MitjaNemec/Kicad_action_plugins #4: Replicate layout not replicating all tracks

sww1235, 2018-02-01:
Hello:

Using the [Bugfix_circular_in_y_linear](https://github.com/MitjaNemec/Kicad_action_plugins/tree/Bugfix_circular_in_y_linear) branch

Unfortunately, I didn't notice if this happened on nightly 2017-09-29, but it does happen on 2018-02-01.

What appears to happen is that the bounding box for replication appears to be just around the footprints and does not take tracks into account. 

The top layout is the original layout, the bottom layout is the replicated layout but missing the top most and bottom most traces of the original.

[image]

MitjaNemec, 2018-02-02:
Thanks for finding this out.

Yes, this is inherent to the workings of the plugin, and I did not spot it as I tested it only with tight layouts

The easiest way to find which track are meant to be replicated is just to select the tracks which are contained/intersecting the bounding box of the modules which are to be replicated.

There is probably a way to find the tracks which are on the nets which connect only modules which are to be replicated.

I'll leave this open until I find the time to address this

MitjaNemec, 2018-02-02:
Try the latest commit "7947a7c" in "Bugfix_circular_in_y_linear"

sww1235, 2018-02-02:
I tried the plugin as of commit 7947a7cc4129f1005fc2012279e01f0959c1108a and got the following results. It looks like some of the traces were replicated, but got moved from their module relative position, while others didn't. I used all 4 checkboxes this time to test, so I could provide a clean image.

Its not a huge issue for me, as the majority of my tracks are inside, its just 1 or 2 that stray a little bit. If you can't figure out a way around the bounding box issue, just put a bigger warning in the readme.

[image]

[image]

MitjaNemec, 2018-02-06:
Would you mind sending me layout (.kicad_pcb) for testing as I can not reproduce the issue. You can delete part of the layout which is not required for me to test the replication.

sww1235, 2018-02-06:
The current kicad_pcb file for the project is at https://github.com/sww1235/edc-mux-board/blob/ard-eth-no-pwr-distro/edc-mux-board.kicad_pcb

There are actually several repeated modules on the sheet so it will be a good test.

MitjaNemec, 2018-02-17:
Hi again.

Thanks for the project, but I can not reproduce the track replication issue with 2018-02-13.

Anyhow, I've changed the radio button replication, so if you would be so kind and test the 60cf82c in the "Bugfix_circular_in_y_linear" branch and report back I would really appreciate it.

closed 2018-02-22 by MitjaNemec