MitjaNemec/Kicad_action_plugins #6: archive_project doesn't work depending on language setting

automesslar, 2018-05-22:
I am using german language and the closing of pcbnew at the end of the process doesn't work.
This is because the key combination changes with language. When I change the language to english it works fine.
I modified my .py file to use the german combination wich is Alt+d ->b ->tab ->enter and it works fine, too.

But nonetheless a great plugin, thanks for your work!

MitjaNemec, 2018-05-23:
Thanks for spotting it. As there I don't see any way of solving it, I am going to leave this issue open, and I'll add a documentation describing and a possible way of solving the issue.

hildogjr, 2018-09-23:
Just to understand, this is issue is KiCad current language dependent or dependent of the which language was used to KiCad that save the file by the last time?

MitjaNemec, 2018-09-23:
The issue is with which language of KiCad interface is selected when the plugin is run.

As the plugin is editing .sch and .lib files eeschema has to be closed, obviously, and the plugin checks for this. The plugin is also editing .kicad_pcb file as a text file and for this it opens the file. Thus the .kicad_pcb file is opened by pcbnew and the plugin in parallel. Obviously this can lead to problems and to prevent this, when the plugin finishes, it closes the pcbnew, blocking the pcbnew to override file changes done with plugin. And the only way I found to do this is to emulate keypress events. So the plugin is hitting the keypress event which close pcbnew when the interface language is english. 

I expect to get rid of this in 5.1 (not 5.0.1) when I'll be able to change 3D model settings from within the pcbnew python API (big thanks goes to Andrew Lutsenko who supplied the patch to pcbnew source)

MitjaNemec, 2019-03-20:
This was fixed in 4f501e10, closing the issue

closed 2019-03-20 by MitjaNemec