MitjaNemec/Kicad_action_plugins #7: Broken dialog in PCBnew

herostrat, 2018-05-29:
Thanks for the plugins!

I am using the kicad 5rc2
It seems that the plugin dialog is broken here.

The dialog when executing:
[image]

Clicking the radiobuttons does not change that both are ticked.
I can however switch to the circular window and back.
[image]

Observation 1:
I can input the linear data. If I execute the script, it is interpreted as circular data (x offset being the radius and y offset the angle)

Observation 2:
Circular execution is working as expected

My guess is that in the dialog initialization something is broken.
With a quick lock in the code didn't find the reason.

If you need any more informations or pics just let me know.

Thanks a lot!

MitjaNemec, 2018-05-30:
Hi,

Thanks for reporting.

I'll not be able to start fixing it until 6.6.2018.

In the mean time, can you tell me what OS did you run this on?

And if you are in a hurry, you could try any commit from Feb 2nd. I was changin GUI to solve an issue in MacOS and it might trigger a regression in other oses.

herostrat, 2018-06-01:
Hey,  thanks for the fast answer.

I am running KUbuntu 17.10 and KiCad5r2

Your guess is true, it seems the GUI fix triggered the regression.
I testes df8c6fb6240bae4f50a3ac327d447a8297917517 and this version worked flawlessly.

Let me know if you want me to test your changes, if you find time for it :)

MitjaNemec, 2018-06-05:
@herostrat thanks for the testing. I really appreciate it. Keep in mind there were a few bugs since that commit that were fixed. You might get by only taking `action_replicate_layout.py` from that revision.

@spyro2 you have sharp eyes. As I don't change the documentation with each new revision, the video/animation still shows older revision.

Anyhow it seems that implementation with wxRadioButton has a problem in Ubuntu (or maybe even in other Linux distributions) and implementation with wxRadioBox has a problem on MacOS. Both implementations work in Win7 (and hopefully even 8, 8.1 and 10). I'll have to figure out how to properly solve this.

spyro2, 2018-06-07:
The correct solution would be to file a bug with Ubuntu (and Debian?), against wx, I think.

MitjaNemec, 2018-06-07:
I am not really certain it is a bug in Ubuntu or MacOS, so it would not be fair to raise a bug report. As said, I am looking for a better solution and it looks like I will replace a radiobutton/radiobox with ToggleButton.

MitjaNemec, 2018-06-11:
Fixed by Seth Hillbrand

closed 2018-06-11 by MitjaNemec