MitjaNemec/Kicad_action_plugins #17: Create a 'update mode' on `replicate_layout`

hildogjr, 2018-09-14:
`replicate_layout` already have the XY align and the circular align, so create a 'update mode' on `replicate_layout` to just update the already created replicated layouts. Use the selected component in the base layout as offset/origin in all the other ones; so the position of these components will not change in Pcbnew.
This may be useful to update the various replicated layout after integrated in the big layout.

hildogjr, 2018-09-19:
Exactly.
About the step 4. Because we usually select one component of the pivot (master) to use the plugin, it could be use this orientation (XY and angle) to others.
This should be useful when the different replicated layouts are already placed on the PDB and I just want to make same track/footprint update that will affect all them.

MitjaNemec, 2018-09-24:
Partial fixed pushed right now, but I don't know if supporting circular layout is even possible without additional input from the user

hildogjr, 2018-11-15:
Closing this as fixed by the function "Grab offset from layout".

closed 2018-11-15 by hildogjr