MitjaNemec/Kicad_action_plugins #21: Replicate silk annotations

hildogjr, 2018-09-24:
A interesting feature would be replicate also (by one checkbox option) replicate the silk annotation inside the boundary of the hierarchical layout.

Procedure (internal idea to the algorithm):
1) Replicate the parts and tracks;
2) Check the silk that are inside (or at least some part inside) the boundary defined by the already made layout (we have to check how to do this with the `pcbnew` package);
3) Replicate this silk annotations.

MitjaNemec, 2018-09-25:
Doable via board.DrawingsList(). I'll have to figure it out how to implement it in within the plugin though

hildogjr, 2018-09-25:
Maybe a workflow:
1) During the parts and tracks replicate action, collect the maximum boundary of each footprint/track and keep in a variable. Iterating and replace by maximum and minimum during the iteration;
2) Finish all the parts and tracks procedures;
3) Iterate the `board.DrawingsList()` and replicate that one that cross the above `boundary` (draws which some point of the boundary are internal of the boundary calculated on (1) ).

MitjaNemec, 2018-09-25:
At the moment, this (general concept) is not doable for the lack of API. The `board.GetDrawings()` or  `board.DrawingsList()` returns list of EDA_RECT obecjts. And EDA_RECT object lacks methods to check on which layer it is or to put it on a certain layer. I'll leave the issue open, but until the pcbnew API changes I don't see a way of solving it

qu1ck, 2019-04-05:
`board.DrawingsList()` returns BOARD_ITEM list which do have `Get/SetLayer()`

MitjaNemec, 2019-04-08:
Thanks for the info. I really appreciate it. So the issue is back on my TODO list

MitjaNemec, 2019-04-09:
Fixed. Though I have to say I really feel silly that I did not spot the option as I already used `board.GetDrawings()` for text replication.

closed 2019-04-09 by MitjaNemec