MitjaNemec/Kicad_action_plugins #25: Replicate layout: Multiple copies end up at the same position

gohai, 2018-11-19:
First: thank you for this plugin, I greatly appreciate your work!

I have the following issue running the latest c3395f2b35358ba93c5a2d6709b510e64ada9bc8: my board has two copies of a single hierarchical sheet. I want both copies to each adapt layout and traces from the third (pre-configured) copy. But when I run the script, the two instances always end up at the same (overlapping) position. This also happens when I try to deselect one of the two copies in the dialog.

Am I doing something wrong, or is this a genuine issue with the plugin?

Demonstration: https://sukzessiv.net/~gohai/kicad/overlap.mp4

If this matters: here is my hierarchy
* NOT2
* CLK
  * instance of NOT2
  * instance of NOT2

MitjaNemec, 2018-11-19:
I am not certain this is really a bug. Could you please attach `replicate_layout.log` which should be in the project folder.

gohai, 2018-11-20:
Here are the lines that get added to `replicate_layout.log` when I do what I did in the screen capture:

```
11-20 09:10:34 replicate_layout.action_replicate_layout 293:Action plugin Replicate layout started
11-20 09:10:34 replicate_layout.action_replicate_layout 327:Preparing replicator with Q10 as a reference
11-20 09:10:34 replicate_layout.replicatelayout 128:Initializing replicator object
11-20 09:10:34 replicate_layout.action_replicate_layout 331:Calculating initial spacing
11-20 09:10:34 replicate_layout.replicatelayout 193:Calculating spacing
11-20 09:10:34 replicate_layout.action_replicate_layout 335:Showing dialog
11-20 09:10:40 replicate_layout.replicatelayout 193:Calculating spacing
11-20 09:10:46 replicate_layout.replicatelayout 193:Calculating spacing
11-20 09:10:46 replicate_layout.action_replicate_layout 379:Replicating layout
11-20 09:10:46 replicate_layout.replicatelayout 985:Replicating layout
11-20 09:10:46 replicate_layout.replicatelayout 330:Preparing for replication
11-20 09:10:46 replicate_layout.replicatelayout 662:Replicating modules
11-20 09:10:46 replicate_layout.replicatelayout 807:Replicating tracks
11-20 09:10:46 replicate_layout.replicatelayout 891:Replicating zones
11-20 09:10:46 replicate_layout.replicatelayout 775:Replicating text
11-20 09:10:46 replicate_layout.action_replicate_layout 386:Replication complete
```

MitjaNemec, 2018-11-20:
Thanks for your effort, but sadly I did not get enough info as my instrumentation logs just some basic info .

So I'd like to ask you if you can run the newest version of the plugin a821ee4 and send me the log file again as it should offer more detail. Or you can just send me whole project.

gohai, 2018-11-25:
Thank you for putting in this extra instrumentation.

Running the most recent commit I get the following output:

```
11-25 11:00:00 replicate_layout.action_replicate_layout 295:Action plugin Replicate layout started
11-25 11:00:07 replicate_layout.action_replicate_layout 295:Action plugin Replicate layout started
11-25 11:00:07 replicate_layout.action_replicate_layout 329:Preparing replicator with Q7 as a reference
11-25 11:00:07 replicate_layout.replicatelayout 128:Initializing replicator object
11-25 11:00:07 replicate_layout.action_replicate_layout 333:Calculating initial spacing
11-25 11:00:07 replicate_layout.replicatelayout 193:Calculating spacing
11-25 11:00:07 replicate_layout.action_replicate_layout 337:Showing dialog
11-25 11:00:15 replicate_layout.replicatelayout 193:Calculating spacing
11-25 11:00:17 replicate_layout.replicatelayout 193:Calculating spacing
11-25 11:00:17 replicate_layout.action_replicate_layout 381:Replicating layout
11-25 11:00:17 replicate_layout.replicatelayout 986:Replicating layout
11-25 11:00:17 replicate_layout.replicatelayout 987:Will replicate sheets:
	[u'5BEB8B38']
11-25 11:00:17 replicate_layout.replicatelayout 988:Pivot modules are:
	Q6
	Q7
	R2
11-25 11:00:17 replicate_layout.replicatelayout 330:Preparing for replication
11-25 11:00:17 replicate_layout.replicatelayout 662:Replicating modules
11-25 11:00:17 replicate_layout.replicatelayout 676:Replicating module: Q1
11-25 11:00:17 replicate_layout.replicatelayout 676:Replicating module: Q2
11-25 11:00:17 replicate_layout.replicatelayout 676:Replicating module: Q3
11-25 11:00:17 replicate_layout.replicatelayout 676:Replicating module: Q4
11-25 11:00:17 replicate_layout.replicatelayout 676:Replicating module: R3
11-25 11:00:17 replicate_layout.replicatelayout 676:Replicating module: R4
11-25 11:00:17 replicate_layout.replicatelayout 808:Replicating tracks
11-25 11:00:17 replicate_layout.replicatelayout 892:Replicating zones
11-25 11:00:17 replicate_layout.replicatelayout 776:Replicating text
11-25 11:00:17 replicate_layout.action_replicate_layout 388:Replication complete
```

Please also find my project (stripped down to just show this issue) here: [reproducer.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/2613392/reproducer.zip)

Thanks again!

MitjaNemec, 2018-11-27:
Thanks for the report and the project. As I am currently quite busy it will be a while before I start working on this, but I will look into it

MitjaNemec, 2019-02-03:
So as of today, there is a partial fix for this issue. But I've uncovered a big architectural issue regarding complex hierarchy setups regarding replication. As I have plans to completely rewrite the plugin in order to enable replication with custom layout (as per https://forum.kicad.info/t/replicate-layout-action-plugin/8539/13?u=mitjan) I will have to think whether I address this issue in full, or I stop the development here, and proceed with rewrite.

gohai, 2019-02-09:
Thank you @MitjaNemec - I just tried out your changes locally, and those help me a lot in what I was trying to accomplish. This is greatly appreciated!

MitjaNemec, 2019-02-09:
Thanks for the feedback.

closed 2019-02-09 by MitjaNemec