MitjaNemec/Kicad_action_plugins #27: Swap pins fails sometimes

robert128, 2018-12-01:
Hello Again,

Although you fixed swap_pins, I've found that it fails to move a pin that has already been swapped.  I think it's related to the way the nets are drawn on the schematic.  When swap_pins moves a pin, it's moving the label to a non-existent net on top of the symbol pin instead of moving the label with it's attached net.  If I fix the schematic to include a net, then the swap works.

The anchor for the label could use some adjustment if it's possible.  Looks like labels have their anchor on the side of the text that is closest to the symbol, so when labels are moved from one side of a symbol to the other, maybe the anchor can move as well.

Here is the exception I get when I try swap a pin that has already been swapped before:

[image]

and here is what the schematic looks like as a result of swapped pins:

[image]

robert128, 2018-12-04:
After a day of swapping without trouble, I finally checked the schematic against the layout, and found that the schematic often wasn't getting the changes is the layout.  I had to do many of the changes by hand.

I'm sorry I haven't been able to identify any clear patterns for what's happening.  I thought it was all related to the labeling issue I highlighted above, but now I'm not so sure.

Here's a new log if that helps.
[swap_pins.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/2645793/swap_pins.log)

MitjaNemec, 2018-12-04:
Firstly, I apologize for the data loss.

Secondly, looking at the logs, you are doing some serious testing, and I really appreciate it. I've written the plugin and I've tested it on real projects that I am working on, but my designs are much simpler.

Thirdly, I've located the bug which caused a crash mentioned in the first [post](https://github.com/MitjaNemec/Kicad_action_plugins/issues/27#issue-386495298) and I've pushed the fix.

Regarding the labeling issue, from the your figure did you have the labels connected directly to the pins before swapping? If so, is the main problem the orientation of the labels?

And finally for the data loss, I have one suspect, but as it is late and my head is not working anymore, I am going to deal with this tomorrow.

robert128, 2018-12-06:
Here's another exception and log:

[image]

[swap_pins.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/2651115/swap_pins.log)

MitjaNemec, 2018-12-06:
Man you are really testing it.

This looks like a new issue. If you can recreate it I'd like you to update the plugin as I should have a bit more instrumentation available.