MitjaNemec/Kicad_action_plugins #29: pad2pad_track_distance - support entire nets

evanshultz, 2019-01-08:
The existing functionality is impressive! Thank you!

Especially when doing designs that have primary circuits connected to an AC mains and secondary circuits which need isolation, there are creepage rules. This will apply to an entire net and not just the pads/tracks on the net (also zones, for example). It would be great to have the tool that checks the distance between one net and another net, combing through each element on the net to find the closest distance between nets.

MitjaNemec, 2019-02-20:
Here you go, I've published `net2net distance` action plugin. It works only on tracks, and it is unlikely that it is possible to extend it to account also for zones.

evanshultz, 2019-02-25:
I understand that you have separated the pad2pad and net2net plugins based on if the selected elements (right now only pads) are on the same net. It would be nice to have a single plugin that work regardless of if the elements are on the same net or not.

MitjaNemec, 2019-02-26:
I don't see it. One plugin specifically calculates shortest distance of traces between two pads, while the other calculates shortest distance between two nets. It is different functionality. I am open for discussion and you might persuade me otherwise, but at the moment even naming the plugin would be a challenge.

evanshultz, 2019-02-26:
Perhaps it's a matter of function. May I ask what you intended the purpose of this plugin to do? Some of your plugins seem obvious in purpose, but here I'm not sure why you may have made this plugin and it could be that I assume the purpose is different from your actual purpose.

When I saw this plugin, I was initially thinking it could be helpful to find the air gap between two pads. Why would someone want to do this, you ask?

The distance between two conductive elements determines the voltage at which the insulation between those elements breaks down. This is called creepage along the surface of a PCB. Knowing this distance can be helpful in many circumstances, but between primary and secondary circuits in a product it is required for worldwide safety. I mentioned this briefly above but perhaps this detail in helpful?

To perform this function, not only pads but also any copper on a net should be considered. This is why I originally asked about being able to determine distance between nets. The new plugin you created currently requires selecting two pads but it really calculates track2track, if I understand. But at any rate, it can still be helpful for measuring air gap when checking if acceptable creepage distance has been met.

Finding the air gap between pads is what I was hoping to use this plugin for, but it turns out the pad2pad measurement is between the pad centers. And net2net is only the minimum distance between tracks. So it's quite possible that I'm abusing the plugin's use, in which case it's perfectly fine to tell me to leave you alone because the plugins meet your goals. :)

MitjaNemec, 2019-02-27:
net2net functionality is what you describe, obtaining creepage distance. And you should need minimum value only. And yes it is track to track. It will have to be expanded to include track2pad and pad2pad and then print the minimum value. Sadly including zones is not an option from what I can tell.

While pad2pad measures track length between two pads on the same net. I had a design, where I did not have a big voltage tolerance budget and I had to find out how much will voltage drop will I have on specific track. As KiCad is lacking a FEM tool similar to Altium PDN (for now), I just threw together and published this plugin, to get length of tracks between two pads. If you know the track width and thickness, you can get the resistance and from this you get voltage drop at specific current. I could/should update this plugin to calculate the resistance directly, which should be easy enough. Calculating it this way is much faster than using full blown FEM. Also the plugin could act as a foundation if you'd want to obtain track self inductance (somewhat more difficult to calculate, but should be doable). On the other hand with link between KiCad and FreeCad getting better and better it is just a matter of time that FreeCad gets support for electrostatic FEM, and then doing FEM analysis will be much easier.

Come to think of it FreeCad should probably be much better for getting creepage distance between two copper objects including zones. Did you consider this?

evanshultz, 2019-02-28:
Considering pads and tracks makes sense.

To catch creepage, though, it will be missing the air gap part. If the plugin checks the center of a pad and the center of a track's width, half of the pad and half of the track width will be missing. So the air gap is not the resulting number given by the tool and thus the creepage distance is not what is reported. Your plugin would need to understand pad geometries (including custom ones) and track width to give a better number. All but custom pad shapes are relatively straightforward to process, but you would need to know the exact location of the minimum distance (displaying that distance as a line on the PCB would be instructive) to properly take into account the pad and track geometries. Or possibly just consider easy geometries and then warn the user if a custom pads or at least one zone are on either net so they can be aware that the number given may not be correct.

Thanks for the explanation. That makes sense. I've used the PDN tool in Allegro to do the same thing.

I have not tried FreeCAD for this. Actually, I'm not sure where to start but it sounds like a good idea. Have you explored that option?

MitjaNemec, 2019-02-28:
At the moment the plugin accounts for track width. But as I've just fooled around with FreeCad [Kicad copper parser](https://github.com/realthunder/fcad_pcb) it really does not make any sense for me to put any additional effort in this plugin. Using this parser you can import the copper and using FreeCad measurement tools you can easily get creepage distance including zones and pads.