MitjaNemec/Kicad_action_plugins #30: KiCad 5.02, replicate layout

PerMalmberg, 2019-01-19:
After installing the replicate plugin in KiCad 5.02 (Ubuntu 18.04), PCBNew crashes on every start up.

Is there a log somewhere from where I can get more information?

MitjaNemec, 2019-01-19:
Can you paste exact version info (kicad project manager/help/copy versiin info)

PerMalmberg, 2019-01-19:
Here it is:

Application: kicad
Version: 5.0.2-bee76a0~70~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.0g zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-43-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 2.24
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.3.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

PerMalmberg, 2019-01-19:
Hm,can KICAD_SCRIPTING_WXPYTHON being off have something to do with it?

If so, can it be enabled with less than a recompile?

MitjaNemec, 2019-01-20:
Yup, this is the issue. As I am familiar only with MS Windows any tip from is untested. But for some distributions, the solutions was to install [wxpython built against gtk2](https://forum.kicad.info/t/arch-linux-problem-with-wxpython-3-0-2-0-7-reverting-to-3-0-2-0-6-works-fine-kicad-nightly/9493/4?u=mitjan).

Other than that, recompilation would be a solution, though you might want to check if there is a nightly version for your distribution available and built against gtk3 (thought, this might have other issues, as KiCad support for GTK3 has been added only recently and from what I know has not been tested seriously).

I will add a proper handling and notification for the user for this issue as the user experience that you went through is really really bad. And when I do, I'll ask you to test it.

And if you can not solve the issue, you can delete the `__init__.py` and `action_replicate_layout.py` so that KiCad will at least start and you can use `replicatelayout.py` in the pcbnew scripting console. The API is not really user friendly as I developed it mainly toe ease the testing, but you can look at the examples in the `test_*` functions from line [1003 ](https://github.com/MitjaNemec/Kicad_action_plugins/blob/master/replicate_layout/replicatelayout.py#L1003)forward. Obviously when running from pcbnew scripting console you load the currently open board with pcbnew.GetBoard() and you save the board from pcbnew GUI.

MitjaNemec, 2019-01-20:
And if you don't have pcbnew scripting console, you can run the `replicatelayout.py` outside of KiCad. Then you need to load the boards with filename, and save it as in the `test_*` examples

PerMalmberg, 2019-01-20:
I'm installing KiCad nightly 5.1 which specifically addresses this issue. iiuc.

PerMalmberg, 2019-01-20:
Turns out the nightly is v6 :\ With KICAD_SCRIPTING_WXPYTHON=OFF too.

PerMalmberg, 2019-01-20:
[v 6 is apparently 5.1](https://forum.kicad.info/t/solved-pcbnew-crashing-linux-v5-02/14774/9)

Anyway, I tried using the scripting console, but it seems not even that is included, at least I can't find it in any menu. Is there a trick to open it?

MitjaNemec, 2019-01-20:
Sorry to mislead you, if you don't have WXPYTHON, then you don't have scripting console

PerMalmberg, 2019-01-21:
Yeah, I figured that out.

I ran the script in Windows yesterday so I'm good for now. Once KiCad 5.1 is released this will no longer be an issue iiuc.

I'm happy to try it out if you make changes to prevent the crashes.

MitjaNemec, 2019-01-21:
Here you go. I'd like to know if pcbnew still crashes, and if it does not, what happens when you run the plugin. You should get a messagebox (which might not be in foreground or will loose focus on any mouse move) informing you of wxpython issues.

PerMalmberg, 2019-01-23:
Sorry for the delay.

I tried it now, in Ubuntu Linux 18.04.1, KiCad as per previous version info.

Sadly, it still crashes without any message.

MitjaNemec, 2019-01-23:
Then it is out of my hands, with what is doable within KiCad. Thanks for the testing though.