MitjaNemec/Kicad_action_plugins #32: place_footprints only arranges selected components, not all elements on sheet

capaulson, 2019-02-19:
I've been exploring the capabilities of the new place_components and noticed it only moves the component selected. This is when 'Sheet Number' is selected in the "Place by?" dialog. I'm digging into the code, but I wanted to flag the issue.

MitjaNemec, 2019-02-19:
This is intentional and it is intended to supplement revised `replicate_layout` plugin.

For more flexible operation of `replicate_layout` plugin I had to split the placement part from replication part.

The workflow for sheet replication now is:
1. do a complete layout of one sheet instance (placement, layout, ...)
2. place anchor footprints for other sheet instances either manually or using `place_footprints` "Place by sheet" functionality
3. replicate layout to other sheets using `replicate_layout`

capaulson, 2019-02-19:
Perfect! It worked. I did get an error in the matrix option. I got an issue where it wasn't getting an int. I think it's just missing .Get value().

MitjaNemec, 2019-02-19:
Thanks for spotting the bug. It has been fixed.

closed 2019-02-19 by capaulson