MitjaNemec/Kicad_action_plugins #35: pad2pad: Crash using latest nightly build

evanshultz, 2019-02-25:
When using the KiCad version at the bottom of this post, from 24 Feb 2019, I see the following error message:
[image]

Note that net2net works for me.

Application: pcbnew
Version: (5.1.0-rc1-136-g8ad15b1d6), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.61.1 OpenSSL/1.1.1 (WinSSL) zlib/1.2.11 brotli/1.0.6 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) nghttp2/1.34.0
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.68.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.61.1
    Compiler: GCC 8.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

MitjaNemec, 2019-02-26:
Can I assume, that in previous versions of KiCad the plugin worked correctly with same `.kicad_pcb` and the crash is specifically tied to KiCad version?

Or it might be a crash related to this specific design?

evanshultz, 2019-02-26:
I do not recall what version or board file I may have used before. So sorry, I can't give any certain feedback.

The same error message I reported can be found in the KiCad version above using the board at https://github.com/DavidGriffith/g3ruh-modem/blob/master/g3ruh-modem.kicad_pcb (just a random one online so we can both try it) when selecting U1.14 and U2.4. With a bit more testing it appears the error appear when selecting two THT pads. With two SMT pads or one of each type I do not get an error, at least in testing so far. It is probably more nuanced than that but at least you have a test case now.

MitjaNemec, 2019-02-27:
Thanks for this. I'll look into it, but my first hunch is that my handling of THT pads is buggy as I did not test the plugin with THT pads

MitjaNemec, 2019-02-27:
Should be fixed now

evanshultz, 2019-02-27:
Thanks! The test case I gave above works, along with several others.

I get the same crash using pad2pad from U1.1 to U2.2 on the same board. After a bit of playing around, perhaps because of non-round pads?

MitjaNemec, 2019-02-28:
It is not a sqare pad, it is something else and it will require deeper probing. Thanks for the report though.