MitjaNemec/Kicad_action_plugins #36: Open issues with python3

T0jan, 2019-03-22:
As mentioned in the kicad forum thread, there are open issues with the scripts on python3 systems.  only swap_pins, swap_units and Pad2Pad_Distance plugin appear in the external plugin list after refreshing, every other plugin is missing.

Pad2Pad_distance raises an AttributeError: module 'pad2pad_track_distance' hads no attribute 'Distance' for code line 95 for action_pad2pad_track_distance.py

swap_pins raises an AttributeError: module 'wx' has no attribute 'wx' for line 49 in action_swap_pins.py

swap_units raises an AttributeError: module 'wx' has no attribute 'wx' for line 49 in action_swap_units.py

Kicad info:

```
Application: kicad
Version: 5.1.0-060a0da~80~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.0g zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-46-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.3.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

T0jan, 2019-03-22:
other weird behavior:

while refreshing kicad seems to run once the replicate layout plugin for the first hierarchical sheet but only places the components in the default grid and ignores any subsheets.

MitjaNemec, 2019-03-22:
First of all, thank you for reporting and testing. I really appreciate this.

The pad2pad bug is a weird one there might be an issue regarding module imports. I'll have to study how this change in py3.

I've fixed the swap_pins and swap_units bug (it was also present in archive_project)

Do you have any `.log` files in the project folder where you tested the plugins? If you do please attach them.

Also does the [InteractiveHtmlBom](https://github.com/openscopeproject/InteractiveHtmlBom) work with your setup. If it does, this will give me pointers how should I approach this

MitjaNemec, 2019-03-22:
Also, can you run pcbnew, open scripting console and type the following, and copy-paste the output here
```
import pcbnew
print pcbnew.NOT_LOADED_WIZARDS
print '---'
print pcbnew.FULL_BACK_TRACE
print '---'
```

T0jan, 2019-03-22:
> I've fixed the swap_pins and swap_units bug (it was also present in archive_project)

new bug

```
Traceback (most recent call last):
  File "~/.kicad_plugins/swap_pins/action_swap_pins.py", line 113, in Run
    swap_pins.swap(board, pad1, pad2)
AttributeError: module 'swap_pins' has no attribute 'swap'
```

> Do you have any .log files in the project folder where you tested the plugins? If you do please attach them.

only swap_pins generated a log , only with the bug message above inside

> Also does the InteractiveHtmlBom work with your setup. If it does, this will give me pointers how should I approach this

nope doesn't work either. 

```Traceback (most recent call last):
  File "~/workspace/hardware/tools/kicad-bom-script/InteractiveHtmlBom/generate_interactive_bom.py", line 9, in <module>
    import pcbnew
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 38, in <module>
    import _pcbnew
ImportError: dynamic module does not define init function (init_pcbnew)
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 63, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 30, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 23, in <module>
    from apport.packaging_impl import impl as packaging
  File "/usr/lib/python3/dist-packages/apport/packaging_impl.py", line 24, in <module>
    import apt
  File "/usr/lib/python3/dist-packages/apt/__init__.py", line 23, in <module>
    import apt_pkg
ImportError: No module named apt_pkg

Original exception was:
Traceback (most recent call last):
  File "~/workspace/hardware/tools/kicad-bom-script/InteractiveHtmlBom/generate_interactive_bom.py", line 9, in <module>
    import pcbnew
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 38, in <module>
    import _pcbnew
ImportError: dynamic module does not define init function (init_pcbnew)```

> Also, can you run pcbnew, open scripting console and type the following, and copy-paste the output here

```
import pcbnew
print pcbnew.NOT_LOADED_WIZARDS
Skip subdir~/.kicad_plugins/__pycache__
~/.kicad_plugins/no_wxpython.py
print '---'
---
print pcbnew.FULL_BACK_TRACE
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 3567, in LoadOnePlugin
    mod = __import__(ModuleName[:-3], locals(), globals() )
  File "~/.kicad_plugins/no_wxpython.py", line 23, in <module>
    import Tkinter as tk
ModuleNotFoundError: No module named 'Tkinter'

print '---'
---
```

> First of all, thank you for reporting and testing. I really appreciate this.

you are welcome, its also in my interest to get it running again :grin:

MitjaNemec, 2019-03-23:
There seems to be a difference how the python3 handles relative imports, but currently my knowledge of the topic is just superficial, so I am mostly tinkering. I've changed how I handle relative imports and when you find the time, please test it. If it still does not work, I'll have to put some serious effort into this.

T0jan, 2019-03-25:
sorry had no access to my machine during the weekend.

First of all the good news: 

* kicad now recognizes all plugins except archive_project and delete_selected
* Length stats seems to work completely 
* Net2Net distance seems to work completely

the bad news: several bugs

* swap_units: only works for components on schematic main sheet. components in hierarchical sheets generate:
```
"swap_units.py", line 181, in swap 
unit_1_sch_file = current_sch_file[:unit_1_loc] + unit_2 + current_sch_file[unit_1_loc + len(unit_1):]
UnboundLocalError: local variable 'unit_1_loc' referenced before assignment
```
* swap_pins: seems also only to work on components on first level of schematic. on subsheet components I get the errror:
```
"swap_pins.py", line 112, in swap
logger.info("Looking for: " relevant_sch_files[0][3] + "in chache.lib")
IndexError: list index out of range
```
* Pad2Pad distance: works for normal via connections, crashes for unconnected or zone connected pads
* Place footprints: generates two different crashes
  * for Reference nr:
```
place_by_reference_GUI.py, line 22, in __init__
self.SetSizeHintsSz(wx.Size(257,-1),wx.DefaultSize)

python3/dist-packages/wx/core.py, line 82, in deprecated_func
return item(*args)

python3/dist-packages/wx/core.py, line 2250, in _Window_SetSizeHintsSz
return self.SetSizeHints(minSize, maxSize, incSize)

TyperError: SetSizeHints() takes 3 positional arguments but 4 were given
```
  * for Sheet nr:
```
place_by_reference_GUI.py, line 22, in __init__
self.SetSizeHintsSz(wx.Size(257,-1),wx.DefaultSize)

python3/dist-packages/wx/core.py, line 82, in deprecated_func
return item(*args)

python3/dist-packages/wx/core.py, line 2250, in _Window_SetSizeHintsSz
return self.SetSizeHints(minSize, maxSize, incSize)

TyperError: SetSizeHints() takes 3 positional arguments but 4 were given
```
```
action_place_footprints.py, line 55, in SetSizeHints
super(InitDioalog.self).SetSizeHints(sz1, sz2)

Type Error: super(type, obj): obj must be an instance or subtype of type
```
* Replicate Layout: raises no exceptions but not recognizes all sheets and moves also unselected module components around

MitjaNemec, 2019-03-25:
> 
> sorry had no access to my machine during the weekend.
> 
You really have nothing to be sorry about. Everybody has his own life with schedules, family, various commitments ... I can only be happy that you are contributing with testing. And if you would stop without notice, I can only say "it was fun while it lasted" but I'll be the last to judge you as I don have the slightest idea about you, your life and other circumstances. The same goes for me. I try to be prompt but sometimes it might take me a couple of weeks to respond. /philosophizing off

The rest you response is pure gold.
1. `swap_units`: It does not look like a py3 issue. Could you take a look at `.log` file and attach it here? If they reveal too much information could you find any other way of sending them over?
2. `swap_pins`: the same as 1
3. `Pad2Pad`: The plugin does not work with zones and assumes there is connectivity. I'll have to add proper checking so that the failure is graceful with meaningful output. Thanks for spotting this.
4. `Place_footprints`: this is either a py3 issue or it might be a combo of wxpython/py3 issue
5. `Replicate_layout`: Could you take a look at `.log` file and attach it here? Or if you don't mind sharing your project that would be even better.

MitjaNemec, 2019-03-25:
Also if you could test new version (21cf9ae), as I've refactored module importing which should resolve some issues I've been having from the last change when I try to run the plugins in standalone mode

T0jan, 2019-03-26:
> swap_units: It does not look like a py3 issue. Could you take a look at .log file and attach it here? If they reveal too much information could you find any other way of sending them over?

There are no new entries in the `.log` file for this error. it only logs an error with the logging process. 
[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3008587/swap_units.log)

> swap_pins: the same as 1

[swap_pins.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3008602/swap_pins.log)

> Replicate_layout: Could you take a look at .log file and attach it here? Or if you don't mind sharing your project that would be even better.

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3008710/replicate_layout.log)

The best I can give you is an overview over the hierarchical structure and my intended replication:

[image]

as you probably can image I want in general to replicate the routing of each of the modules over to the other ones of the same kind. some of the modules need some different routing but this should be not important for the plugin as far as i know?

selecting the anchor components of each module and then running the plugin generates the following screens:

[image][image][image][image]

Another remark: due to the error of the logger, only the first plugin run in a session only seems to log its behavior

> Also if you could test new version (6e40bad), as I've refactored module importing which should resolve some issues I've been having from the last change when I try to run the plugins in standalone mode

this fix seemed to fix the crash of the Place_footprints plugin. nevertheless it has the same issue recognizing my structure as the Replicate_layout

all other plugins work (or not) like before, archive_project and delete_selected are not included by Kicad

MitjaNemec, 2019-03-26:
OK there is a lot of different bug hiding in there. Again, thank you for your patience

I should have resolved the logger issue, and I've added some instrumentation so please do the following:
1. update the plugins' code
2. Test the swap_units and send me the log. 
3. Test the swap_pins and send me the log. And if you could also send me the partial .sch file contents in which the R8101 is located (everything between $Comp and $EndComp delimiters should be fine)
4. Run the replicate_layout again and send me the .log file so that I should have better overview of your project hierarchy.
5. Look in archive_project and delete_selected plugin folders and send me the corresponding `_error.log` files. I've added additional instrumentation which should give me a hint what is the problem here and with delete_selected, I've fixed a couple of bugs so it might appear anyway (you will not see a `_error.log` file in this case

T0jan, 2019-03-27:
> OK there is a lot of different bug hiding in there. Again, thank you for your patience

You are welcome, I have some scripts written by myself so I know the pain with bugs well enough. but its also part of the fun

> Test the swap_units and send me the log.

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3012775/swap_units.log)

> Test the swap_pins and send me the log. 

[swap_pins.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3012791/swap_pins.log)

> And if you could also send me the partial .sch file contents in which the R8101 is located (everything between $Comp and $EndComp delimiters should be fine)

[schematic_snippet.txt](https://github.com/MitjaNemec/Kicad_action_plugins/files/3012811/schematic_snippet.txt)

had to alter it a bit, hopefully it has some use for you

> Run the replicate_layout again and send me the .log file so that I should have better overview of your project hierarchy.

for modules where it does not find any other module:
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3012835/replicate_layout.log)

for modules where he does find only one other (every time the same, regardless which other module is selected) and touches every time the same components of a not selected module:
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3012847/replicate_layout.log)

> Look in archive_project and delete_selected plugin folders and send me the corresponding _error.log files. I've added additional instrumentation which should give me a hint what is the problem here [...]

[Archive_project_error.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3012857/Archive_project_error.log)

delete_selected now works without any issues in first test cases

MitjaNemec, 2019-03-27:
swap_pins and swap_units crash when dealing with units on sheets which are referenced multiple times in hierarchy. I've opened another issue as it is not python3 related, but I will keep you posted in this thread. It might take me some time to build proper test case, fix the bug, test the solution and publish the fix

As for the replicate layout, I am still fumbling in the dark, so I've added additional instrumentation and I am asking you again to run the plugin and send me the .log file

And finally with you help (.log file) I've located and fix this python3 compatibility bug. If archive_project is not availible send me the _error.log file. If it is, please run the plugin, but make sure that you backup your project beforehand. and then send me the .log file

T0jan, 2019-03-28:
In all plugins: the problem with the logger is still there, which makes it a bit painfull to handle the logging itself.

```
action_replicate_layout.py, line 116 in Run 
disable_existing_loggers=False)

/usr/lib/python3.6/logging/__init__.py, line 1829, in basicConfig 
raise ValueError('Unrecognised argument(s): %s' % keys)

ValueError Unrecognised argument(s):disable_existing_loggers
```

> As for the replicate layout, I am still fumbling in the dark, so I've added additional instrumentation and I am asking you again to run the plugin and send me the .log file

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3017418/replicate_layout.log)

also an interesting behavior I discovered now during testing:
I now tried the Plugin on the mostly finished Board, where all blocks are set, routed etc. There it grabs again random the same components as in the earlier version of the project but sorts them at least partially correct and creates the connected wires and zones (replication at 11:19 in the log file)

Then I have gone back to the earlier stage of the project I tested with before and there it has the same behavior like in the last tests (replication at 11:25)

For me it looks like if it does not took the selected components as replication reference but another one (may the C14908 shown in the Sheets to replicate list?)

> And finally with you help (.log file) I've located and fix this python3 compatibility bug. If archive_project is not availible send me the _error.log file. If it is, please run the plugin, but make sure that you backup your project beforehand. and then send me the .log file

[Archive_project_error.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3017341/Archive_project_error.log)

MitjaNemec, 2019-03-29:
I've managed to also fix swap_pins and it would be great if you could test it.

T0jan, 2019-03-29:
swap_pins works now in several tested cases on capacitors and resistors with two pins but on a IC with multiple pins it raises an exception:

[swap_pins.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3022579/swap_pins.log)

the bug with the logger ssems to be gone for all working scripts :+1:

MitjaNemec, 2019-04-01:
> swap_pins works now in several tested cases on capacitors and resistors with two pins but on a IC with multiple pins it raises an exception:
> 
> [swap_pins.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3022579/swap_pins.log)

I can see no obvious cause for the fault. Can you post the screenshot of partial schematics showing pins AD1 and AD2 of U7801.

BTW, did you managed to test swap_units?

T0jan, 2019-04-01:
> I can see no obvious cause for the fault. Can you post the screenshot of partial schematics showing pins AD1 and AD2 of U7801.

[image][image]

> BTW, did you managed to test swap_units?

yes, for parts with the same unit pin count it works, for parts with different pin counts it raises:

`LookupError: Units to swap have a different pin count`

probably intended?

Did you managed to get behind the problem with replicate layout?

MitjaNemec, 2019-04-01:
Thanks for the feedback. 

I've found a bug in swap_pins, and I've opened a new issue for it (#40). Thanks for spotting this.

Regarding swap_unit, the behaviour is intended. Can you give me an example when would swapping units with different pad count make sense and how it would work?

And finally, I haven't touched `replicate_layout` and once I fix or at least triage #40 I'll put my efforts there. Thanks for understanding and support.

T0jan, 2019-04-01:
> Regarding swap_unit, the behaviour is intended. Can you give me an example when would swapping units with different pad count make sense and how it would work?

In general in multiunit opamps I started to avoid a single unit for the supply pins and added them instead to the first opamp unit. Was partly because it created problems for me with not deleted power pins which placed/hold not used footprints on my layout. also I tried to simplify the unit placement because kicad does not offer an automatic placement of the next unit of a symbol but I have to select them individual. therefore removing one unit reduces error cases and simplifies the placement.

MitjaNemec, 2019-04-01:
Ok, I get it. I don't know exactly how would swapping proceed. If I don't touch the symbols, then the most I can to is::
- check if all lower pin count unit's pins are also on other unit at same locations
- ask the user if he wants to force the swap
- swap the units, leaving the user to sort other pins.

This would leave .sch and .kicad_pcb out of sync. As pcb would see power pins still connected and .sch would see them disconnected (hopefully, as maybe a wire is on new location so the power pins might get connected to different net)

I am not really enthusiastic about this.

T0jan, 2019-04-01:
I am totally fine if you keep it like it is, I only mentioned this because it raised an exception instead of a tooltip which some other misshandlings of the plugin do ("You need to close eeshema and then run the plugin again!" for example"

but one bug I have to mention regarding this:

> This would leave .sch and .kicad_pcb out of sync.

After switching the units in the layout, the plugin also switched them in the schematic. so far so good, but now it gets weird: on updating the the netlist (either with F8 option or with dedicated creation of the netlist) it switches the units only in the layout back.

I also tried switching units of a component which is part of subsheet I use multiple times. Switching units on one of these components does not lead to a change of the schematic and so the component loses synch with the schematic, the next reload of the netlist will undo the switch.

T0jan, 2019-04-02:
> As for the first issue with swap_units, did you managed to keep the .sch open? Maybe in separate process?

no the schematic was closed the whole time, I started the editor afterwards to review the changes and there it looked good but the issue with the reload of the netlists was there. Interesting fact: after pulling the latest updates it worked also after reloading the netlist.

> And the second issue sounds familiar. Can you recall when did you test the swap_units. I am asking because I spotted an issue last week and I've fixed it and your description sound like the issue I've fixed.

this issue is still there.

> And for the swap_pins issue the gist of the issue is that the swap_pins can only work robustly for pins which have labels directly attached to them (for more see #40).

ok good to know, i will track of the other issue and as long as there is no other solution I will keep this limitation in mind

> Can you please test the archive_project plugin again. If it works, I'd like to close the issue and move other issue to separate thread to keep things under control.

still not there:

[Archive_project_error.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3033904/Archive_project_error.log)

(also there was no update on the latest pull?)

> So I am starting to work on replicate_layout. And to locate the issue more precisely, I've added instrumentation, so I would like for you to test it and send me back the log. You don't need to run it to the end, just to the GUI and select each of available levels and then you can cancel.

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3033926/replicate_layout.log)

MitjaNemec, 2019-04-02:
And we go again with archive_project: please test the archive_project plugin again. I've requested for python3 windows build and hopefully I'll receive it so that I can at least solve the syntactical and grammar issues. Until then, you are my only help, and I appreciate it.

MitjaNemec, 2019-04-02:
> 
> no the schematic was closed the whole time, I started the editor afterwards to review the changes and there it looked good but the issue with the reload of the netlists was there. Interesting fact: after pulling the latest updates it worked also after reloading the netlist.
> 

Just to clear any language confusion. By "it worked" you mean that the swap_units plugin worked, or that the issue happened again? If the latter can you save and post the report from the netlist updater, and the swap_units.log

> I also tried switching units of a component which is part of subsheet I use multiple times. Switching units on one of these components does not lead to a change of the schematic and so the component loses synch with the schematic, the next reload of the netlist will undo the switch.

Can you pleas post the swap_unit log for this case.

MitjaNemec, 2019-04-02:
replicate_layout, I am getting closer but I need additional info, so I am asking you to do the same as before ( just to the GUI and select each of available levels and then cancel and then send me the .log)

T0jan, 2019-04-02:
> And we go again with archive_project: please test the archive_project plugin again. I've requested for python3 windows build and hopefully I'll receive it so that I can at least solve the syntactical and grammar issues. Until then, you are my only help, and I appreciate it.

first the good news: kicad now recognizes the plugin

but some problems occurred (of cause):

first a small issue: the GUI item opens with a to small window size.

[image]

bigger issue: after pressing ok, it's crashing.

[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3034588/archive_project.log)

> Just to clear any language confusion. By "it worked" you mean that the swap_units plugin worked, or that the issue happened again? If the latter can you save and post the report from the netlist updater, and the swap_units.log

The script does in this case what it should and also after reopen the schematic and reloading the netlist everything stays switched.

> Can you pleas post the swap_unit log for this case.

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3034633/swap_units.log)

> replicate_layout, I am getting closer but I need additional info, so I am asking you to do the same as before ( just to the GUI and select each of available levels and then cancel and then send me the .log)

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3034651/replicate_layout.log)

MitjaNemec, 2019-04-02:
I think I fixed replicate_layout. The issue was incorrect handling of sheet names with spaces in them. So instead of getting "Switch 1" I got "Switch". It was hiding in a totally inexpected place and I could have seen it earliers. Again thanks for the testing, and please test it again.

Also please test the archive_project as I fixed on python3 compatibility issue

T0jan, 2019-04-03:
> I think I fixed replicate_layout. The issue was incorrect handling of sheet names with spaces in them. So instead of getting "Switch 1" I got "Switch". It was hiding in a totally inexpected place and I could have seen it earliers. Again thanks for the testing, and please test it again.

yes I can confirm that the plugin works like intended, at least on this project. I will test it probably tomorrow on an other one and will come back to you if there are any further issues.

The old problem with whitespaces in names and software ^^

> Also please test the archive_project as I fixed on python3 compatibility issue

Runs now smoothly also the GUI is fixed, I have an additional feature request but I will put this to a dedicated issue.

So only the swap_units problem left, i think there is some land in sight

MitjaNemec, 2019-04-03:
Thanks for the feedback.

As for the swap_units, I am a bit lost in this thread so I'd really like if you describe what is an issue with it, preferably with as much words as you can and please attach the .log file and the netlist importer log if applicable.

T0jan, 2019-04-04:
Sorry the issue was with swap_units. it only works on the first level of a schematic. components in hierarchical sheets, at least in sheets which are used multiple times, are not touched. so of cause he does change the switch back, while reloading the netlist:

```
Reconnect IC5201 pin 7 from /Switch Circuit 6/1 to /Switch Circuit 6/2.
Reconnect IC5201 pin 1 from /Switch Circuit 6/2 to /Switch Circuit 6/1.
Reconnect IC5201 pin 2 from /Switch Circuit 6/2 to /Switch Circuit 6/1.
Reconnect IC5201 pin 3 from /Switch Circuit 6/2 to /Switch Circuit 6/1.
Reconnect IC5201 pin 5 from /Switch Circuit 6/1 to /Switch Circuit 6/2.
Reconnect IC5201 pin 6 from /Switch Circuit 6/1 to /Switch Circuit 6/2.
```

considering the log the plug claims, it changed something in the subsheet, but it doesn't:
[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3043081/swap_units.log)

may another bug or its the thing with labels directly on the pins we already discussed before: when trying to switching the pins of the resistor below (R302) the plugin claims, that it wont do it because it makes no sense to reconnect unconnected pins. it also generates no .log file for this

[image]

[image]

MitjaNemec, 2019-04-05:
swap_units should now work, please test it.

As for the swap pins in this case it can not really work. 
 It works 100% when the label is directly on the pin.
If the label is near the pin connected through short wire segment it works:
- if the label is the same as the net name
- there is no junctions, etc.

Thus I am considering removing the code supporting the swapping pins with label near them. Or I will keep the code in, but I will not mention the ability.

T0jan, 2019-04-12:
sorry had some other issues to deal with, but no i could find some time to test:

swap units now raises an exception while running and kills the schematic file (major project schematic as well as subsheets). it only saves an empty file.

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3073802/swap_units.log)

also the changes in delete_selected lead to a state, where Kicad does not recognize it anymore:

[Delete_selected_error.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3073805/Delete_selected_error.log)

MitjaNemec, 2019-04-12:
Oh man, sorry about the swap_units bug. This was a careless one and it points out that I have to put additional testing and automate it.

Both of them should be fixed.

T0jan, 2019-04-15:
> Both of them should be fixed.

delete_selected works again, swap_units does not raise an exception anymore, but:

it now kills the B-unit of the component (in subsheets as well as in the first level schematic):

[image]

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3079376/swap_units.log)

MitjaNemec, 2019-04-15:
First of all, thank you for your persistence. I really appreciate it.

Secondly, I really need to implement unit/integration tests for schematics, as this already worked, so the test should (if I had them) have caught this.

Anyhow, the bug has been fixed.

T0jan, 2019-04-15:
> First of all, thank you for your persistence. I really appreciate it.

you are welcome :)

> Anyhow, the bug has been fixed.

for first level schematics i can confirm this, on subsheets nevertheless the bug is still there.

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3080261/swap_units.log)

MitjaNemec, 2019-04-19:
OK, took me a while, as I've added test code, so I should catch regressions more easily. And I've fixed a bug, but do please test it as you have a real life project, which as the history shows can and did expose a number ob bugs.

T0jan, 2019-04-23:
Swap_units now again kills the B-unit in the major schematic as well as in subsheets.

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3107326/swap_units.log)

pad2pad_distance now raises an exception (and creates no log file):

[image]

place_footprints works only in `Place by Reference nr.`. in `Place by Sheet nr. it also moves the selected footprint and uses the component with the lowest reference designator as reference. It also does nothing anymore for higher sheet selection but only for the lowest level (in which the selected footprint is). (also no log file here)

the rest works completely fine.

MitjaNemec, 2019-04-24:
pad2pad should be fixed
swap_units should be fixed. Thanks for endurance.

as for place_footprints:
- it always takes lowest reference designator as a reference footprint. I know it is not really flexible. I've open an issue #45 regarding this, and I'll try to address it in the future.
- could you be a bit more verbose regarding higher and lowest level.
- there should be a log file

And starting from tomorrow I'll be gone until 7.5.2019

T0jan, 2019-05-16:
sorry to let you hang in the air, had to cover some time stealing test campaign.

> pad2pad should be fixed

I can confirm this, but the Highlight button does simply nothing and the UI scalling is a bit off:

[image]

> swap_units should be fixed. Thanks for endurance.

For hierarchical sheets it again kills the B unit, but for the first level schematic it works now:

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3186627/swap_units.log)

> could you be a bit more verbose regarding higher and lowest level.

I have multilevel hierarchical sheets like:

main_schematic
|-- switch_control_circuit 1
    |-- Switch 1
    |-- Switch 2
|-- switch_control_circuit 2
...

when I select a component from Switch 1, it only places the components of all selected sheets, when I select them via the UI directly. if I instead select the level above `switch_control_circuit` it does nothing.

[place_footprints.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3186672/place_footprints.log)

MitjaNemec, 2019-05-16:
I don't mind the delay. 

The highlight buton was just a test with no functionality behind it, but I've managed to solve the issue, so it should be implemented in a while. Thanks for the screenshot, I'll work on the layout. Thought there might be an iteration or two if this is a issue cause by differences between platforms.

Regarding swap_unit, thanks for the log. I'll have to look into it, but I now have some test project, and this works there. It will take a week or so.

Regarding place_footprints, also thanks for the log and description, I'll look into it

MitjaNemec, 2019-05-17:
pad2pad_disantce: OK, I've solved the highlighting issue and GUI might look somewhat better

swap_units: I've added a bit of instumentation. Can you please:
- store the original .sch files (pit-switch.sch)
- run the plugin again
- look into the log file search for line: `Unit # description is around lines: #`
- copy the full component description from original and modified file(which should be around the lines mentioned in .log file) here.
- attach the .log file here

you can also try running the example which is included with the plugin and you can try swapping U7 units on pin 7 and 14 (B and D) and report back what do you see and what would you expect to see

T0jan, 2019-05-17:
> pad2pad_disantce: OK, I've solved the highlighting issue and GUI might look somewhat better
I can confirm this, the GUI scaling is better but not completely fine:

> look into the log file search for line: Unit # description is around lines: #
with the latest version you pushed to master the log does not contain this info:

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3192238/swap_units.log)

may tell me, what i have to look for in the schematic files, I have some experience to dig through them ^^

> you can also try running the example which is included with the plugin and you can try swapping U7 units on pin 7 and 14 (B and D) and report back what do you see and what would you expect to see

on the test project the switching works like intended.

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3192271/swap_units.log)

MitjaNemec, 2019-05-20:
Sorry, I've forgot tu push the latest changes, in  version 19 you should have the additional info available

T0jan, 2019-05-20:
> copy the full component description from original and modified file(which should be around the lines mentioned in .log file) here.

original file:

```
$Comp
L bst_opamp:LM158 IC201
U 2 1 5CE5C513
P 3200 5650
AR Path="/5CAC71BF/5CE5C513" Ref="IC201"  Part="2" 
AR Path="/5CB3E342/5CE5C513" Ref="IC1201"  Part="2" 
AR Path="/5CB582E3/5CE5C513" Ref="IC2201"  Part="2" 
AR Path="/5CA3C508/5CE5C513" Ref="IC3201"  Part="2" 
AR Path="/5CA43896/5CE5C513" Ref="IC4201"  Part="2" 
AR Path="/5CA4FBFD/5CE5C513" Ref="IC5201"  Part="2" 
F 0 "IC201" H 3300 6050 45  0000 C CNN
F 1 "LM158" H 3300 5950 45  0000 C CNN
F 2 "SO8" H 3230 5800 20  0001 C CNN
F 3 "" H 3100 5850 60  0001 C CNN
F 4 "LM158DT" H 3200 5950 60  0001 C CNN "MPN"
F 5 "497-6990-1-ND" H 3300 6050 60  0001 C CNN "OC_Digi-Key"
	2    3200 5650
	1    0    0    -1  
$EndComp
```

modified file:

```
$Comp
L bst_opamp:LM158 IC201
U 1 1 5CE5C513
P 3200 5650
AR Path="/5CAC71BF/5CE5C513" Ref="IC201"  Part="1" 
AR Path="/5CB3E342/5CE5C513" Ref="IC1201"  Part="1" 
AR Path="/5CB582E3/5CE5C513" Ref="IC2201"  Part="1" 
AR Path="/5CA3C508/5CE5C513" Ref="IC3201"  Part="1" 
AR Path="/5CA43896/5CE5C513" Ref="IC4201"  Part="1" 
AR Path="/5CA4FBFD/5CE5C513" Ref="IC5201"  Part="1" 
F 0 "IC201" H 3300 6050 45  0000 C CNN
F 1 "LM158" H 3300 5950 45  0000 C CNN
F 2 "SO8" H 3230 5800 20  0001 C CNN
F 3 "" H 3100 5850 60  0001 C CNN
F 4 "LM158DT" H 3200 5950 60  0001 C CNN "MPN"
F 5 "497-6990-1-ND" H 3300 6050 60  0001 C CNN "OC_Digi-Key"
	2    3200 5650
	1    0    0    -1  
$EndComp
```

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3198875/swap_units.log)

also a small thing I discovered right now: if you run several plugins in one session it will store the logs of all of them in the log file of the first plugin you run, not in the corresponding log files of each plugin. That's probably also why I claimed several times before, that there is no log file at all while testing several plugins right after each other.

T0jan, 2019-05-20:
> copy the full component description from original and modified file(which should be around the lines mentioned in .log file) here.

original file:

```
$Comp
L some_lib:LM158 IC201
U 2 1 5CE5C513
P 3200 5650
AR Path="/5CAC71BF/5CE5C513" Ref="IC201"  Part="2" 
AR Path="/5CB3E342/5CE5C513" Ref="IC1201"  Part="2" 
AR Path="/5CB582E3/5CE5C513" Ref="IC2201"  Part="2" 
AR Path="/5CA3C508/5CE5C513" Ref="IC3201"  Part="2" 
AR Path="/5CA43896/5CE5C513" Ref="IC4201"  Part="2" 
AR Path="/5CA4FBFD/5CE5C513" Ref="IC5201"  Part="2" 
F 0 "IC201" H 3300 6050 45  0000 C CNN
F 1 "LM158" H 3300 5950 45  0000 C CNN
F 2 "SO8" H 3230 5800 20  0001 C CNN
F 3 "" H 3100 5850 60  0001 C CNN
F 4 "LM158DT" H 3200 5950 60  0001 C CNN "MPN"
F 5 "497-6990-1-ND" H 3300 6050 60  0001 C CNN "OC_Digi-Key"
	2    3200 5650
	1    0    0    -1  
$EndComp
```

modified file:

```
$Comp
L some_lib:LM158 IC201
U 1 1 5CE5C513
P 3200 5650
AR Path="/5CAC71BF/5CE5C513" Ref="IC201"  Part="1" 
AR Path="/5CB3E342/5CE5C513" Ref="IC1201"  Part="1" 
AR Path="/5CB582E3/5CE5C513" Ref="IC2201"  Part="1" 
AR Path="/5CA3C508/5CE5C513" Ref="IC3201"  Part="1" 
AR Path="/5CA43896/5CE5C513" Ref="IC4201"  Part="1" 
AR Path="/5CA4FBFD/5CE5C513" Ref="IC5201"  Part="1" 
F 0 "IC201" H 3300 6050 45  0000 C CNN
F 1 "LM158" H 3300 5950 45  0000 C CNN
F 2 "SO8" H 3230 5800 20  0001 C CNN
F 3 "" H 3100 5850 60  0001 C CNN
F 4 "LM158DT" H 3200 5950 60  0001 C CNN "MPN"
F 5 "497-6990-1-ND" H 3300 6050 60  0001 C CNN "OC_Digi-Key"
	2    3200 5650
	1    0    0    -1  
$EndComp
```

[swap_units.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3198875/swap_units.log)

also a small thing I discovered right now: if you run several plugins in one session it will store the logs of all of them in the log file of the first plugin you run, not in the corresponding log files of each plugin. That's probably also why I claimed several times before, that there is no log file at all while testing several plugins right after each other.

MitjaNemec, 2019-05-20:
Thanks for the file snippets, but could you also attach the snippets for the other unit (the one going from 1->2)

MitjaNemec, 2019-05-21:
Additionally, I've fixed a bug in the logging code, so the logs should be generated as user would expect. If you can test this also on your end (linux and python3) I'd appreciate it.

T0jan, 2019-05-23:
> Thanks for the file snippets, but could you also attach the snippets for the other unit (the one going from 1->2)

oh sorry, of cause (actually the log claims the wrong lines for unit 2):

original file:

```
$Comp
L some_lib:LM158 IC4201
U 1 1 5CDE9A00
P 2200 5650
AR Path="/5CA43896/5CDE9A00" Ref="IC4201"  Part="1" 
AR Path="/5CAC71BF/5CDE9A00" Ref="IC201"  Part="1" 
AR Path="/5CB3E342/5CDE9A00" Ref="IC1201"  Part="1" 
AR Path="/5CB582E3/5CDE9A00" Ref="IC2201"  Part="1" 
AR Path="/5CA3C508/5CDE9A00" Ref="IC3201"  Part="1" 
AR Path="/5CA4FBFD/5CDE9A00" Ref="IC5201"  Part="1" 
F 0 "IC201" H 2300 6050 45  0000 C CNN
F 1 "LM158" H 2300 5950 45  0000 C CNN
F 2 "SO8" H 2230 5800 20  0001 C CNN
F 3 "" H 2100 5850 60  0001 C CNN
F 4 "LM158DT" H 2200 5950 60  0001 C CNN "MPN"
F 5 "497-6990-1-ND" H 2300 6050 60  0001 C CNN "OC_Digi-Key"
	1    2200 5650
	1    0    0    -1  
$EndComp
```

modified file:

```
$Comp
L some_lib:LM158 IC4201
U 3 1 5CDEA626
P 4000 5650
AR Path="/5CA43896/5CDEA626" Ref="IC4201"  Part="3" 
AR Path="/5CAC71BF/5CDEA626" Ref="IC201"  Part="3" 
AR Path="/5CB3E342/5CDEA626" Ref="IC1201"  Part="3" 
AR Path="/5CB582E3/5CDEA626" Ref="IC2201"  Part="3" 
AR Path="/5CA3C508/5CDEA626" Ref="IC3201"  Part="3" 
AR Path="/5CA4FBFD/5CDEA626" Ref="IC5201"  Part="3" 
F 0 "IC201" H 4150 5700 45  0000 L CNN
F 1 "LM158" H 4150 5650 45  0000 L CNN
F 2 "SO8" H 4030 5800 20  0001 C CNN
F 3 "" H 3900 5850 60  0001 C CNN
F 4 "LM158DT" H 4000 5950 60  0001 C CNN "MPN"
F 5 "497-6990-1-ND" H 4100 6050 60  0001 C CNN "OC_Digi-Key"
	3    4000 5650
	1    0    0    -1  
$EndComp
```

> Additionally, I've fixed a bug in the logging code, so the logs should be generated as user would expect. If you can test this also on your end (linux and python3) I'd appreciate it.

* delete_selected: did not generated any log at all
* all other plugins write their results into the log of the first plugin which created a log and also overwriting the information of the previous ran plugin

MitjaNemec, 2019-05-23:
Are you sure that the snippets are from before and after, as there is no difference between them.
Also could you paste a section where log says for unit 2

Regarding the logging, I should have fixed this by now and it should work also under python3. Delete selected does not generate the log, and I'll probably delete this plugin as the same functionality can be achieved with selection filtering, that is natively present in KiCad.

T0jan, 2019-05-23:
> Are you sure that the snippets are from before and after, as there is no difference between them.

Yeah I checked it again. the difference is in the unit I posted a few days ago (https://github.com/MitjaNemec/Kicad_action_plugins/issues/36#issuecomment-494064435)

> Also could you paste a section where log says for unit 2

the log points for unit 2 to the first component of the schematic independent from what a component this is.

> Regarding the logging, I should have fixed this by now and it should work also under python3

Jep works now for all plugins.

MitjaNemec, 2019-05-24:
> the log points for unit 2 to the first component of the schematic independent from what a component this is.
Aaaaaa. So just to confirm, for example if you want to swap units 2 and 3 in IC4201, the log points to the unit 3 of the IC201 and IC201 is the first one in the .sch file? Obviously your design is more complex than my test cases, and I'll need to update my test suite. Is there a way you could share your project with me? You don't need to attach it here, you can look me up on google and you should find my email and send it over email. If you can't, I understand, but it will likely take a bit more time.

BTW, I'll be offline from tomorrow till 10th of June

T0jan, 2019-05-24:
> Aaaaaa. So just to confirm, for example if you want to swap units 2 and 3 in IC4201, the log points to the unit 3 of the IC201 and IC201 is the first one in the .sch file? 

no I wanted to switch unit 1 and 2 of IC4201. IC4201 is the same as IC201 but in a different sheet. the log points for the second unit every time to the first symbol in the schematic, in my case this is a resistor. the other two units I presented you above I searched for myself in the schematic.

> Is there a way you could share your project with me?

I will setup a demo-project which covers this issue. May take a while.

MitjaNemec, 2019-05-24:
> in my case this is a resistor.
Can you paste in a description block from the .sch file for this resistor.

T0jan, 2019-05-24:
sorry it is a capacitor but here it is:

> $Comp
> L some_lib:100n@50V[0603] C?
> U 1 1 5C8ABC3B
> P 2850 2300
> AR Path="/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C173F2E/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C249527/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C301C11/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C310E33/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C320055/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C37F005/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C3906DA/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C39066B/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C390607/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C32F277/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5BFE0C05/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5BFE0C17/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5BFE0C29/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C8AB945/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C903BD2/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C925F8C/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C938119/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C941E60/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C9466A8/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C9898F2/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C94EE49/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C965B42/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5C965B98/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5CB5FA73/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5CB5FA92/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5CB5FAB1/5C8ABC3B" Ref="C?"  Part="2" 
> AR Path="/5CAC71BF/5C8ABC3B" Ref="C201"  Part="1" 
> AR Path="/5CB3E342/5C8ABC3B" Ref="C1201"  Part="1" 
> AR Path="/5CB582E3/5C8ABC3B" Ref="C2201"  Part="1" 
> AR Path="/5CA3C508/5C8ABC3B" Ref="C3201"  Part="1" 
> AR Path="/5CA43896/5C8ABC3B" Ref="C4201"  Part="1" 
> AR Path="/5CA4FBFD/5C8ABC3B" Ref="C5201"  Part="1" 
> F 0 "C201" H 2550 2350 50  0000 L CNN
> F 1 "100n@50V[0603]" V 3000 1850 50  0000 L CNN
> F 2 "some_lib:C0603" H 2880 2450 50  0001 C CNN
> F 3 "" H 3060 2615 50  0001 C CNN
> F 4 "C0603C104K5RACTU" H 2110 2715 50  0001 C CNN "MPN"
> 	1    2850 2300
> 	1    0    0    -1  
> $EndComp

MitjaNemec, 2019-05-24:
Thanks. It looks like I am just looking for the reference number (4201 from the C4201). I'll have to confirm it

T0jan, 2019-05-24:
here the promised test project, once before and once after using the switch_units plugin:

[test_before_switch.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/3216450/test_before_switch.zip)
[test_after_switch.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/3216451/test_after_switch.zip)

MitjaNemec, 2019-05-24:
Thanks. I'll attack this when I come back

MitjaNemec, 2019-06-14:
Ok I am back on it and for the life of me, I can not recreate it with the test project. Can you please rerun on the test project and attach the .log file?

T0jan, 2019-06-14:
the log is in test_after_switch.zip

[25 more replies]

closed 2019-10-06 by MitjaNemec