MitjaNemec/Kicad_action_plugins #41: request: add archiving of footprints to archive_project

T0jan, 2019-04-03:
It would be nice if archive_project could call the pcbNew internal function to archive the footprints on its own to provide a full snapshot of the used libraries.

PureBasic, 2019-04-03:
This is a must have.

MitjaNemec, 2019-04-03:
@PureBasic I don't know if you are aware, but currently pcbnew has an option to archive footprints.

@T0jan I agree it would be nice, but currently the underlying function is not exposed to python API. I've looked into wx and maybe I could trigger the menu pragmatically leaving the user to select the folder location and name and click OK.

If I'll ask for this function to be exposed it will be some time before this is implemented by KiCad developers. And even then it will most likely go into V6 branch only.

Currently my aim is to support V5.1.

iveyj1, 2019-04-15:
It seems like the footprint archive function doesn't fix up the schematic symbol footprint fields to point to the new archive.  This seems less useful than it might be.  Am I missing something?

MitjaNemec, 2019-04-15:
I really cannot comment on intended functionality of footprint archive function. That said, I can imagine that the developers did not really feel the need to do this, as the layout is quite portable even without the archived footprint library. This functionality only enables to extract footprints from the design, so that you can place them into your personal library.

As for my plugin, my main intention was to make KiCad projects portable, and I believe that I achieved that. Yes archiving footprints and remapping the schematics and would be a step forward, but I currently plan to support KiCad 5.1 for quite some time. 

Such an expansion would require additional SWIG interface which can be introduced only in V6 nightlies and they will be somewhat less stable that the late V5 nightlies were before V5 release. So even if SWIG interface for this gets implemented, I'll start to work on this when V6 nightlies become more stable.

iveyj1, 2019-04-16:
I understand, thanks for the good work; I plan to use the symbol archiving functionality.

iveyj1, 2019-04-17:
Nice!