MitjaNemec/Kicad_action_plugins #42: Not copyng module (improve error message)

hildogjr, 2019-04-10:
I am using the last KiCad NIgthly on Ubuntu 16 and constantly getting the message: "More or less than 1 module selected. Please select exactly one module and run the script again.".

[image]

As far I know, I selected exactly the necessary one instance of the module. So, I don't known if it is a version compatibility error.

Could this message be improved to give me some tip about (each components are missing or were selected in excess).

MitjaNemec, 2019-04-10:
> ![Screenshot from 2019-04-10
> 09-55-38](https://user-images.githubusercontent.com/2211474/55880416-0551e600-5b77-11e9-8c3e-acf04f39ce25.png)

From the screenshot I see that you selected all the modules in one sheet. You have to select only one module in the sheet.

The confusion comes from the KiCad nomenclatrue where module means footprint. So you have to select only one footprint in the sheet. Keep in mind that you have to place the corresponding footprints in other sheet to desired final position.

I'll have to adjust the UI language as it seems that KiCad asimilated the common industry terms.

hildogjr, 2019-04-10:
I pick just one element/part/footprint, I am getting:

[image]

MitjaNemec, 2019-04-10:
> 11-06-12](https://user-images.githubusercontent.com/2211474/55885638-b0b36880-5b80-11e9-8e2e-cfa3b75dcd45.png)

You opened a .kicad_pcb file which does not have a corresponding .sch file. If you open multiple_hierarchy.kicad_pcb it should work.

I'll have to handle this error in a sane way though

hildogjr, 2019-04-10:
In my personal project appear to be some string space-split. Check the image:

[image]

MitjaNemec, 2019-04-10:
Can you send me the `.log` file which should be in the project folder. I'll try to fix this today or tomorrow.

hildogjr, 2019-04-10:
`replicate_plugin.log` is not created at the project folder.

hildogjr, 2019-04-10:
I found the source of the issue: adding `logger.info('TEST1 -> {}'.format(filename)) #TODO` I have a log:

    04-10 12:27:31 replicate_layout.action_replicate_layout 122:Replicate layout plugin version: 11 started
    04-10 12:27:31 replicate_layout.action_replicate_layout 153:Preparing replicator with CNN201 as a reference
    04-10 12:27:31 replicate_layout.replicatelayout 190:getting project hierarchy from schematics
    04-10 12:27:31 replicate_layout.replicatelayout 191:TEST1 -> /home/h/Documents/UNICAMP/LabREI/Projeto painel/Boards/Analog conditioning.sch
    04-10 12:27:31 replicate_layout.replicatelayout 98:TEST1 -> /home/h/Documents/UNICAMP/LabREI/Projeto painel/Boards/Analog conditioning.sch
    04-10 12:27:31 replicate_layout.replicatelayout 98:TEST1 -> /home/h/Documents/UNICAMP/LabREI/Projeto painel/Boards/Analog

hildogjr, 2019-04-10:
In this function the "F1" line search return:

    ['$Sheet', 'S 3300 900  1350 900 ', 'U 5C9F82F2', 'F0 "Antialiasing & signal conditioning 0" 50', 'F1 "Analog conditioning - antialiasing.sch" 50', 'F2 "outA+" O R 4650 1000 50 ', 'F3 "outA-" O R 4650 1100 50 ', 'F4 "outB+" O R 4650 1200 50 ', 'F5 "outB-" O R 4650 1300 50 ', 'F6 "outC+" O R 4650 1400 50 ', 'F7 "outC-" O R 4650 1500 50 ', 'F8 "outD+" O R 4650 1600 50 ', 'F9 "outD-" O R 4650 1700 50 ', '']

An you break it at the space `subsheet_path = line.split()[1].rstrip("\"").lstrip("\"")`.

I replace that line by `subsheet_path = re.findall("\s\"(.*.sch)\"\s", line)[0]` and now works good.

closed 2019-04-10 by MitjaNemec

MitjaNemec, 2019-04-10:
Thanks for spotting, reporting and fixing the bug. I should have fixed it a week ago when I dealt with the handling of space in schematics name (5-6 lines higher)

commit 9d58bc68 (2019-04-10): Merge pull request #43 from hildogjr/master

Fix #42 - allow space in the file sheet name
commit aec55a79 (2019-04-10): Fix #42 - allow space in the file sheet name