MitjaNemec/Kicad_action_plugins #49: Remove duplicated track

hildogjr, 2019-05-16:
Sometimes we duplicate a layout and forgot to check the "remove existing tracks/zones/drawings", making place of duplicated symbols / silks / tracks in each instance of the layout.

Should be interesting check for this occurrences and delete them, even in the source layout.

MitjaNemec, 2019-05-16:
While I understand your frustration, I will certainly not delete any elements of the PCB layout without users consent. I can probably check the destination area for existing elements and notify the user. Now that I've managed to use the higlight functionality I could even highlight which items the plugin found.

As a workaround, does is currently not work if you re-run the plugin and check the "remove existing tracks/zones/drawings"?

hildogjr, 2019-05-16:
Thanks @MitjaNemec, I am using the work around. The issue is that:
1) I am sharing a project...
2) The other designer replicated one part several times so sync them;
3) Now each individual part could have multiple tracks with same begging-end point / width;
4) I would like to check this occurrences  and delete them even in the layout choose as source for the replicate algorithm.

MitjaNemec, 2019-05-16:
Whoa, this is a hard to solve with current tools if you don't have at least one sheet where there are no duplicates. Now if you do, select this one as a source and rerun the plugin.

Obviously I'll have to add the check if there is already an existing item in the same place, and if it is, skip the replication for this item only. But this will slow down the plugin, and it is not a trivial addition. As it will take some time before I attempt this you will have to solve your issues on your own. Perhaps you might get around with a tool that finds line/paragraph duplicates in text files?

hildogjr, 2019-05-16:
Maybe add a get additional ckeckbox "avoid duplicated elements", even it will delete nothing in the parent layout, this will allow to remove them making a reverse sync (one children - > parent).

This tool for duplicated could be a interesting propose for the developer in the new Pcbnew DRC...

MitjaNemec, 2019-05-16:
This is a nice idea. Thanks

Regarding the DRC, I wouldn't even dare mention this idea. It is not the job of the KiCad developers to handle cases which are caused by external tools. If the kicad internal router does place tracks over the existing ones exactly (when the "remove redundant tracks" is disabled) then this might be one and only case.

Though this is an idea for additional action plugin "delete_duplicates"

MitjaNemec, 2019-10-20:
Implemented within the replicate_layout plugin, please test when you find the time

hildogjr, 2019-10-20:
Work fine. Tested in Pcbnew Version 5.99.0-unknown-c3175b4~86~ubuntu16.04.1

closed 2019-10-21 by MitjaNemec