MitjaNemec/Kicad_action_plugins #50: Replicate plugin not replicate filled zone corner smoothing

hildogjr, 2019-05-16:
Replicate plugin not replicate filled zone corner smoothing.

1) Set corner smoothing to fillet in the source layout
2) Replicate
3) The children will be configured as None

MitjaNemec, 2019-05-16:
Thanks for the report. It will take me a while to get on it especially since I've used the code from Miles Mccoo for cloning zones.

Furthermore I am sceptical if I can even solve it as I think that corner smoothing is a new feature in KiCad. Do you know when it was introduced? I am afraid that there is no python API to do this. But I'll have to dig in.

hildogjr, 2019-05-16:
I remember of this feature start 2 month ago in Nightly release, before the 5.1 release. But I don't known if is present in last stable.

If I found something that can guide the implementation (the API point / function), I will post here.

MitjaNemec, 2019-05-16:
How do you do corner smoothing in pcbnew?

hildogjr, 2019-05-16:
It is in the settings of the filled copper zone
[image]

The source is

    (zone (net 43) (net_name "/Phase measurement 0/Voltage measure 1/GND_iso") (layer "B.Cu") (tstamp 5CDD2BC9) (hatch edge 0.508)
        (priority 5)
        (connect_pads (clearance 0))
        (min_thickness 0.254)
        (fill yes (arc_segments 32) (thermal_gap 0.508) (thermal_bridge_width 0.508) (smoothing fillet) (radius 1))
        (polygon
          (pts
            (xy 158.192 146.5915) (xy 158.192 153.0915) (xy 174.867001 153.0915) (xy 174.867001 146.5915)
          )
        )
        (filled_polygon
          (pts
            (xy 174.03719 146.735876) (xy 174.200851 146.785522) (xy 174.351673 146.866138) (xy 174.483871 146.97463) (xy 174.592363 147.106828)
            ...
          )
        )
      )

The `(smoothing fillet) (radius 1)` directive was not copied to children.

MitjaNemec, 2019-05-16:
Ok, this is a V6 feature, so this will be quite low in my list of priorities

hildogjr, 2019-05-16:
Agree.

MitjaNemec, 2019-07-29:
First of all, I was wrong saying that zone corner smoothing (fillet or chamfer) is not part of 5.1.x, as it certainly is present in 5.1.2 (don't know when exactly it was introduced though).

Secondly, I've refactored zone replication and it should replicate corner smoothing also. Please test it.

hildogjr, 2019-07-29:
I am getting the follow error, even using the examples in the package (but I am using the last Nightly).

File "/home/h/.kicad_plugins/replicate_layout/action_replicate_layout.py", line 240, in Run
    clear_highlight_on_module(mod)
File "/home/h/.kicad_plugins/replicate_layout/action_replicate_layout.py", line 55, in
    clear_highlight_on_module
    drawings = module.GraphicalItemsList()
File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 18916, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, MODULE, name)
File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 74, in _swig_getattr
    return _swig_getattr_nondynamic(self, class_type, name, 0)
File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 69, in _swig_getattr_nondynamic
    return object.__getattr__(self, name)
AttributeError: type object 'object' has no attribute '__getattr__'

MitjaNemec, 2019-07-29:
Yeah, it seems there were a couple of python API changes done in V6 nightlies (see #55) where some of the methods were removed. I'll have to get my hands on V6 doxygen documentation as the one available at https://kicad-downloads.s3.cern.ch/doxygen-python/index.html is obviously for V5.

It lists both methods as available:
https://kicad-downloads.s3.cern.ch/doxygen-python/classpcbnew_1_1MODULE.html#a131f43c3d91e6812413293b48b7e2c94

https://kicad-downloads.s3.cern.ch/doxygen-python/classpcbnew_1_1MODULE.html#a7204a8afe5ecbc2d7d5501f96c906cc0

I seem to recall from the developers mailing list that only methods which return DLIST_ITERATOR_WRAPPER are preferred, but I can not find the discussion.

Anyhow the issue regarding `GraphicalItemsList` should be fixed, but I suspect that I have similar issues still lurking, so I'd appreciate further testing.

qu1ck, 2019-07-29:
DLIST was removed completely you have to use non-List() methods now.

MitjaNemec, 2019-07-30:
Thanks for the info. Now I need to get my hands on V6 documentation to find out all the removed methods

qu1ck, 2019-07-30:
You can generate it yourself but it's easier to just look at stubs in pcbnew.py. It's huge and can be hard to search but it has same comments as in doxygen docs.

MitjaNemec, 2019-07-30:
Thanks, just did the comparison of 5.1.2 and V6nightly pcbnew.py and it seems that only PadsList and GraphicalItemsList are gone. I'll sleep a little bit better now.

hildogjr, 2019-07-30:
Almost worked (it made all the position but got a error to replicate the tracks) with the 16ac87ae1861b527a6c8b4e8d2e177dabfd07353 (30-July) commit.
Now I am getting:

Traceback (most recent call last):
  File "/home/h/.kicad_plugins/replicate_layout/action_replicate_layout.py", line 277, in Run
    drawings=rep_drawings)
  File "/home/h/.kicad_plugins/replicate_layout/replicatelayout.py", line 964, in replicate_layout
    self.replicate_tracks()
  File "/home/h/.kicad_plugins/replicate_layout/replicatelayout.py", line 720, in replicate_tracks
    newvia.SetViaType(track.GetViaType())
  File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 11580, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, TRACK, name)
  File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 74, in _swig_getattr
    return _swig_getattr_nondynamic(self, class_type, name, 0)
  File "/usr/lib/python2.7/dist-packages/pcbnew.py", line 69, in _swig_getattr_nondynamic
    return object.__getattr__(self, name)
AttributeError: type object 'object' has no attribute '__getattr__'

MitjaNemec, 2019-07-30:
Whoa, looks like it either does not find `SetViaType` or `GetViaType` methods. But it should as this code is in the if clause which checks if the track s VIA. And if the track is VIA it should have these two methods. I'd appreciate if you could paste me the contents of `/usr/lib/python2.7/dist-packages/pcbnew.py` around line number 11580.

Also, I've refactored tracks replication and if you could test it again that would be great

hildogjr, 2019-07-31:
Appear all fine now.

The content of `pcbnew.py`.

    _pcbnew.MIN_VIA_DRAW_SIZE_swigconstant(_pcbnew)
    MIN_VIA_DRAW_SIZE = _pcbnew.MIN_VIA_DRAW_SIZE
    class TRACK(BOARD_CONNECTED_ITEM):
        """Proxy of C++ TRACK class."""
    
        __swig_setmethods__ = {}
        for _s in [BOARD_CONNECTED_ITEM]:
            __swig_setmethods__.update(getattr(_s, '__swig_setmethods__', {}))
        __setattr__ = lambda self, name, value: _swig_setattr(self, TRACK, name, value)
        __swig_getmethods__ = {}
        for _s in [BOARD_CONNECTED_ITEM]:
            __swig_getmethods__.update(getattr(_s, '__swig_getmethods__', {}))
        __getattr__ = lambda self, name: _swig_getattr(self, TRACK, name)
        __repr__ = _swig_repr
    
        def ClassOf(aItem):
            """ClassOf(EDA_ITEM aItem) -> bool"""
            return _pcbnew.TRACK_ClassOf(aItem)
    
        if _newclass:
            ClassOf = staticmethod(ClassOf)
        __swig_getmethods__["ClassOf"] = lambda x: ClassOf
    
        def __init__(self, aParent, idtype=PCB_TRACE_T):
            """
            __init__(TRACK self, BOARD_ITEM aParent, KICAD_T idtype=PCB_TRACE_T) -> TRACK
            __init__(TRACK self, BOARD_ITEM aParent) -> TRACK
            """
            this = _pcbnew.new_TRACK(aParent, idtype)
            try:
                self.this.append(this)
            except Exception:
                self.this = this

[pcbnew.py.txt](https://github.com/MitjaNemec/Kicad_action_plugins/files/3453484/pcbnew.py.txt)

closed 2019-08-19 by MitjaNemec