MitjaNemec/Kicad_action_plugins #51: replicate_layout.log is missing

pgregson1, 2019-05-20:
Hello,

I upgraded my computer from an older Mac Air running OS X 10.12.6 Sierra to MacBook Air running OS X 10.14.4 Mojave with 16 Gbytes of RAM.  

The file "replicate_layout.log" for my current project is missing from my current computer.  Ones from previous projects exist on my old computer, but none with dates more recent than March 9.  

To determine where the problem was, I copied the KiCad project that produced a "replicate_layout.log" file on my previous computer to the new one.  I then used the new machine to see if the "replicate_layout.log" would be produced when I ran "Place_footprints" and then "replicate_layout" again on a layout that was know to produce that file on the older machine.  The log file was not created on the new machine.

So, things that have changed for that test case:

1.  A new computer, but merely an upgraded version of the old one,
2. A new version of the operating system (paths may have changed),
3. A new version of KiCad (the old version is (5.0.2-5)-5, release build; the new one is 5.1.0-0, release build),
4  A new install of Action Plugins "Place_footprints" and "Replicate_layout".

Things that have not changed:

1.  the test file,
2.  the procedure followed.

I need to get the "replicate_layout.log" because I am trying to track down a few other issues.  Any help would be greatly appreciated.

Best regards,
Peter

MitjaNemec, 2019-05-20:
This might be similar as #52.

Also on windows, you have to close the KiCad completely in order for log to get written.

Any it might be in different location (did you open different project and run a plugin, then opened different project and rerun the plugin. The log might be in the first project location.

Can you check this out and confirm it.

pgregson1, 2019-05-20:
In answer to your questions,

"This might be similar as #52."  It isn't.  No replicate_layout.log file is written at all, neither with Place_footprints nor with Replicate_layout, and certainly not when both are used.

"Also on windows, you have to close the KiCad completely...".  I'm using a Mac, not Windows.  Exiting KiCad and re-opening it did  not result in the writing of a replicate_layout.log file.

"Any it might be in different location (did...".  No I didn't do that.   However, I just ran my test case.  For it, I ran Place_footprints first, and then Replicate_layout.  I found the replicate_layout information at the end of the place_footprints.log file.  That's a bug.

Best regards,

Peter

MitjaNemec, 2019-05-21:
OK there is at least one bug in the logging code. The bug has been fixed so the plugins should generate .log files as one would expect. But as I can only test it on Windows platform I'd appreciate it if you can test it on macOS.

pgregson1, 2019-05-21:
Will do.  I am in the middle of something so I won't be able to get to it until this evening.

MitjaNemec, 2019-05-21:
There is no hurry. Everybody has his own life with schedules, family, various commitments ... I can only be happy that you are contributing with testing. And if you would stop without notice, I can only say "it was fun while it lasted" but I'll be the last to judge you as I don have the slightest idea about you, your life and other circumstances. The same goes for me. I try to be prompt but sometimes it might take me a couple of weeks to respond, so please keep this in mind.

MitjaNemec, 2019-05-24:
The logging issue should be solved now (https://github.com/MitjaNemec/Kicad_action_plugins/issues/36#issuecomment-495282294)

closed 2019-06-18 by MitjaNemec

pgregson1, 2019-06-18:
Thanks Mitja.  Much appreciated.

Best regards,

Peter