MitjaNemec/Kicad_action_plugins #53: archive_project doesn't archive the page layout description file

Sembazuru, 2019-05-31:
As the title says, a configured page layout file isn't archived so does not become portable.

I'd like the archive_project to copy the defined page layout file for both schematic and pcb to the project folder and then redirect the path to the project directory.

The setting for the page layout for both EEschema and PCBNew is in the project .pro file, so that might make this difficult...

MitjaNemec, 2019-06-03:
Thanks for reporting it (I think it was mentioned before either on kicad-info forum or within some issue here on GitHub) so that it will not be forgotten.

As you said this is probably not doable as KiCad manager might rewrite any changes that are made from within the pcbnew. I'll try this out and see where it gets me. If you could supply a test project along with page layout description this would be even better.

MitjaNemec, 2019-09-04:
@Sembazuru, I've pushed an implementation of worksheet archivation in the archive_worksheet branch. It works on windows and if you can test it on any other OS I'd appreciate it.

Sembazuru, 2019-09-08:
Unfortunately for testing, I only have access to Windows as well.

MitjaNemec, 2019-10-20:
Merged to master

closed 2019-10-20 by MitjaNemec