MitjaNemec/Kicad_action_plugins #54: Replicated Layout don't create any tracks

tulskij-pryanic, 2019-06-17:
Hello. I tried to use plugins both on MacOS and Win7. 
First I want to say that it start working when I put it in different folder than specified in README.md
On MacOs it was inside the packet ~Kicad/contents/sharedsupport/scripting/plugins not in ~Library. On windows C:\Program Files\KiCad\share\kicad\scripting\plugins
On MacOS it's ok. But on Win7 all components goes to the same places, but no any traces create. And I have a message fatal error when making an instance of replicator.

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3296146/replicate_layout.log)

MitjaNemec, 2019-06-17:
Hi, thanks for the report
Regarding the path on macOS, I suspect things are different for different macOS version, so if you'd be so kind and let me know which version you are using, I could put more detailed instructions.

As for the bug, I'll get on it, but I am currently short on time available it will take a while.

MitjaNemec, 2019-06-17:
Come to think of it, you did place the anchor footprints well appart?

If all of the anchor footprints are on the same position as the pivot footprint, this would explain why the footprints are place on top of each other. Obviously I should be able to detect this and give a proper error message instead of crashing.

tulskij-pryanic, 2019-06-17:
Hello, MacOS HiSierra)
No, no. The target footprints doesn't place on top to source position. All components places near the same footprint as i choose but there is no traces between pads. 
Log on the first message)

MitjaNemec, 2019-06-17:
Ok just to eliminate any misunderstandings the footprints are placed on correct locations?

MitjaNemec, 2019-06-17:
Ok I see

MitjaNemec, 2019-06-17:
Regarding the plugin folder, can you run the following code in python scripting console in pcbnew
```
import pcbnew
pcbnew.GetKicadConfigPath()
```
and paste me back the return path (you can filter the id part of the path)

tulskij-pryanic, 2019-06-17:
```
locals()

Type: <type 'dict'>

Value: {'pp': <bound method Display.setItem of <wx.py.crust.Display; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x2f6160e0> >>, 'shell': <wx.py.shell.Shell; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x2f615640> >, '__builtins__': <module '__builtin__' (built-in)>, '__package__': None, 'filling': <wx.py.filling.Filling; proxy of <Swig Object of type 'wxSplitterWindow *' at 0xed8fb10> >, 'notebook': <wx._controls.Notebook; proxy of <Swig Object of type 'wxNotebook *' at 0xbc68230> >, 'pcbnew': <module 'pcbnew' from 'C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.pyc'>, '__name__': '__main__', '__doc__': None}
```
I made a new simple schematic with hierarchy and route it. No problem. Then I delet second sheet in my "problem" project and add it again. After all ok - the traces routes too.
I don't know... Maby problem was in schematic file(

MitjaNemec, 2019-06-17:
I'll have to correct myself (obviously there are difference between OSes)
```
import pcbnew
print pcbnew.GetKicadConfigPath()
```
and if this does not work then try this:
```
import pcbnew
print (pcbnew.GetKicadConfigPath())
```
and paste me back the return path (you can filter the id part of the path)

MitjaNemec, 2019-06-17:
As for the bug experienced, it is certainly not schematics.

When replicating footprints, I also replicate footprint's text items. It seems that the footprint which I tried to replicate (I don't know which one) did not have the same number of text items as its pivot modules

MitjaNemec, 2019-06-18:
I've added some additional error checking for equality of the footprints under replication, so if you hit this again, there should be a bit more information in the log

tulskij-pryanic, 2019-06-21:
The same
locals()

```
Type: <type 'dict'>

Value: {'pp': <bound method Display.setItem of <wx.py.crust.Display; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x984e040> >>, 'shell': <wx.py.shell.Shell; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0xa227630> >, '__builtins__': <module '__builtin__' (built-in)>, '__package__': None, 'filling': <wx.py.filling.Filling; proxy of <Swig Object of type 'wxSplitterWindow *' at 0xa1c5860> >, 'notebook': <wx._controls.Notebook; proxy of <Swig Object of type 'wxNotebook *' at 0xa20d750> >, 'pcbnew': <module 'pcbnew' from 'C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.pyc'>, '__name__': '__main__', '__doc__': None}
```
[image]

MitjaNemec, 2019-06-21:
Aha, you have to enter line by line
[image]
I don't know why the difference though.

tulskij-pryanic, 2019-06-22:
[image]

MitjaNemec, 2019-06-26:
I was probably not clear enough first you have to enter
`import pcbnew` and make sure you hit enter (return) after entering.
and then you enter
`print pcbnew.GetKiCadConfigPath()`

closed 2019-07-29 by MitjaNemec