MitjaNemec/Kicad_action_plugins #55: Replicate layout do not work after kicad update

Joelguerreiro, 2019-07-02:
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3352225/replicate_layout.log)

MitjaNemec, 2019-07-03:
On which KiCad version are you running the plugin? Which python version do you use?

Currently only 5.1.x branch is supported by all my plugins.

Joelguerreiro, 2019-07-03:
Hello,

Sorry for not sending the details. I thought the log already contained it.
Kicad 6.0
Python 2.7.12
I think my system does not support your plugins then. Thank you anyways.

MitjaNemec, 2019-07-03:
Hi,

I can not get the KiCad information into the log as the pcbnew python API does not support this, But I could get the python version into it.

As for the plugin, yes it seems that the method `pcbnew.MODULE.PadsList()` is either not existent anymore or is just not exposed via SWIG. I don't have a clue why was it removed since as far as I know currently development has been focused elsewhere.

Anyhow I intend to support only the 5.1.x branch since the V6 branch is under heavy development and I'd rather wait for the rate of change to settle down. So I'll defer dealing with this until 6.0 is officially released.

Sorry, but you are on your own. You might get by with replacing `pcbnew.MODULE.PadsList()` with
`pcbnew.MODULE.Pads()`

MitjaNemec, 2019-07-29:
Hi @Joelguerreiro 

I've replaced all the instances of using `pcbnew.MODULE.PadsList()` to `pcbnew.MODULE.Pads()` and it works in V5.1.x. 

So if you don't mind testing in in V6 nightlies, that would be great.

closed 2019-08-19 by MitjaNemec