MitjaNemec/Kicad_action_plugins #56: Request: Archive datasheets

rfporter, 2019-07-05:
It would be quite useful if the project archiver would pull all of the datasheets from the web and place them in a datasheets or docs folder, then update the datasheet field to point to the new path.

This way, if you open the project in 10 years, you will still have access to the datasheets.

MitjaNemec, 2019-07-08:
This is a great idea. I'll have to resolve a couple of other issues first, but I plan to implement this.

T0jan, 2019-07-22:
should come definitely with a option to disable it (like for the other options) in case some has a different method to store the datasheets locally

MitjaNemec, 2019-07-25:
Thanks for the remark.

MitjaNemec, 2019-07-28:
Feature implemented, please test it.

There is an issue with wxpython (at least on windows) where documentation checkbox can be modified by the user even when the schematics checkbox is unchecked. But this is just a GUI issue.

T0jan, 2019-07-29:
at least for my side the integration into kicad fails:

[Archive_project_error.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3442159/Archive_project_error.log)

MitjaNemec, 2019-07-29:
An obvious python3 compatibility issue, thanks for spotting it, should be fixed now

T0jan, 2019-08-01:
integration into kicad works again, but it seems like it does not archive the datasheets. not sure if it is because i store my datasheets on my local server or if it is a general problem. at least the log file does not mentions a try of archiving any datasheet but this could also be a logging issue since the log only contains information about the 3D models.

MitjaNemec, 2019-08-01:
There was a bug in the logging code. It was fixed, so please run it again and attach .log file.

T0jan, 2019-08-01:
I can confirm the archive function works for weblinks but not for links to local server files but this is probably nothing you have to consider.

for the logging the bug seems to be still there:

[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3456971/archive_project.log)

MitjaNemec, 2019-08-02:
I'd also like to solve the local server files issue.

You did update the plugin? You can also get the first part of the .log file if you run the plugin and when the plugin finishes with schematics it displays a dialog saying "Schematics archived sucessfuly!". If you don't click on the OK, you should be able to open the .log file. If you do, please attach the .log file contents here. Thanks.

T0jan, 2019-08-02:
> I'd also like to solve the local server files issue.

you sure you want to do this? this needs a completely different approach with potential password prompt etc.

> If you do, please attach the .log file contents here. Thanks.

hmm seems the logging works now, may I forgot to reload the plugins/restart kicad

nevertheless log after schematics + pdfs: 
[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3460711/archive_project.log)

PS: it would be good either for testing but also for use in normal projects to have a undo button which resets the file links to the global paths and deletes all local libraries. should I open a new issue for this?

MitjaNemec, 2019-08-02:
> you sure you want to do this? this needs a completely different approach with potential password prompt etc.

Now that I've seen the log, not really. I though you have the files on a networked volume (NFS or something like that) and the plugin handled filepaths the wrong way. For the web access behind username/password account, I am open for merging PR if somebody does this, but I probably won't even try to implement this. Do open an issue though, so that it stays documented please.

And for the undo, also open a new issue for it, I see that it is better to have multiple issues opened each for separate topic than the one that we've had.

closed 2019-08-19 by MitjaNemec