MitjaNemec/Kicad_action_plugins #58: Can't install

GongSiqiu, 2019-08-01:
I copied  relevant folder into  ~/.kicad_plugins and refresh plugins. But these tools didn't show up in external plugins. I tried to copy kicad/demos/python_scripts_examples/action_plugin_test_undoredo.py there, this script works well. So I want to know some details about the installation.
Thank you!

Application: kicad
Version: 5.1.2+dfsg1-1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.0 OpenSSL/1.1.1c zlib/1.2.9 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3
Platform: Linux 4.19.0-5-amd64 x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.64.0
    Compiler: Clang 7.0.1 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

MitjaNemec, 2019-08-01:
You do have to restart KiCad. Just refreshing from within KiCad is not enough (at least on Windows).

And the exact folder setup is 
`~/.kicad_plugins/replicate_layout/` for replicate_layout plugin for example?

GongSiqiu, 2019-08-01:
I tried restart and  undoredo worked well while replicate_layout failed

~/.kicad_plugins]$ tree                                              
.
├── action_plugin_test_undoredo.py
├── __pycache__
│   └── action_plugin_test_undoredo.cpython-37.pyc
└── replicate_layout
    ├── action_replicate_layout.py
    ├── compare_boards.py
    ├── duplicate-replicate_layout.svg
    ├── duplicate-replicate_layout.svg.png
    ├── __init__.py
    ├── multiple_hierarchy
    │   ├── Leg.sch
    │   ├── multiple_hierarchy-cache.lib
    │   ├── multiple_hierarchy.kicad_pcb
    │   ├── multiple_hierarchy_outer.kicad_pcb
    │   ├── multiple_hierarchy_outer.sch
    │   ├── multiple_hierarchy_outer_test.kicad_pcb
    │   ├── multiple_hierarchy.pro
    │   ├── multiple_hierarchy.sch
    │   ├── multiple_hierarchy_test_alt.kicad_pcb
    │   ├── multiple_hierarchy_test.kicad_pcb
    │   ├── Power.sch
    │   └── Sensor.sch
    ├── no_wxpython.py
    ├── __pycache__
    │   ├── __init__.cpython-37.pyc
    │   └── no_wxpython.cpython-37.pyc
    ├── Replicate_layout_error.log
    ├── replicate_layout_GUI.fbp
    ├── replicate_layout_GUI.py
    ├── replicatelayout.py
    └── version.txt

MitjaNemec, 2019-08-02:
Thanks for persisting. I know it can be a bit frustrating as I am at computer only twice a day for less than an hour so the discussion tempo is quite slow.

Just to avoid any misunderstandings due to the markdown representation the path to `replicatelayout.py` is `~/.kicad_plugins/replicate_layout/replicatelayout.py`?
Also the init.py has two underscores in prepended and appended?
Please confirm this.

Can you delete `Replicate_layout_error.log` and launch KiCad again and if `Replicate_layout_error.log` gets created, please attach it here.

GongSiqiu, 2019-08-03:
It's very kind of you. Thanks for your patience.

replicatelayout.py is ~/.kicad_plugins/replicate_layout/replicatelayout.py
the init.py has two underscores in prepended and appended

Replicate_layout_error.log has only one line 
ModuleNotFoundError("No module named 'wx'")
[Replicate_layout_error.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3463252/Replicate_layout_error.log)

GongSiqiu, 2019-08-03:
Installed something about wx, I maybe solved this problem. Thank you very much!

closed 2019-08-03 by GongSiqiu