MitjaNemec/Kicad_action_plugins #61: Fatal error when executing place footprints in matrix

nxxsr, 2019-08-20:
08-20 08:09:23 place_footprints.action_place_footprints 687:Fatal error when executing place footprints
Traceback (most recent call last):
  File "C:\Users\xiaoxu.niu.WONDERLAN\AppData\Roaming\kicad\scripting\plugins/place_footprints/action_place_footprints.py", line 682, in Run
    placer.place_matrix(sorted_modules, step_x, step_y, nr_columns)
TypeError: place_matrix() takes exactly 6 arguments (5 given)

MitjaNemec, 2019-08-21:
Sorry about it, although I test the backend regularly, the UI is left for manual testing, and I forgot to test all cases. Should be fixed now

closed 2019-08-21 by MitjaNemec

nxxsr, 2019-08-23:
Thank you. Appreciate the plugin. Huge time saver!

commit 4d1518fc (2019-08-21): Bugfix for matrix layout

Fixes #61