MitjaNemec/Kicad_action_plugins #62: place footprints only select part of available footprints

nxxsr, 2019-09-04:
only part of the footprints are selected by reference. None of the parts are selected by Hierarchy level.

[image]

[image]

MitjaNemec, 2019-09-04:
First, thanks for reporting. I really appreciate thorough testing that only real life cases can provide. Now if you can, then please send me the `place_footprints.log` file so I can start locating the bug. I might ask for some additional info, but the .log file should be a good start

MitjaNemec, 2019-09-09:
@nxxsr can you attach the log file please?

MitjaNemec, 2019-09-09:
Select by reference should be working now, for select by hierarchy sheet I'll need the .log file

closed 2019-10-04 by MitjaNemec