MitjaNemec/Kicad_action_plugins #63: Replicate layout: If run before saving all sheets in eeschema gives criptic message.

dinoghi, 2019-09-09:
To reproduce the issue:
-Create a hierarchical sheet with some  components in it (assign also footprints), don't save it.
-Copy and paste the hierarchical sheet (don't save).
Syncrhronize pcb (F8)
Use the script on one of the newly created sheets. An error pop up is shown with the message "u'5D76A583" and the "ok" button  (that seems to be the unique timestamp of the new sheet with the components to be placed).

Tested on master branch kicad, 
Cheers,
Dino.

dinoghi, 2019-09-09:
Also verified on kicad stable 5.1.2 (Linux)

MitjaNemec, 2019-09-10:
Yes the plugin has to read .sch files in order to get additional info not available from the .kicad_pcb. As there is currently no mechanism to check if the schematics was saved before running the plugin, there is not much that I can do. Maybe I could make the error message more user friendly. I'd appreciate the `replicate_layout.log` file to see where and how it fails. Thanks for reporting it though.

dinoghi, 2019-09-10:
Yes, having a  more descriptive error popup is a good solution (like "unable to open file xxx.sch: please check the sheet file has been saved in eeschema" or something similar that helps the user to understand that he missed the "save" step)
...and here is the log file.

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3594549/replicate_layout.log)

Cheers, 
Dino.