MitjaNemec/Kicad_action_plugins #64: Replicate layout: It does not fully handle components on both top and bottom layers.

dinoghi, 2019-09-09:
Trying to replicate a layout where the components are placed on both layers does not work correctly:

When the anchor and the pivot components are on a different layers the pivot layer is not changed.
In the attached file
[replicateLayout.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/3591791/replicateLayout.zip)

whe you select R10 (placed on bottom) to replicate the layout, you see the issue.

MitjaNemec, 2019-09-10:
I can't open the schematics with 5.1.4. Can you attach the .log file please?

dinoghi, 2019-09-10:
In this new attached zip file there is the full project directory including the .log file. (this time not zipped using kicad "zip" icon but manually compressing the whole folder.
[replicateLayout_5_1_4.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/3594517/replicateLayout_5_1_4.zip)

Cheers, Dino.

MitjaNemec, 2019-09-10:
The archive includes only an empty folder

dinoghi, 2019-09-10:
D'oh! ...doing it again (and checking twice, this time...)
[replicateLayout_5_1_4.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/3594719/replicateLayout_5_1_4.zip)

MitjaNemec, 2019-09-10:
There is something quite strange with this project. I still can not open it. Looking at .sch files I can see the following hierarchy:
the root file `replicateLayout.sch` is almost empty as it consist of only one hierarchical subsheet `file5D7697E6.sch`
This first level subsheet `file5D7697E6.sch` also consists of several hierarchical subsheets. The sheets are: `ReplicateLayout.sch`, `ReplicateLayout.sch`, `ReplicateLayout.sch`, `file5D7683FA.sch`, `file5D7683FA.sch`
 Thus you have circular reference, which results in infinite hierarchy (`ReplicateLayout.sch`->`file5D7697E6.sch`->`ReplicateLayout.sch`->...), thus the eeschema crashes, and running the plugin results in "maximum recursion depth exceeded"

How did you manage to create it? How do you see the hierarchy? And all this hierarchy issues are diverting the focus from the main issue.

The way I see it you want to place pivot footprint on let's say top layer and anchor footprint on either top or bottom layer. The plugin should consider this and appropriately mirror the replicated section. Obviously this is currently not supported and as a stop-gap, I'll try to add checks for this to notify the user and cancel the plugin. But in the long run this could be implemented but it'll be probably a while before I start working on it.

dinoghi, 2019-09-10:
Quite weird: I can still open it using 5.1.4, 5.1.2 and nightly without even a warning, but looking at the .sch file there is the circular reference so the question is "how can it work on my own machine?".
Anyway, Now I deleted the unneeded child sheets and made a simplified schematic/layout to check it.

[replicateLayout.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/3595121/replicateLayout.zip)

Cheers, 
Dino.

dinoghi, 2019-09-10:
...May be I found the reason why on your machine the project does not open correctly:  I'm using a case sensitive file system (Linux), so "replicateLayout.sch" and "ReplicateLayout.sch" are two different files when unzipped.. 
If you are using a case-insensitive file system (like dos/windows usually does) those files become the same and the circular reference is created.
If you still have a problem, delete the file "ReplicateLayout.sch" (uppercase "R" from zip file).
Cheers, 
Dino

MitjaNemec, 2019-09-10:
Aha, there was a nagging message from the unzip tool, but I thought it was overwriting the `fp-info-cache` so we could have caught this earlier

MitjaNemec, 2019-09-11:
I've added the checks for this condition so that the plugin emits meaningful message and stops the plugin.

dinoghi, 2019-09-11:
Hi Mitja,
 I just got it from git and on the same test project as before I get:
On stable versions (tested on kicad 5.1.2 and 5.1.4) it seems to work.
Develpment version (master branch): selecting R12 and launching the script I get a "Fatal error when running replicator...
[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3602546/replicate_layout.log)

From the log it seems that one of the used functions changed in the latest dev versions.

MitjaNemec, 2019-09-12:
Thanks for reporting.

The Flip method changed signature. Now it takes an additional argument specifying the direction of the flip (`bool aFlipLeftRight`). I'll try to handle it

MitjaNemec, 2019-09-12:
Fix commited if you could please test it I'd appreciate it.

MitjaNemec, 2019-09-16:
The source of this is KiCad code commit [37af3adffb0e4dc8643f40f644c982c1c44fe2be](https://github.com/KiCad/kicad-source-mirror/commit/37af3adffb0e4dc8643f40f644c982c1c44fe2be), which is connected to [bug/wish](https://bugs.launchpad.net/kicad/+bug/1836267).

Now from my perspective changing legacy `Flip(aCentre)` with `Flip(aCentre, False)` should behave the same. There is a possibility that there is a bug in KiCad code, but I'll have to research this more into detail