MitjaNemec/Kicad_action_plugins #66: Allow zoom on pcbnew when replicated plugin is on focus

hildogjr, 2019-09-20:
Is it possible allow the Pcbnew to get zoom out/in when the "Replicate layout" windows are in focus?

I am not sure because could be a KiCad limitation when running the external plugin.

MitjaNemec, 2019-09-25:
This is doable, and I can see the sense in it, but I'd have to make my dialog non-modal and it requires non trivial code changes. If I start to work on it, is this the only plugin that would benefit from non-modal dialogs or do you see (would you prefer) any other plugins also behaving this way?

hildogjr, 2019-09-30:
The other plugins that I use and you designed already have a dialog window that allow to interact with Pcbnew canvas.
In this specific case, I have a "big" (A4 paper size) board and I have to zoom in to choose the part to replicate. If was allowed to zoom out after the replicate windows is opened (because I have multiple hierarchical blocks and sublocks) I could be visually sure about what hierarchical I had choose.

MitjaNemec, 2019-10-01:
Yea with the recently added brighten feature I can see it makes sense to allow interaction with canvas. I've started to work on this

MitjaNemec, 2019-10-02:
I've managed to implement this, but the GUI part of the code was changed significantly and I'd rather not push it to `master` branch just yet. The plugin is available in the `Replicate_layout_non_modal` branch. So if you find the time, please test it. There is no hurry though.

hildogjr, 2019-10-02:
Works just fine! Good job. But it create a issue that I think we could workaround it:
After zoom in / zoom out once Pcbnew move the mouse cursor to the center of the canvas and this behavior is fine. When I have your replicate dialog opened, it coincides with the center of the dialog. Because this, I just can do a single movement (zoom in / zoom out).
A solution / workarournd is: load the dialog window in the right corner of the monitor or right enough to it be above the "Layers Manager" part of Pcbnew.

MitjaNemec, 2019-10-03:
Thanks for the feedback. Should be solved now, but as this is a GUI thing it will certainly behave differently on other platforms. Do you have a high DPI display, as I'd like to test it also there

hildogjr, 2019-10-04:
Don't, my primary (work PC) is 1920x1080 and secondary 1280x1024. It is a 2 monitor expanded setup and I found a interesting behavior.
If I start the replicate dialog with Pcbnew on my primary everything is as expected (the dialog load close to the Layers Manager) :-). The problem is using Pcbnew on secondary display, the replicate dialog is loaded centered in the first monitor.

MitjaNemec, 2019-10-04:
Thanks for testing. I really appreciate it. I was expecting some funny business with dual monitor setups and with HighDPI setups. I'll find someone with HighDPI setup, so don't worry about it. But for the issue you've found, I'll probably need you for testing the solution.

hildogjr, 2019-10-04:
If you add some log / debug info or want some specific behavior on the test, let me known.

MitjaNemec, 2019-10-04:
It's a GUI issue so the logs will not help much. I'll first try to reproduce it, then fix it and only then I'll ask you to confirm the fix.

MitjaNemec, 2019-10-07:
Now it works on MSW dual monitor setup, please test it on your setup, and if it works, I'll merge the changes to master.

hildogjr, 2019-10-07:
All fine in a Linux-Ubuntu 16.04.
I didn't test the other branch (about wx4 and Py3) because KiCad on Ubuntu 16 still using the Py2.

MitjaNemec, 2019-10-07:
Thanks for prompt response. This has been now merged to master

closed 2019-10-07 by MitjaNemec