MitjaNemec/Kicad_action_plugins #69: Assertion `aNetInfo->GetBoard() == GetBoard()' failed.

lumostor, 2019-10-08:
This is a the assertion failed as in #68 

I checked out  `Added_logging_for_track_setnet` branch. And did test on the ubuntu machine and no crash, and work well :).

I  did also with the debian machine and tried with the project in `Kicad_action_plugins/save_restore_layout/Source_project` restore run but crash. Here is the log: (NOTE: I replaced the my home directory portion of the path with ~)

Any clue?

```
10-07 22:29:17 save_restore_layout.action_save_restore_layout 154:Plugin executed with python version: '3.7.4+ (default, Sep  4 2019, 08:03:05) \n[GCC 9.2.1 20190827]'
10-07 22:29:17 save_restore_layout.action_save_restore_layout 155:Save/Restore Layout plugin version: 23 started
10-07 22:29:19 save_restore_layout.action_save_restore_layout 266:Restore layout chosen
10-07 22:29:21 save_restore_layout.save_restore_layout 362:Getting board info
10-07 22:29:21 save_restore_layout.save_restore_layout 364:Getting schematics info
10-07 22:29:21 save_restore_layout.save_restore_layout 150:getting project hierarchy from schematics
10-07 22:29:21 save_restore_layout.save_restore_layout 152:Project hierarchy looks like:
{'5B767FE4': ['Leg1', '~/src/kicad-stuff/Kicad_action_plugins/save_restore_layout/Source_project/Leg.sch'], '5B767FF2': ['Power+', '~/src/kicad-stuff/Kicad_action_plugins/save_restore_layout/Source_project/Power.sch'], '5B767FF5': ['Sensor', '~/src/kicad-stuff/Kicad_action_plugins/save_restore_layout/Source_project/Sensor.sch'], '5B772791': ['Power-', '~/src/kicad-stuff/Kicad_action_plugins/save_restore_layout/Source_project/Power.sch'], '5B76D023': ['Leg2', '~/src/kicad-stuff/Kicad_action_plugins/save_restore_layout/Source_project/Leg.sch'], '5B76D031': ['Leg3', '~/src/kicad-stuff/Kicad_action_plugins/save_restore_layout/Source_project/Leg.sch'], '5C575278': ['Power_extra', '~/src/kicad-stuff/Kicad_action_plugins/save_restore_layout/Source_project/Power.sch']}
10-07 22:29:21 save_restore_layout.save_restore_layout 366:Getting layout info
10-07 22:29:21 save_restore_layout.save_restore_layout 225:getting a list of all footprints on board
10-07 22:29:21 save_restore_layout.save_restore_layout 368:Updating layout info with schematics info
10-07 22:29:21 save_restore_layout.save_restore_layout 707:Loading saved design
10-07 22:29:21 save_restore_layout.save_restore_layout 714:Source levels is:['Leg.sch', 'Power.sch']
10-07 22:29:21 save_restore_layout.save_restore_layout 725:Destination levels is:['Leg.sch', 'Power.sch']
10-07 22:29:21 save_restore_layout.save_restore_layout 730:All sch files required are: ['Power.sch']
10-07 22:29:21 save_restore_layout.save_restore_layout 732:Getting current schematics hash
10-07 22:29:21 save_restore_layout.save_restore_layout 742:Source hash is:'f2e7e18230d755050d3e042d519e794d'
10-07 22:29:21 save_restore_layout.save_restore_layout 743:Destination hash is: 'f2e7e18230d755050d3e042d519e794d'
10-07 22:29:21 save_restore_layout.save_restore_layout 760:Get layout data from saved board
10-07 22:29:21 save_restore_layout.save_restore_layout 225:getting a list of all footprints on board
10-07 22:29:21 save_restore_layout.save_restore_layout 473:Replicating footprints
10-07 22:29:21 save_restore_layout.save_restore_layout 575:Replicating tracks
10-07 22:29:21 save_restore_layout.save_restore_layout 610:Setting track net to: 1, '+15V'
```

MitjaNemec, 2019-10-08:
Okay, I think I am on it. It fails on the first track. I think that assert in the code checks if the net is actually on the board. The code was introduced in 2016 and hasn't been changed since. So it is unclear why different behaviour. 

Can you be specific how do you proceed with restore part of the plugin. I assume you:
1. start with blank pcb
2. update the pcb from eeschema
3. you select pivot footprint
4. run the restore part of the plugin.

Is there any difference if you add a step between 2 and 3, where you save the pcb. 

Also can you stop there and open the .kicad_pcb file with text editor and copy-paste the list of nets on the board.

MitjaNemec, 2019-10-08:
There is also the option that you've got different builds, and in the one that does not crash the asserts are disabled. So maybe my plugin was doing something bugworthy all the time, but it shows only with specific build configurations.  I'll summon @qu1ck and maybe he can confirm if this is one possibility

qu1ck, 2019-10-08:
Yeah, this is just a sanity check in kicad. Before setting net add the net to board and use the net object from the board instead of your own.
This will only popup on kicad builds with asserts enabled, not release builds.

lumostor, 2019-10-09:
Looks like it is the case. I created PR #70 which partially correct the problem. Partially because it crash elsewhere with another assertion fail.

MitjaNemec, 2019-10-10:
Okay, something funny is going on. It is obvious that there is/was a couple of things wrong with my code.
1. Why do I need `new_track.SetNetCode(to_net_code)` and `new_track.SetNet(to_net_item)`. Either one should be ok.
2. While the `new_track.SetNet(to_net_item)` works on `replicate_layout` plugin it is obvious that I have a problem here, as I load a saved board, thus the net_item is from the saved board. And then I try to add the track from the destination board to this net_item. So the assert in `BOARD_CONNECTED_ITEM::SetNet(NETINFO_ITEM *)` is correctly triggered

Obviously we should only need `new_track.SetNetCode(to_net_code)` so the PR #70 is in the right direction why it fails I currently don't know.

I'll have to think more about it.

lumostor, 2019-10-11:
Got a bit of time tonight, I reverted to your code and commented out `new_track.SetNet(to_net_item)`.
It goes goes thought up to `BOARD_CONNECTED_ITEM::SetNetCode(int, bool): Assertion `m_netinfo' failed.`. So PR #70 is of no real use. I should have more time tomorrow.

MitjaNemec, 2019-10-11:
Aha, here's a thought. I am not really familiar with C++ but I guess the 
`if( ( aNetCode >= 0 ) && board )`
checks if the netcode exist on the board **where the track is**. As the track at this point was not added to the board, this is obviously not true. Then the if clause will return `m_netinfo` as null (&NETINFO_LIST::ORPHANED_ITEM) which will trigger assert later on.

So can you try rearranging the code that the `new_track` is added to the board before it is assigned a net.

Whole code snippet 
```
bool BOARD_CONNECTED_ITEM::SetNetCode( int aNetCode, bool aNoAssert )
{
    if( !IsOnCopperLayer() )
        aNetCode = 0;

    // if aNetCode < 0 ( typically NETINFO_LIST::FORCE_ORPHANED )
    // or no parent board,
    // set the m_netinfo to the dummy NETINFO_LIST::ORPHANED

    BOARD* board = GetBoard();
    //auto connectivity = board ? board->GetConnectivity() : nullptr;
    //bool addRatsnest = false;

    //if( connectivity )
        //addRatsnest = connectivity->Remove( this );

    if( ( aNetCode >= 0 ) && board )
        m_netinfo = board->FindNet( aNetCode );
    else
        m_netinfo = &NETINFO_LIST::ORPHANED_ITEM;

    if( !aNoAssert )
        assert( m_netinfo );

    // Add only if it was previously added to the ratsnest
    //if( addRatsnest )
    //    connectivity->Add( this );

    return ( m_netinfo != NULL );
}
```

lumostor, 2019-10-12:
Bang on :). You were right. I put the add track (and zone) just after the duplicate and bang no more crash. I also left the `new_track.SetNet(to_net_item)` in place, with the add track/zone before they were no longer crashing, I was not sure enough if they were really needed or not. The pull request has been amended with the new fix. There was also the same issue with the zones.

Thanks a lot for this awesome plugin.

MitjaNemec, 2019-10-12:
Thanks for reporting testing and submitting the PR.

closed 2019-10-12 by MitjaNemec