MitjaNemec/Kicad_action_plugins #71: ArchiveProject: problem with kicad_wks file

judge2005, 2019-11-08:
I tried running ArchiveProject but I get this error.
[image]

I tried deleting the file, but I still get the same error!

MitjaNemec, 2019-11-09:
Thanks for reporting. The archive worksheet functionality has only been added recently and has not been tested thoroughly. A potential fix has been pushed just now and I'd really appreciate it if you would test it and report back.

judge2005, 2019-11-09:
Thanks for that. Now I get a permission problem. FYI EESchema was not open, but project and pcbnew were. I tried it with just having pcbnew open, but get the same error:

[image]

judge2005, 2019-11-09:
Just took a look at the code. You are assuming that the worksheet file is not already in the project folder. That is not the case with my projects.

MitjaNemec, 2019-11-11:
Thanks for testing. I can not recreate the issue and it might be due to OS configuration. Base upon my previous experience, issues like this will probably take a bit back and forth to find the root cause and to fix it. So, I've added some instrumentation and I'd like you to try again, and report back. Please attach `archive_project.log` (it can be redacted to hide any project information) and any screenshots required.

judge2005, 2019-11-12:
A different error this time:

[image]

[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3834907/archive_project.log)

MitjaNemec, 2019-11-12:
I should have caught this earlier, so I apologize for a bit of back and forth. 

Anyhow your reference to pcbnew worksheet file is empty and the plugin did not handle this properly. I hope I've fixed this.

judge2005, 2019-11-12:
Now I get this:

[image]
[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3837918/archive_project.log)

MitjaNemec, 2019-11-14:
Whoa! Since the line 150 is a comment line it should not raise any exceptions. There is something funny going on and it took me a while to consider all the possibilities. And I can narrow it down to two:
1. For whatever reason python interpreter did not regenerate `archive_project.pyc` file and while the `archive_project.py` was updated, the interpreter executed previous version of the script
2. You had KiCad already open, when you updated `archive_project.py` file. As `.pyc` files regenerate upon start of KiCad, interpreter executed previous version of the script

So could you please delete all the `.pyc` files in the plugin folder, start KiCad and run the plugin again.

Thanks for the persistence.

judge2005, 2019-11-15:
It says it archived successfully, then when I hit OK I got this:
[image]

MitjaNemec, 2019-11-16:
Man you are really testing the plugins. This time you uncovered a bug in the archive 3D models functionality. The path `Capacitors_SMD.3dshapes/C_1210.wrl` is neither absolute nor relative (at least to my understanding.

Do you by any chance have a `Capacitors_SMD.3dshapes` subfolder in the project folder, and `C_1210.wrl` is in this subfolder? If this is the case, do you see the 3D model for this footprint in the KiCad's 3D viewer? If so, then I'll need to properly parse such paths.

judge2005, 2019-11-16:
No. It is relative to C:\Program Files\KiCad\share\kicad\modules\packages3d. i.e. there is a file C:\Program Files\KiCad\share\kicad\modules\packages3d\Capacitors_SMD.3dshapes\C_1210.wrl

Which is basically what it says in your log file:
`11-15 11:03:42 archive_project.archive_project 542:KISYS3DMOD path: C:\Program Files\KiCad\share\kicad\modules\packages3d
11-15 11:03:42 archive_project.archive_project 569:Trying to copy: Capacitors_SMD.3dshapes/C_1210.wrl
`

MitjaNemec, 2019-11-17:
Yes, but if it is relative to path defined with KiCad or enviroment variable it should be completely define with the variable. In your case this should be `${KISYS3DMOD}/Capacitors_SMD.3dshapes/C_1210.wrl`.

As such I am treating your path definition incomplete as the plugin can not know to which variable the path is relative to. This is why I am asking how is the 3D viewer behaving.

I could try all possible combinations, but what if the path is legal for two different variables? Obviously the lazy thing would be to stop at the first successful match.

MitjaNemec, 2019-11-17:
To be fair, even if the path is incomplete, the plugin should handle this gracefully without crashing

judge2005, 2019-11-17:
I will check what the 3D viewer does. BTW, which file are you getting this path from?

MitjaNemec, 2019-11-17:
If you take new release of the plugin, the info should be in the .log file. Also the plugin should not crash anymore. Thanks for testing.

judge2005, 2019-11-19:
I checked the 3D view. It works fine. I would expect KiCAD to stop at the first successful match for KISYS3DMOD - that is how paths are supposed to work - the order is important.

Here are the results of using your latest version:

[image]

Followed by:

[image]

Log file:

[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/3864100/archive_project.log)

MitjaNemec, 2019-11-20:
Thanks for checking with 3D viewer. I am not really happy that such an ambiguous path gets resolved, but I'll align the plugin with KiCad behavior. I might even raise a bug report with KiCad just to see the reasoning behind it.

But for the time being, I'll assume with ambiguous paths that they are relative to ${KISYS3DMOD}. The fix is in and it should also cover the latest bug that you've uncovered.

As always I'll appreciate the testing and feedback.

judge2005, 2019-11-20:
Thanks. Everything appears to work now.

MitjaNemec, 2019-11-21:
Thanks for persistent testing. It is much appreciated.

closed 2019-11-21 by MitjaNemec