MitjaNemec/Kicad_action_plugins #72: Allow alternative angle

a7hybnj2, 2019-12-24:
Allow for circular layout as illustrated in attached screenshot.
[image]

MitjaNemec, 2019-12-24:
This is already doable.

The footprint that you select when calling the plugin has to be oriented vertically, as the selected serves as an anchor and the plugin and goes  clockwise/counterclokwise (increasing/decereasing references/sheet numbers)

Before
[image]

After
[image]

a7hybnj2, 2019-12-24:
@MitjaNemec Yep, I had just tried this. I came back here to let you know I figured it out and you beat me too it!

Seems awesome and exactly what I wanted. Thanks!

closed 2019-12-24 by a7hybnj2