MitjaNemec/Kicad_action_plugins #73: place_footprints circle center point

a7hybnj2, 2019-12-24:
This is great but I am having a little trouble. I am trying to align my rings of parts. For parts that have an easy to find center point like a circle with 36 parts it is easy. But, for circles with 24 parts it is more difficult to move the ring from a central point. Even for my rings of 36 parts with a easy to find center I ended up moving an extra resistor to the center point so that I could move it with that location as 0.
[image]

I know the highlight reference part is used as the 12 o'clock part but is it possible to specify a center point to wrap the parts around?

a7hybnj2, 2019-12-24:
This would also be easier if the radius displayed was for the center of the part and not for the very edge. That way you could come down from the 12 o'clock part by the radius but since the radius looks to be calculated by the CrtYd that point isn't always on a grip point.

a7hybnj2, 2019-12-24:
I just added a target to the fab layer for the ones I could find center for to test if it could be used to grab from. It looks like it can. Could you just calculate a target for the center of the circle of parts?
[image]

MitjaNemec, 2019-12-25:
I understand your issue and I agree that some way of specifying center point and radius type (edge of component or center of component) would make it more usable. I'll have to think about it how to implement it. But I'll avoid relying on any external data within the design (markers on other layers). Probably user will have to enter center point within the plugin GUI, but at the moment I don't have a clear view how to change the GUI so that it will be intuitive

a7hybnj2, 2019-12-25:
My idea for the crosshairs wasn’t that the user would specify or create them. It was something that would be generated by place_footprints so the user can then select and use it as a grid point to reference center.

So, keep everything as is but generate a crosshair on the fab layer to indicate center.

MitjaNemec, 2019-12-26:
OK, I understand what you want. But what I don't understand is what can a user do with this info?

If you want to place two concentric circles as per your figure, when placing the second one it would be way easier if user would specify the center point before running the plugin.

And the feature you're asking for might be worth doing it also as a standalone plugin (find midpoint or center point).

a7hybnj2, 2019-12-26:
In my initial test when you bulk select the circle and hit 'm' with the mouse nearest the center of the circle for the crosshair it would snap-to that point. So, you could then move the entire circle with from the center point and stack them up.

I agree adding a center point to the function would be the best final solution.

I also agree that having a script for 'mark center' would be great. Would be cool even on larger singular parts that have 0,0 on pad 1.

MitjaNemec, 2019-12-27:
Thanks for the feedback. It is clearer now. I'll still go with the GUI change though. I don't know why and but I don't agree that a plugin (or any kind of tool) which modifies only placement should produce any kind of graphics on any layer. If KiCad would include a temporary layer which would not be saved as a part of kicad_pcb file, the I might consider it.

So my current plan is to first solve #74 and while I'm at it add GUI for center point specification.

MitjaNemec, 2019-12-29:
Okay, I've played around a bit and the feature to rotate around the bottom edge of the footprint instead of center point of the footprint is quite annoying when trying to position the circular placement at precise rotation point. So I've changed this part so that the plugin will rotate parts around component center point.

I think that this solves most of the issues, as you can move the component to desired offset from the rotation point using grids. And then you can use this offset as a radius when calling the plugin. So I've managed to place concentric circles with ease.

As for your idea regarding the annotation, I have an idea for partial solution and I'd like to run it by you.
The plugin could draw graphical lines denoting rotation center and highlight them (so that they appear bright greed) while plugin is active, and when the plugin finishes, the lines would be deleted. Thus the plugin would not modify the layout (besides footprint positions).

a7hybnj2, 2019-12-29:
The way you describe changing the rotational point leads me to believe marking the center point would be unnecessary. I was going to use the center point to move the circles around with after creating them.

I would, off board, make the circular parts and put a 1 grid circle with the center @ the center of the component circle. Then when I selected the circle I could use that as a grab point to move the parts around. I only did this because it was confusing to get concentric circles lined up.

I would say skip it or have a checkbox to make and keep a central marker on a layer like fab where it shouldn't negatively effect anything.

a7hybnj2, 2019-12-29:
I will pull and test the changes tonight when I get back to my workspace.

closed 2020-08-03 by MitjaNemec