MitjaNemec/Kicad_action_plugins #74: Matrix doesn't obey selection

a7hybnj2, 2019-12-27:
I have 250 resistors on my board. I took 1-50 and turned them into circles. I then tried to take the remainder and turn them into a matrix but it included my 1-50 that had already been placed. I tried multiple times to make sure I had the selection correct and every time it matrix'ed the entire 250 resistors.

MitjaNemec, 2019-12-27:
Yeah that is a bug. Thanks for spotting and reporting.

I don't see any other way of solving it than generating a list of footprints and letting the user select the ones suitable for placement. This will need a complete GUI rehaul, which I'll do with #73.

Current workaround is to annotate them so that you separate them by at least one count. (1-50, 52-251)

MitjaNemec, 2019-12-29:
Disregard my previous comment. It was written hastily as I did not remember that I already had the selection part in the GUI, and I did not read your report thoroughly. This should be a lesson for me never to comment on the issues via phone.

Anyhow, I've found the bug and fixed it. Thanks for reporting it.

a7hybnj2, 2019-12-29:
I just pulled and tested. It does appear to be fixed in my test case.

Cool!

closed 2019-12-29 by a7hybnj2