MitjaNemec/Kicad_action_plugins #75: Crash when restoring layout

greenscreenflicker, 2020-01-05:
The following crash occures when restoring a layout
[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4023490/save_restore_layout.log)
[7segment_disp.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/4023491/7segment_disp.zip)

MitjaNemec, 2020-01-07:
First of all, thanks for reporting it. But it will take me a couple of days to get on it. From the brief overview of the .log file I can tell that saving of the layout worked without issues, but there is no messages from the restore part of the layout. Can you reproduce this and if you can what kind of messages you get (attach screenshot if needed). Thanks  for the feedback.

MitjaNemec, 2020-01-11:
OK, I've looked into it. From the data available (.pckl file and .log file) I can not really tell what went wrong. The only think that I can tell is that the layout has been successfully saved. As there might be an issue (bug) with logging and the restore part was not logged, can you try restoring the layout after restarting KiCad (not just the layout, also the project manager).

closed 2020-08-03 by MitjaNemec