MitjaNemec/Kicad_action_plugins #79: Crash while duplicate layoute

xDraconix, 2020-02-15:
KiCAD Version: (5.1.5)-3, release build

Ive got following error while replicate:

[image]

here are the recent log file:

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4208489/replicate_layout.log)

Bratwurstdealer, 2020-02-15:
I'm having the same problem with version (5.0.1)-3, release build running on windows 10.

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4208505/replicate_layout.log)

xDraconix, 2020-02-15:
Sorry, that was my fault! I have delete one compoment after routing them. After that, the footprint still exist at pcbnew, but in the newly importet second hierarchical sheet, they cant route to the non existing footprint.

Closed for my issue.

closed 2020-02-15 by xDraconix

MitjaNemec, 2020-02-16:
@Bratwurstdealer the error window within the KiCad is the same for any error caught. Looking at the .log file, I can see that you have a different issue. It might be that 5.0.1 does not have the SetKeepUpright method implemented.

Can you confirm this by opening the python console within pcbnew and enter
```
import pcbnew
repr(pcbnew.TEXTE_MODULE.SetKeepUpright)
```
you should get `'<unbound method TEXTE_MODULE.SetKeepUpright>'` as an answer.

if you get something like `AttributeError: type object 'TEXTE_MODULE' has no attribute 'SetKeepUpright'` then you have two options:
1. either you upgrade KiCad to 5.1. branch
2. use [older revision](https://github.com/MitjaNemec/Kicad_action_plugins/commits/master) of the plugin (the latest commit that should still work is d3655a)

MitjaNemec, 2020-02-16:
@xDraconix Thanks for letting me know. I'll still look int it as it would be nice if the plugin reported more specific message

Bratwurstdealer, 2020-02-16:
Thanks for the response @MitjaNemec!
The problem was the outdated Kicad version I used. I updated my Kicad and now it works fine.