MitjaNemec/Kicad_action_plugins #80: Save/Restore gives error when restoring

roboart2019, 2020-02-16:
Using the example , I am getting the following error when restoring. 

u'Leg.sch' is not on the list

Log

02-16 09:57:50 save_restore_layout.action_save_restore_layout 154:Plugin executed with python version: '2.7.16 (default, Aug 15 2019, 17:51:40)  [GCC 9.2.0 64 bit (AMD64)]'
02-16 09:57:50 save_restore_layout.action_save_restore_layout 155:Save/Restore Layout plugin version: 30 started
02-16 09:57:52 save_restore_layout.action_save_restore_layout 266:Restore layout chosen
02-16 09:58:06 save_restore_layout.save_restore_layout 362:Getting board info
02-16 09:58:06 save_restore_layout.save_restore_layout 364:Getting schematics info
02-16 09:58:06 save_restore_layout.save_restore_layout 150:getting project hierarchy from schematics
02-16 09:58:06 save_restore_layout.save_restore_layout 152:Project hierarchy looks like:
{'5CAF0FB9': ['Leg 2', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Leg.sch'], '5CADA2A6': ['power', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Power.sch'], '5CADA6B4': ['Leg', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Leg.sch'], '5B772791': ['Power-', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Power.sch'], '5B767FF5': ['Sensor', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Sensor.sch'], '5B767FF2': ['Power+', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Power.sch']}
02-16 09:58:06 save_restore_layout.save_restore_layout 366:Getting layout info
02-16 09:58:06 save_restore_layout.save_restore_layout 225:getting a list of all footprints on board
02-16 09:58:06 save_restore_layout.save_restore_layout 368:Updating layout info with schematics info
02-16 09:58:06 save_restore_layout.save_restore_layout 707:Loading saved design
02-16 09:58:06 save_restore_layout.save_restore_layout 714:Source levels is:[u'Leg.sch']
02-16 10:03:16 save_restore_layout.action_save_restore_layout 303:Error when restoring layout
Traceback (most recent call last):
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/action_save_restore_layout.py", line 296, in Run
    restore_layout.restore_layout(pivot_mod, layout_file)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 721, in restore_layout
    indx = anchor_mod.filename.index(last_level)
ValueError: u'Leg.sch' is not in list

MitjaNemec, 2020-02-17:
Hi, thanks for reporting this.

Could you be more specific what steps did you take that resulted in the crash of the plugin. Please be as verbose as you can. 

Also can you specify which version of KiCad are you running.

Thanks fro the feedback

roboart2019, 2020-02-18:
Nightly build.  

I was following the instructions on Github, with the demo project.

Select one footprint which is in the hierarchy that you want to save the layout for.
Run the plugin and choose "Save layout"
Select into which file should the layout be saved
Create a new project. Copy the chosen .sch files used for hierarchy to the project folder and link it into schematics
Create a new layout.
Place and select the anchor footprint. This defines the position and orientation of restored layout.
Run the plugin.
Choose "Restore layout" 9: Select in which file is tha layout data stored
Voila, observe restored layout

MitjaNemec, 2020-02-20:
Ok, I''ve run the plugin with 5.1.5 and I did not had any issues saving/restoring the test project at either hierarchical level (leg, power)

On the other hand with today's nightly (5.99.0-885-g625249322) the `leg` hierarchical level is restored properly, while the `power` level is not restored properly (tracks are not placed properly). More specifically if I restore from the layout saved by 5.1.5, the layout is restored fine. This leads me to believe that the save part of the plugin has issues with 5.99.

Nevertheless I could not reproduce you're issues. As the issue might be present only with specific nightly revisions, please be more specific about the version.

As is, I am only actively supporting the 5.1.x branch, but I'd like to track this down anyway. As I can not reproduce the issue I'll need your help. Would be kind enough to help track this down? It will mainily require that you download new releases of the plugin run it and send me back output files (.pckl and .log). Usually a couple of back and forth messages are enough.

roboart2019, 2020-02-22:
Yes whatever you need, thanks

Version (5.99.0-748-gb3af41e1b)  release build
Platform Windows 8(build 9200) 64 bit edition, 64 bit

Application: Pcbnew
Version: (5.99.0-748-gb3af41e1b), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Build date: Jan 14 2020 22:17:26
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

MitjaNemec, 2020-02-24:
So, I've just pushed a couple of changes, so that the .log file should have more details, so I'll be able to narrow it down. 

Please update the plugin and rerun it and send me back the .log file.

MitjaNemec, 2020-03-03:
@roboart2019  ping. Would you mind testing the latest version?

roboart2019, 2020-03-03:
Ok, sorry for delay working on it now, Thanks

roboart2019, 2020-03-03:
I am new so I hope I am not wasting your time with a mistake I am making.  I have included the logs now from the run today.  I will be watching closely for your responses in order to provide you faster feedback.  Let me know if you want me to do anything else.

03-03 10:46:59 save_restore_layout.action_save_restore_layout 154:Plugin executed with python version: '2.7.16 (default, Aug 15 2019, 17:51:40)  [GCC 9.2.0 64 bit (AMD64)]'
03-03 10:46:59 save_restore_layout.action_save_restore_layout 155:Save/Restore Layout plugin version: 30 started
03-03 10:47:01 save_restore_layout.action_save_restore_layout 192:Save layout chosen
03-03 10:47:01 save_restore_layout.save_restore_layout 805:Saving the current board temporary in order to leave current layout intact
03-03 10:47:01 save_restore_layout.save_restore_layout 815:Getting schematics info
03-03 10:47:01 save_restore_layout.save_restore_layout 150:getting project hierarchy from schematics
03-03 10:47:01 save_restore_layout.save_restore_layout 152:Project hierarchy looks like:
{'5B76D031': ['Leg3', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Leg.sch'], '5B76D023': ['Leg2', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Leg.sch'], '5B772791': ['Power-', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Power.sch'], '5B767FE4': ['Leg1', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Leg.sch'], '5C575278': ['Power_extra', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Power.sch'], '5B767FF5': ['Sensor', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Sensor.sch'], '5B767FF2': ['Power+', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Power.sch']}
03-03 10:47:01 save_restore_layout.save_restore_layout 817:Getting layout info
03-03 10:47:01 save_restore_layout.save_restore_layout 225:getting a list of all footprints on board
03-03 10:47:01 save_restore_layout.save_restore_layout 819:Updating layout info with schematics info
03-03 10:47:02 root 47:highlighting module: u'C301'
03-03 10:47:02 root 47:highlighting module: u'C302'
03-03 10:47:02 root 47:highlighting module: u'C401'
03-03 10:47:02 root 47:highlighting module: u'C501'
03-03 10:47:02 root 47:highlighting module: u'C502'
03-03 10:47:02 root 47:highlighting module: u'D301'
03-03 10:47:02 root 47:highlighting module: u'D501'
03-03 10:47:02 root 47:highlighting module: u'J201'
03-03 10:47:02 root 47:highlighting module: u'J202'
03-03 10:47:02 root 47:highlighting module: u'Q301'
03-03 10:47:02 root 47:highlighting module: u'Q302'
03-03 10:47:02 root 47:highlighting module: u'Q501'
03-03 10:47:02 root 47:highlighting module: u'Q502'
03-03 10:47:02 root 47:highlighting module: u'R301'
03-03 10:47:02 root 47:highlighting module: u'R302'
03-03 10:47:02 root 47:highlighting module: u'R401'
03-03 10:47:02 root 47:highlighting module: u'R501'
03-03 10:47:02 root 47:highlighting module: u'R502'
03-03 10:47:03 root 47:highlighting module: u'U301'
03-03 10:47:03 root 47:highlighting module: u'U401'
03-03 10:47:03 root 47:highlighting module: u'U501'
03-03 10:47:11 save_restore_layout.save_restore_layout 889:Calculating hash of the layout schematics
03-03 10:47:11 save_restore_layout.save_restore_layout 893:Saving hash for files: [u'Leg.sch', u'Power.sch']
03-03 10:47:11 save_restore_layout.save_restore_layout 912:Removing everything else from the layout
03-03 10:47:11 save_restore_layout.save_restore_layout 837:Removing text
03-03 10:47:11 save_restore_layout.save_restore_layout 823:Removing drawing
03-03 10:47:11 save_restore_layout.save_restore_layout 851:Removing zones
03-03 10:47:11 save_restore_layout.save_restore_layout 867:Removing tracks
03-03 10:47:11 save_restore_layout.save_restore_layout 884:Removing modules
03-03 10:47:11 save_restore_layout.save_restore_layout 931:Saving layout in temporary file
03-03 10:47:11 save_restore_layout.save_restore_layout 934:Reading layout as text
03-03 10:47:11 save_restore_layout.save_restore_layout 941:Saving layout data
03-03 10:47:11 save_restore_layout.save_restore_layout 948:Succesfully saved the layout

03-03 10:47:53 save_restore_layout.action_save_restore_layout 154:Plugin executed with python version: '2.7.16 (default, Aug 15 2019, 17:51:40)  [GCC 9.2.0 64 bit (AMD64)]'
03-03 10:47:53 save_restore_layout.action_save_restore_layout 155:Save/Restore Layout plugin version: 30 started
03-03 10:47:55 save_restore_layout.action_save_restore_layout 266:Restore layout chosen
03-03 10:47:59 save_restore_layout.save_restore_layout 362:Getting board info
03-03 10:47:59 save_restore_layout.save_restore_layout 364:Getting schematics info
03-03 10:47:59 save_restore_layout.save_restore_layout 150:getting project hierarchy from schematics
03-03 10:47:59 save_restore_layout.save_restore_layout 152:Project hierarchy looks like:
{'5CAF0FB9': ['Leg 2', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Leg.sch'], '5CADA2A6': ['power', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Power.sch'], '5CADA6B4': ['Leg', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Destination_project\\Leg.sch'], '5B772791': ['Power-', u'C:\\Users\\ [... cut ...]
03-03 10:48:01 save_restore_layout.action_save_restore_layout 303:Error when restoring layout
Traceback (most recent call last):
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/action_save_restore_layout.py", line 296, in Run
    restore_layout.restore_layout(pivot_mod, layout_file)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 721, in restore_layout
    indx = anchor_mod.filename.index(last_level)
ValueError: u'Leg.sch' is not in list

MitjaNemec, 2020-03-03:
> I am new so I hope I am not wasting your time with a mistake I am making. I have included the logs now from the run today. I will be watching closely for your responses in order to provide you faster feedback. Let me know if you want me to do anything else.

Thanks for the feedback. As I am a bit short on time it will happen from time to time that it might take ma a week or more to analyze the logs, change the code and ask you again to test it.

roboart2019, 2020-03-03:
Is my python version correct?

MitjaNemec, 2020-03-03:
Yup, the plugin should work with py2.7.x and py>3.6

roboart2019, 2020-03-03:
Ok good, well I believe that a run with the recent version of your plugin.  Perhaps I will download the new nightly?

MitjaNemec, 2020-03-03:
If you have the time, go ahead, but then also post the version of the KiCad you used when appending .log file

roboart2019, 2020-03-03:
Application: Pcbnew
Version: (5.99.0-748-gb3af41e1b), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Build date: Jan 14 2020 22:17:26
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

roboart2019, 2020-03-03:
Current Nightly , Yikes I am getting a different error, maybe I copied something wrong.  Let me know if you have any ideas, Thx

Application: Pcbnew
Version: (5.99.0-1023-gd6706c257), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Build date: Mar 3 2020 09:03:41
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013
Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

03-03 14:30:00 save_restore_layout.action_save_restore_layout 154:Plugin executed with python version: '2.7.16 (default, Aug 15 2019, 17:51:40)  [GCC 9.2.0 64 bit (AMD64)]'
03-03 14:30:00 save_restore_layout.action_save_restore_layout 155:Save/Restore Layout plugin version: 30 started
03-03 14:30:02 save_restore_layout.action_save_restore_layout 192:Save layout chosen
03-03 14:30:02 save_restore_layout.save_restore_layout 805:Saving the current board temporary in order to leave current layout intact
03-03 14:30:02 save_restore_layout.save_restore_layout 815:Getting schematics info
03-03 14:30:02 save_restore_layout.save_restore_layout 150:getting project hierarchy from schematics
03-03 14:30:02 save_restore_layout.save_restore_layout 152:Project hierarchy looks like:
{'5B76D031': ['Leg3', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Leg.sch'], '5B76D023': ['Leg2', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Leg.sch'], '5B772791': ['Power-', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Power.sch'], '5B767FE4': ['Leg1', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Leg.sch'], '5C575278': ['Power_extra', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Power.sch'], '5B767FF5': ['Sensor', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Sensor.sch'], '5B767FF2': ['Power+', u'C:\\Users\\gdott\\Desktop\\Kicad\\save_restore_layout\\save_restore_layout\\Source_project\\Power.sch']}
03-03 14:30:02 save_restore_layout.save_restore_layout 817:Getting layout info
03-03 14:30:02 save_restore_layout.save_restore_layout 225:getting a list of all footprints on board
03-03 14:30:02 save_restore_layout.action_save_restore_layout 198:Fatal error when creating an instance of SaveLayout
Traceback (most recent call last):
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/action_save_restore_layout.py", line 196, in Run
    save_layout = save_restore_layout.SaveLayout(board)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 818, in __init__
    self.layout = PcbData(self.board)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 226, in __init__
    self.modules = self.get_board_modules(board)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 209, in get_board_modules
    mod_id=self.get_module_id(module),
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 185, in get_module_id
    module_path = module.GetPath().split('/')
AttributeError: 'KIID_PATH' object has no attribute 'split'

MitjaNemec, 2020-03-04:
Yeah, the timestamps moved to ID (KIID). So there are some changes to python API. This might be an insurmountable issue, as I really don't want to maintain two codebases (one for 5.1.x and the other for 5.99) so I'll most likely leave this open. I'll let you know.

Now we might want to solve the original issue, but you'll need to revert do an older nightly. Or it is maybe worth checking if you can reproduce the same issue with a 5.1.x branch (either 5.1.5 or the latest 5.1.x nightly). But move slowly. For start just think about it whether this is doable. As the ball is on my side, I'll first check if I can solve this, and if I can't, then and only then we can decide how to proceed.

roboart2019, 2020-03-04:
Ok let me know what version to try it with next. Thanks

MitjaNemec, 2020-03-07:
Thanks for the details, I'll have to take a look. But in the future, can you please drag the files into the message (I've edited your message) so that the thread will be more readable.

roboart2019, 2020-03-07:
Yes absolutely, Thanks

MitjaNemec, 2020-03-08:
Can you please update the plugin and run it again. The nightly compatibility issue should be fixed and I've also added additional instrumentation which should help me to narrow down the problem.
Thanks

roboart2019, 2020-03-09:
Hi, I got an error right after running the plugin on the PCB of the  source project.  This occured before any prompts.   Thanks for your work on this plugin and your help in resolving this.

[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4304509/save_restore_layout.log)

Application: Pcbnew
Version: (5.99.0-1090-gf079c4111), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Build date: Mar  8 2020 21:58:44
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

MitjaNemec, 2020-03-09:
From what I can tell from the .log file, you did not run the latest version of the plugin. You are still on version 30, while current version is 39 (contents of version.txt).

roboart2019, 2020-03-10:
Ok, sorry about that.  

[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4309945/save_restore_layout.log)

Application: Pcbnew
Version: (5.99.0-1090-gf079c4111), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Build date: Mar  8 2020 21:58:44
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

MitjaNemec, 2020-03-10:
Sorry, I did not test the fix for the 5.99 (nightly) compatibility properly. Now it should work.
Please update, run again and send me the .log file. You don't need to report the KiCad version info. The important part is already embedded in the .log file

roboart2019, 2020-03-10:
Thanks, here are the latest results. I renamed leg.pckl to a text file extension to enable uploading.

[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4312694/save_restore_layout.log)

[Leg_pckl.txt](https://github.com/MitjaNemec/Kicad_action_plugins/files/4312703/Leg_pckl.txt)

[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4312708/save_restore_layout.log)

MitjaNemec, 2020-03-10:
OK, I've figured it out. When you've saved the layout you've saved the layout for complex hierarchy (Leg.sch which has three subsheets: Sensor, Power+ and Power-). But when restoring, you tried to restore to the layout only and only for the "power" subsheet, which does not have leg as a parrent.

Obviously I probably could have make the plugin to detect this and then restore only part of the saved layout. But for the time being, I'll add proper detection of this so that plugin generates a meaningful message and fails gracefully. What is your take on it, and what kind of message would make sense to you?

roboart2019, 2020-03-10:
Ok Thanks, I will work with it some more.  I am trying to follow the instructions and gif tutorial to make things easier.   The message could say something like "You selected a part in this PCB which is not in the proper place of the hierarchical structure contained in the previously saved restore file."   I am sure you will come up with something better though. I will play around with it some more. Thanks again for helping.

MitjaNemec, 2020-03-10:
Yeah, I know, my plugins lack proper documentation. This is on my todo list, but I can't seem to get the motivation to do it.

Anyhow, I've pushed a fix for it so now the user should get a more meaningful message.

Thanks for testing, I really appreciate it.

roboart2019, 2020-03-11:
Thanks for your  help.  If you have a chance.  following the tutorial, I noticed some tracks appear in the wrong area upon restoring.  I can take screen shots, but perhaps you will get the same results when you run it.

[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4316277/save_restore_layout.log)

MitjaNemec, 2020-03-11:
This is probably what I've seen in https://github.com/MitjaNemec/Kicad_action_plugins/issues/80#issuecomment-589009743. I'll have to look into it.

roboart2019, 2020-03-12:
I think we are in comment 80, which comment details some tracks appearing in wrong place, then I will read and comment there if necessary.

MitjaNemec, 2020-03-12:
Since the original issue is partly solved, I am closing this issue. For misplaced tracks, I've opened #83 and when I'll make some progress I'll ping you there so that you could test it. Again thanks for all the feedback

closed 2020-03-12 by MitjaNemec

MitjaNemec, 2020-03-20:
@roboart2019 I've fixed the track placement issue. Can you test it? Please report in #83.