MitjaNemec/Kicad_action_plugins #82: place_footprints: ability to place selected footprints

easyw, 2020-02-22:
Hi, it would be possible to position a selection of footprints applying the same offset (x,y) to the corresponding footprints referred in a base sheet?
I mean if I have a Sheet instantiated many times, after having routed the first instance, I would like to select the footprints of the routed part; then running the plugin selecting the sheet to be replicate/positioned, the corresponding footprints would be placed with an offset referred to the position of the selected footprints.

easyw, 2020-02-22:
I have found it can be done starting with this plugin and then using the replicate plugin.
thx

closed 2020-02-22 by easyw

MitjaNemec, 2020-02-22:
You might have been tripped as the original replicate layout plugin worked as you expected. But I've since split it into two plugins as the place footprints can be used by itself. And code management is easier.