MitjaNemec/Kicad_action_plugins #83: Restore layout sometimes misplaces tracks in 5.99

MitjaNemec, 2020-03-12:
I noticed some tracks appear in the wrong area upon restoring.  I can take screen shots, but perhaps you will get the same results when you run it.

[save_restore_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4316277/save_restore_layout.log)

_Originally posted by @roboart2019 in https://github.com/MitjaNemec/Kicad_action_plugins/issues/80#issuecomment-597458658_

MitjaNemec, 2020-03-14:
I've found and fixed this. It was a latent bug which got exposed in 5.99 when the iterators changed from DLIST to std::deque.

@roboart2019 If you can, please test it.

roboart2019, 2020-03-21:
Hi, sorry for delay, it seems to be working with the demo board.  I will do further testing in the upcoming days with the board I am working on.  Thanks for your help addressing these issues.

closed 2020-08-03 by MitjaNemec