MitjaNemec/Kicad_action_plugins #84: UUIDs in latest development versions

jvpap, 2020-03-21:
They are longer than the timestamps used til now, but they contains the value from the timestamps.
So the easiest solution is something like this:
--- a/replicate_layout/replicatelayout.py
+++ b/replicate_layout/replicatelayout.py
@@ -189,13 +189,13 @@ class Replicator():
     @staticmethod
     def get_module_id(module):
         """ get module id """
-        module_path = get_path(module).split('/')
+        module_path = map(lambda x: x[-8:].upper(),get_path(module).split('/'))
         module_id = "/".join(module_path[-1:])
         return module_id
 
     def get_sheet_id(self, module):
         """ get sheet id """
-        module_path = get_path(module).split('/')
+        module_path = map(lambda x: x[-8:].upper(),get_path(module).split('/'))
         sheet_id = module_path[0:-1]
         sheet_names = [self.dict_of_sheets[x][0] for x in sheet_id if x]
         sheet_files = [self.dict_of_sheets[x][1] for x in sheet_id if x]

MitjaNemec, 2020-03-21:
Thanks for pointing this out. My solution was sloppy and I've corrected it in Save/Restore plugin [1], but I did not port the patch here. Your approach is also somewhat more readable than mine, so I'll use your's. If you want to contribute, you can open a PR but move the patch into the get_path() method, so that the plugin is compatible with 5.1.x and 5.99

Thanks

[1]:
```
# V5.1.x and 5.99 compatibility layer
def get_path(module):
    path = module.GetPath()
    if hasattr(path, 'AsString'):
        path = path.AsString()
        # clean up the path
        parts = path.split("/")
        cleaned_parts = []
        for part in parts:
            # tidy up the path ID
            cleaned_id = part.replace("-", "").lstrip("0").upper()
            # left pad zeros, if they were removed
            if 1 < len(cleaned_id) < 8:
                cleaned_id = cleaned_id.rjust(8, '0')
            cleaned_parts.append(cleaned_id)
        path = "/".join(cleaned_parts)
    return path
```

jvpap, 2020-03-22:
If you move the code into get_path(), the logger output in get_sheets_to_replicate() will also use the cleaned path string. Not sure, if you want that...
Anyway, in my opinion, it's in question if the timestamp -> uuid process will be continued.
Just now i'm having a lot of trouble with inconsistencies and even segfaults with kicad 5.99 because of some code use the uuids, some the older timestamp solution.
Have to write a lot of code just to keep my eda projects running.
I think we have to be patient and see were the road takes us with 5.99.
Maybe the cleaning solution from this issue is only useful for nerds who always want to use the latest and newest features of kicad - for the moment ;)

MitjaNemec, 2020-03-23:
Yes, I'd rather keep everything with path mangling in one place.

As far as I can tell timestamp is still 32bit value in KiCad internals. Devs just swapped the container for it and it is the C++->python where this shows. If you look at the `.kicad_pcb` files you can see that `tstamp `is encoded with 8 character (32 bits). They might be trimming the leading zeros and actual ID internally is 64 bit, but we'll see this after 2038.

I'll move this to get_path() in a day or so.

MitjaNemec, 2020-03-28:
I've finally found some time to fix this. If you could test it and report back, I'd appreciate it.

jvpap, 2020-04-05:
Just tested it.
It's working fine.
Thank you very much.

closed 2020-04-05 by jvpap