MitjaNemec/Kicad_action_plugins #85: save_restore_layout.action_save_restore_layout 261:Fatal error when saving layout

pogojens, 2020-03-26:
Hey :)
first things first: thank you for your great Job with this plugin! It's really nice stuff! And it works with a lot of my hirachical sheets.

I found a Problem: when I'm trying to save some of my layouts i recieve a error Message.

03-26 10:12:32 save_restore_layout.action_save_restore_layout 160:Plugin executed on: 'win32'
03-26 10:12:32 save_restore_layout.action_save_restore_layout 161:Plugin executed with python version: '2.7.16 (default, Aug 15 2019, 17:51:40)  [GCC 9.2.0 64 bit (AMD64)]'
03-26 10:12:32 save_restore_layout.action_save_restore_layout 162:KiCad build version: Unknown
03-26 10:12:32 save_restore_layout.action_save_restore_layout 163:Save/Restore Layout plugin version: 43 started
03-26 10:12:33 save_restore_layout.action_save_restore_layout 200:Save layout chosen
03-26 10:12:33 save_restore_layout.save_restore_layout 851:Saving the current board temporary in order to leave current layout intact
03-26 10:12:33 save_restore_layout.save_restore_layout 861:Getting schematics info
03-26 10:12:33 save_restore_layout.save_restore_layout 187:getting project hierarchy from schematics
03-26 10:12:33 save_restore_layout.save_restore_layout 189:Project hierarchy looks like:

//deleted this line cause of personal information

03-26 10:12:33 save_restore_layout.save_restore_layout 863:Getting layout info
03-26 10:12:33 save_restore_layout.save_restore_layout 262:getting a list of all footprints on board
03-26 10:12:33 save_restore_layout.save_restore_layout 865:Updating layout info with schematics info
03-26 10:12:34 root 53:highlighting module: u'FL1401'
03-26 10:12:34 root 53:highlighting module: u'C1403'
03-26 10:12:34 root 53:highlighting module: u'C1402'
03-26 10:12:34 root 53:highlighting module: u'C1401'
03-26 10:12:37 save_restore_layout.save_restore_layout 947:Saving layout for level: ['filter_test', 'sheet5E7CFC80']
03-26 10:12:37 save_restore_layout.save_restore_layout 948:Calculating hash of the layout schematics
03-26 10:12:37 save_restore_layout.save_restore_layout 952:Saving hash for files: [u'001_filter_PWR_PWM.sch']
03-26 10:12:37 save_restore_layout.action_save_restore_layout 261:Fatal error when saving layout
Traceback (most recent call last):
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/action_save_restore_layout.py", line 259, in Run
    save_layout.save_layout(pivot_mod, pivot_mod.sheetname[0:index + 1], layout_file)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 956, in save_layout
    md5hash = self.schematics.get_sch_hash(sch_file, md5hash)
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/save_restore_layout/save_restore_layout.py", line 213, in get_sch_hash
    md5hash.update(line.encode('utf-8'))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 2: ordinal not in range(128)

Does somebody has any ideas?

I'm using:

Application: KiCad
Version: (5.1.5)-3, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013
Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

pogojens, 2020-03-27:
EDIT: found the Problem and a workaround - it has to do with öäü character. It is not possible to use them in footprints and even not in comments. The Workaround is not to use them ;-)
greetings, Jens

MitjaNemec, 2020-03-27:
Thanks for diagnosing it. I'd still rather make this work so that the plugin can handle what KiCad can handle. Can you prepare and attach a test project.

If I'll not be able to fix it, I'll at least generate meaningful error message.

MitjaNemec, 2020-08-03:
@pogojens ping. If you have the time and will could you test the plugin again for this issue as it should be solved now?