MitjaNemec/Kicad_action_plugins #86: Replicate layout: cryptic error in 5.99

CantBelieveThisWorks, 2020-03-26:
Hello,
I tried running replicate layout in a Kicad nightly version and was confronted with the following error: u'00000000-0000-0000-0000-00005e7d2d12. 

[image]

I recreated a minimal project with one main sheet and two (identical) sub sheets, that is attached and creates the same error.

Simply select one of the resistors (all resistors are from the subsheet) of the attached project in pcbnew and invoke replicate layout.

Please have a look at this and see whether this is caused by user fault, incompatibility of replicate layout with KiCad 5.99 or a quirk in the given nightly version.
It should be noted that the most recent nightly version does not show any external plugins.
Thanks!

[replicate_layout.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4390392/replicate_layout.log)
[test.zip](https://github.com/MitjaNemec/Kicad_action_plugins/files/4390383/test.zip)

Kicad Version:
Application: KiCad
Version: (5.99.0-1098-gfbd01f2f7), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    Build date: Mar 11 2020 22:01:41
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

MitjaNemec, 2020-03-27:
Yes this is on me. I've introduced a patch to solve the 5.99 compatibility, but it was incomplete and has a bug. This is the same as #84. I'll try to fix this over the weekend.

Thanks for reporting it

MitjaNemec, 2020-03-28:
I've pushed a fix. Could you try it and report back?

CantBelieveThisWorks, 2020-03-29:
Just tested it, works a treat now! Thanks a lot for your work and the quick fix :D

closed 2020-03-29 by CantBelieveThisWorks