MitjaNemec/Kicad_action_plugins #90: Put a note for nightlies install location in README for new preferences location

a7hybnj2, 2020-04-07:
I am running the macOS nightlies:
`~/Library/Preferences/kicad/5.99/scripting/plugins`
I don't know if this is going to be a permanent change but its how it is for now. Took me a few minutes to figure out what was going on.

MitjaNemec, 2020-04-08:
Thanks for the notice. Before I proceed can you please run 
``` python
import pcbnew
print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)
```
in the pcbnew python console and report back what you get.

a7hybnj2, 2020-04-08:
```
print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)
/private/var/folders/n1/qjm83k_105312fpt4p66q1rc0000gn/T/AppTranslocation/6D98930B-A771-4527-ADE1-B9F5180F295B/d/kicad.app/Contents/SharedSupport/scripting
/private/var/folders/n1/qjm83k_105312fpt4p66q1rc0000gn/T/AppTranslocation/6D98930B-A771-4527-ADE1-B9F5180F295B/d/kicad.app/Contents/SharedSupport/scripting/plugins
/Users/ME/Library/Preferences/kicad/5.99/scripting
/Users/ME/Library/Preferences/kicad/5.99/scripting/plugins
```

MitjaNemec, 2020-04-08:
This is due to the [Nightly users: changes to settings](https://forum.kicad.info/t/nightly-users-changes-to-settings/21385). I'll put a notice up.

closed 2020-04-08 by MitjaNemec

a7hybnj2, 2020-04-08:
Thanks. I just thought it would help someone else not waste 10 minutes trying to figure it out.