MitjaNemec/Kicad_action_plugins #91: Archieve Project: Missing already achieved 3D Models

der-ule, 2020-04-16:
Hello Mitja, first of all, I really appreciate all the effort that you have put to this and your other plugins and I would like to thank you for them, they are such a time saver.

I started using the Archive Project plugin and it works great! However, I noticed that when I had run it already once, if I wanted to run it again on a project that has been achieved already, it kinds of get "confused" when trying to overwrite the existing 3D models.

Here is an screenshot with the error message and the folder containing the 3D models previously archived, as you can see, the already exist in the path that the software tries to save them and instead of reporting that, the software reports that they do not exists. 

This is of course a nitpick and low  priority, just something that I noticed.

I hope this helps, take care!

[image]

[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4488513/archive_project.log)

```
Application: Pcbnew
Version: (5.1.5)-3, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

MitjaNemec, 2020-04-17:
Thanks for reporting.

I'll look into it

MitjaNemec, 2020-06-14:
I cannot recreate this. I put additional logging in the plugin and I'd like to ask you if you can try again and send me back the .log file. You might want to trim the .log file as it reveals a lot of information (all environment variables)

MitjaNemec, 2020-08-03:
@der-ule ping. 

Can you provide any additional feedback on this? As I can not recreate it I'll close the issue otherwise.

der-ule, 2020-08-04:
Hi Mitja, thank you for the reminder, I am attaching the new generated log files. 

Currently I am working in three projects and those where the projects where I got the error to begin with, so I open KiCAD and ran the script in the first project, all worked as expected, from KiCAD I opened the next project, ran the script and got the error as before, I opened the next project, ran the script and the error appeared again, I took a peek to the log file and somehow the path of the models was still referencing the first project that I opened. 

I closed KiCAD, opened again with the last Project loaded and ran the script again, the errors where gone and the log output looks fines unfortunately the previous log file got lost.

[image]

[archive_project1.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/5020760/archive_project1.log)   OK
[archive_project2.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/5020761/archive_project2.log)   NOK
[archive_project3.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/5020762/archive_project3.log)   OK

I hope this helps

MitjaNemec, 2020-08-04:
I can confirm that there is a bug in KiCad. If you open one project, open pcbnew, start python console and run
```python
import os
os.getenv("KIPRJMOD")
```
you'll get the path to the currently opened project. But if you then close this project and open another one, open pcbnew, start the python console and run the same code again, you'll get the path to the previous project. I reported [this](https://gitlab.com/kicad/code/kicad/-/issues/5071).

Thanks for help and debugging. I'll close this issue when there is a fix in KiCad

der-ule, 2020-08-04:
Thank **you** for the extremely useful plugins and the time that you put on this, I am very glad that I could somehow help :)

MitjaNemec, 2020-08-27:
This is now fixed in testing build of 5.1.x branch. Once 5.1.7 is released I'll close the issue.

If somebody wants to stay at 5.1.6 or lower he/she can implement a [workaround](https://gitlab.com/kicad/code/kicad/-/issues/5071#note_392790034) by adding a line
```python
os.environ["KIPRJMOD"] = os.path.abspath(os.path.dirname(board.GetFileName()))
```
in beginning of `archive_3D_models ` function

closed 2021-08-17 by MitjaNemec