MitjaNemec/Kicad_action_plugins #92: 3D model search path

ferdymercury, 2020-06-02:
The archive project plugin works nicely, but there is an issue if one defines a 3D model search path (not an environmental variable), see https://forum.kicad.info/t/help-understanding-3d-search-paths-and-environment-variables/20726

In this case, the footprint is just ABCD.step and the plugin is unable to find it and copy it to the archived folder.

I found a hacky workaround, namely to add in line https://github.com/MitjaNemec/Kicad_action_plugins/blob/master/archive_project/archive_project.py#L582 the following:

```
if not model_path.startswith('$') and not model_path.startswith('/'):
    model_path = '/My3dsearchPath......../' + model_path
```

Probably there is a way to generalize so that a fix can be commited to master?

MitjaNemec, 2020-06-04:
Thanks for reporting The plugin should handle custom environment variables. I'll add some additional logging and I'll ask you to rerun the plugin and send me back the archive_project.log file so that I'll narrow down the issue. It might be an OS related issue.

ferdymercury, 2020-06-04:
https://gitlab.com/kicad/code/kicad/-/issues/3792

MitjaNemec, 2020-06-04:
I've changed the 3D model archiving code a bit. Can you please rerun the plugin and report back the results and send me the `archive_project.log` file.

ferdymercury, 2020-06-05:
It was a colleague of mine in Windows who had the problem and I proposed him the workaround which is now working for him. I can ask him if he can check your new version to see if it is solved by it, too.

gkasprow, 2020-06-05:
I checked the new version. It behaves differently, it does not report error with copying the 3D files. But does not copy them. Below is the log.
[archive_project.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4738101/archive_project.log)
To make the original code working, I added with help of @ferdymercury following lines starting with 584
            if not model_path.startswith('$') and not model_path.startswith('D:\\........\\StepLib\\'):
                logger.info("replace the model path: " + model_path)
                model_path = 'D:\\............\\StepLib\\' + model_path

where the path:\\StepLib is defined:
 KISYS3DMOD	D:\.........\StepLib\

MitjaNemec, 2020-06-14:
@gkasprow I've made a few changes to the plugin and I've added additional logging, which should help me narrow it down. If you can try again and send me back the .log file I'd appreciate it. You might want to trim the .log file first as it will reveal a lot of personal information (all environment variables present on your machine)

aracosta1, 2022-01-20:
With version 6 kicad changed the environment variables so KISYS3DMOD is no more used and in new installation is not even defined. the ew variable is KICAD6_3DMODEL_DIR.

MitjaNemec, 2022-01-20:
@aracosta1 This should not matter as the plugin grabs and looks at all defined environment variables. So it does not matter if the path is `KICAD6_3DMODEL_DIR` or `KISYS3DMOD`. The plugin will look for the variable how the 3D model is defined, and if found, it will expand the path accordingly. If the variable is not defined, the plugin should at least log this in the .log file.

But I agree that the change of the path has caused some issue to some users. One would hope that if you've gotten bit by this, you'll start using your own libraries.