MitjaNemec/Kicad_action_plugins #93: archive project hangs when downloading a file

ferdymercury, 2020-06-02:
The archive project plugin froze KiCad after trying to download one of the manuals.

Here the last lines in the log file after I waited 10 minutes and killed the process:
```
06-02 12:19:06 archive_project.archive_project 433:Trying to archive documentation file: "https://media.digikey.com/pdf/data%20sheets/cooper%20bussmann%20pdfs/ccfl_transformers_rev2015.pdf"
06-02 12:19:06 archive_project.archive_project 440:File is encoded with URL
06-02 12:19:06 archive_project.archive_project 456:Downloading file
```

ferdymercury, 2020-06-02:
I found a workaround by adding the following:
```
link = link.replace('%20',' ')
```

MitjaNemec, 2020-06-04:
Thanks for reporting and providing the workaround. I'll have to take a look into this and provide a more universal solution (there might be links with other % encodings)

MitjaNemec, 2020-06-05:
I'd really like to avoid relying on requests package. It is an external dependency and this will significantly complicate plugin installation in Windows and maybe also on macOs. I am assuming you are running on Linux? On Win and macOs platforms, KiCad uses its own interpreter, which is installed with KiCad and comes with bare minimum of packages. So user would have to install the requests package to KiCad python interpreter.

If I find a nice way to install request at least into KiCad on Windows, I'll consider using it.

ferdymercury, 2020-06-05:
Yes, I am running on Linux and installed the requests package with pip.

I understand, sure, if there is a solution without extra packages, much better. I think it can be probably fixed with some mapping dictionary of special characters.

The 403 error might be however more hard to solve with urlretrieve, I tried this but did not work https://stackoverflow.com/questions/34957748/http-error-403-forbidden-with-urlretrieve. If no workaround solves it, one might think of a hybrid approach: if the standard urlretrieve query failed, check if user has requests installed and try downloading with that one. This does not force Windows user to install it, but at the same time allows other users to get full docu if package is included.