MitjaNemec/Kicad_action_plugins #95: archive_project documentation fails with some urls

ferdymercury, 2020-06-02:
For example:
`https://www.infineon.com/dgdl/irf7601pbf.pdf?fileId=5546d462533600a4015356038df01c74
`
This is because of two reasons:
- it calls the function .lower() and changes fileId to fileid
- it does not accept urls if thy do not end with .pdf

ferdymercury, 2020-06-02:
Using the "requests" package from python3 solves all these issues. My new version also supports non-pdf filenames. See below.

```
from __future__ import absolute_import, division, print_function
import pcbnew
import os
import os.path
import shutil
import sys
import logging
import re
import hashlib
from shutil import copyfile
try:
    from urllib import urlretrieve
except:
    from urllib.request import urlretrieve
import urllib
import requests
import re

logger = logging.getLogger(__name__)

# get version information
version_filename = os.path.join(os.path.dirname(os.path.realpath(__file__)), "version.txt")
with open(version_filename) as f:
    VERSION = f.readline().strip()

# > V5.1.5 and V 5.99 build information
if hasattr(pcbnew, 'GetBuildVersion'):
    BUILD_VERSION = pcbnew.GetBuildVersion()
else:
    BUILD_VERSION = "Unknown"

def balanced_braces(args):
    if isinstance(args, str):
        args = [args]
    parts = []
    for arg in args:
        if '(' not in arg:
            continue
        chars = []
        n = 0
        for c in arg:
            if c == '(':
                if n > 0:
                    chars.append(c)
                n += 1
            elif c == ')':
                n -= 1
                if n > 0:
                    chars.append(c)
                elif n == 0:
                    parts.append(''.join(chars).lstrip().rstrip())
                    chars = []
            elif n > 0:
                chars.append(c)
    return parts

def remove_braced_content(args):
    if isinstance(args, str):
        args = [args]
    parts = []
    for arg in args:
        if '(' not in arg:
            continue
        chars = []
        n = 0
        for c in arg:
            if c == '(':
[... 637 lines cut ...]

# for testing purposes only
if __name__ == "__main__":
    # if debugging outside of this folder change the folder
    os.chdir(os.path.dirname(os.path.realpath(__file__)))

    file_handler = logging.FileHandler(filename='archive_project.log', mode='w')
    stdout_handler = logging.StreamHandler(sys.stdout)
    handlers = [file_handler, stdout_handler]
    # handlers = [file_handler]

    logging.basicConfig(level=logging.DEBUG,
                        format='%(asctime)s %(name)s %(lineno)d:%(message)s',
                        datefmt='%m-%d %H:%M:%S',
                        handlers=handlers
                        )

    logger = logging.getLogger(__name__)
    logger.info("Plugin executed on: " + repr(sys.platform))
    logger.info("Plugin executed with python version: " + repr(sys.version))
    logger.info("KiCad build version: " + BUILD_VERSION)
    logger.info("Archive plugin version: " + VERSION + " started in standalone mode")

    main()
```

MitjaNemec, 2020-06-04:
Thanks for reporting the issue and for providing the workaround. I'll look into it but I'll need to make sure it works with python2 (Windows and macOS). But this might take some time as I currently don't have much free time available.