MitjaNemec/Kicad_action_plugins #99: Replicate Layout doesn't work on 5.1.6

tulskij-pryanic, 2020-07-04:
Hello, after update to 5.1.6 replicate layout unable in plugin list. 
MacOS 10.13.6
[image]
[image]

MitjaNemec, 2020-07-08:
Hi, thanks for reporting it. Can you look at replicate_layout folder and look for a .log file (specifically `Replicate_layout_error.log`) and if it exists can you please attach it here.

tulskij-pryanic, 2020-07-08:
Hello, I haven't this file in plugin folder.
[image]

MitjaNemec, 2020-07-08:
Hmm, which version of the plugin do you have (take a look into `version.txt`)? There was a broken version available on github a week or so ago for a couple of hours and you might have that specific version.

Can you also download the latest version (66) and try again?

tulskij-pryanic, 2020-07-08:
Version 66:
[image]
I tried to reinstall, but nothing happened

[image]
[image]

MitjaNemec, 2020-07-08:
I've just pushed a new version that has a minor bug fixed regarding unicode file support. But I somewhat suspect that this is not the issue. The plugin should at least load. or if it fails it should write the `replicate_layout_error.log`. 

It does not help that I don't have any experiences with macOs.

Where did you install the plugin? Did you require any special privileges to write in the folder where the plugin is installed?

tulskij-pryanic, 2020-07-08:
/Applications/KiCad/kicad.app/Contents/SharedSupport/scripting/plugins/replicate_layout
This way. And before I use this plugin about 1 year without any problems. After update to 5.1.6 it stop working.
Other plugins from you pack works well and I tried some other without any problem.

> I've just pushed a new version that has a minor bug fixed regarding unicode file support. But I somewhat suspect that this is not the issue.
Yes, Replicate layout start working, but now other unable (save-restore, wap units, swap pins). After I put replicate layout from latest and other plugin from previous download all OK) Strange thing...

MitjaNemec, 2020-07-09:
I am totally in the dark. Can you post complete KiCad version info?

tulskij-pryanic, 2020-07-09:
[image]

MitjaNemec, 2020-07-09:
Thnx, but I'd like to see complete info as shown below
[image]

tulskij-pryanic, 2020-07-09:
[image]

MitjaNemec, 2020-07-09:
Thnx, I just wanted to confirm you are on python2.7. As this is a tough nut to crack and it seems it is macOs specific, I'll need your help.

For start I'd like you to:
  1. run pcbnew
  2. open scripting console
  3. execute the following code
    a. `import pcbnew`
    b. `print pcbnew.GetKicadConfigPath()`

Thanks

tulskij-pryanic, 2020-07-09:
[image]

import pcbnew:

```
locals()

Type: <type 'dict'>

Value: {'pp': <bound method Display.setItem of <wx.py.crust.Display; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fe7ae8f5000> >>, 'shell': <wx.py.shell.Shell; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fe7ae8cb400> >, '__builtins__': <module '__builtin__' (built-in)>, '__package__': None, 'filling': <wx.py.filling.Filling; proxy of <Swig Object of type 'wxSplitterWindow *' at 0x7fe7ad945810> >, 'notebook': <wx._controls.Notebook; proxy of <Swig Object of type 'wxNotebook *' at 0x7fe7ad9451a0> >, 'pcbnew': <module 'pcbnew' from '/private/var/folders/5z/724347s91cz2c75nnpj1590w0000gn/T/AppTranslocation/DC5D09FE-4DFD-4154-9C08-911EE457A53F/d/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.pyc'>, '__name__': '__main__', '__doc__': None}
```

[image]

print pcbnew.GetKicadConfigPath():
[image]

```
locals()

Type: <type 'dict'>

Value: {'pp': <bound method Display.setItem of <wx.py.crust.Display; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fe7ae8f5000> >>, 'shell': <wx.py.shell.Shell; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fe7ae8cb400> >, '__builtins__': <module '__builtin__' (built-in)>, '__package__': None, 'filling': <wx.py.filling.Filling; proxy of <Swig Object of type 'wxSplitterWindow *' at 0x7fe7ad945810> >, 'notebook': <wx._controls.Notebook; proxy of <Swig Object of type 'wxNotebook *' at 0x7fe7ad9451a0> >, 'pcbnew': <module 'pcbnew' from '/private/var/folders/5z/724347s91cz2c75nnpj1590w0000gn/T/AppTranslocation/DC5D09FE-4DFD-4154-9C08-911EE457A53F/d/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.pyc'>, '__name__': '__main__', '__doc__': None}
```

MitjaNemec, 2020-07-09:
Thx, can you try moving the plugin from the original folder (`/Applications/KiCad/kicad.app/Contents/SharedSupport/scripting/plugins/replicate_layout`) to `/users/antonbushinsky/Library/Preferences/kicad/scripting/plugins/replicate_layout` and restart KiCad.

MitjaNemec, 2020-07-09:
According to https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Installation you should install the plugins in different folder than you did. So this might be an issue. Or at least is masking an issue as the plugin probably can not write into its folder thus we can not find the `replicate_layout_error.log` file

tulskij-pryanic, 2020-07-09:
Hello, in this folder it doesn't work at all
[image]
[image]
I wrote it about year or year and half ago when I had problem with replicate layout (it didn't create traces, but now it's OK)

tulskij-pryanic, 2020-07-09:
Ok, I tried ti put plugins in 
/Volumes/Macintosh HD/Users/antonbushinskiy/Library/Preferences/kicad/scripting/plugins
but from latest download not all avalible(
[image]

[image]

tulskij-pryanic, 2020-07-09:
Kicad_action_plugins-master (2).zip
Kicad_action_plugins-master (3).zip
From this two downloads. Firs is "old" and second the latest. When you say that fixed unicode.

MitjaNemec, 2020-07-10:
OK, there are a couple of issues. 

Since you've moved the plugin to another folder the plugin is at least registered. But then:
  1. some of the plugins do not register (at least the latest versions)
  2. you get an error when trying to run the plugin.
  3. For whatever reason you can not run the plugins from specific folders (%APPDATA%, ...)

For 1. can you look into the plugin folder for `*_error.log` and attache it here?

For 2. can you run the following code in pcbnew scripting console (you have to copy it line by line not all at once) and paste the output here.
```python
import wx
a = wx.GetTopLevelWindows()
for item in a:
    print item.GetTitle()
```
I suspect that since you are running Cyrillic alphabet the plugin can not find the window with name "pcbnew". I'd like to confirm this.

As for why this folder works with 5.1.6 and the previous folder worked with 5.1.5, if you have a chance run the following code in 5.1.5 and in 5.1.6 and paste the output
```
import pcbnew
print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)
```

Thanks for hanging in here. I know that this remote debugging is a real PITA.

tulskij-pryanic, 2020-07-10:
I haven't error log in plugin folder (when use this way: /Volumes/Macintosh HD/Users/antonbushinskiy/Library/Preferences/kicad/scripting/plugins)

For me it's not a problem put plugin in .app (except I need to recopy after update). Main problem - that few plugins not available after update to 5.1.6. But as I sayd I can resolve this problem when I use some of them from "previous" version. 
In 5.1.5 I used folder inside .app too.
[image]

```
locals()

Type: <type 'dict'>

Value: {'a': wxWindowList: [<wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fdc98859c00> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fdc9751c900> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fdc981e2200> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fdc9c88bb80> >, <wx._windows.MiniFrame; proxy of <Swig Object of type 'wxMiniFrame *' at 0x7fdc9c8a2780> >, <kicad_pyshell.PcbnewPyShell; proxy of <Swig Object of type 'wxFrame *' at 0x7fdc976b6fe0> >], 'pp': <bound method Display.setItem of <wx.py.crust.Display; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fdc98011000> >>, 'shell': <wx.py.shell.Shell; proxy of <Swig Object of type 'wxStyledTextCtrl *' at 0x7fdc9823fe00> >, '__builtins__': <module '__builtin__' (built-in)>, '__package__': None, 'filling': <wx.py.filling.Filling; proxy of <Swig Object of type 'wxSplitterWindow *' at 0x7fdc974d4690> >, 'notebook': <wx._controls.Notebook; proxy of <Swig Object of type 'wxNotebook *' at 0x7fdc974d37d0> >, '__name__': '__main__', '__doc__': None, 'wx': <module 'wx' from '/private/var/folders/5z/724347s91cz2c75nnpj1590w0000gn/T/AppTranslocation/1C2B78DE-B412-4987-9624-9D06AD80D6A9/d/kicad.app/Contents/Frameworks/python/site-packages/wx-3.0-osx_cocoa/wx/__init__.py'>}
```
[image]
```
KiCAD:PCBNEW - Python Shell - PyAlaMode 0.9.8
import pcbnew
print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)
/private/var/folders/5z/724347s91cz2c75nnpj1590w0000gn/T/AppTranslocation/1C2B78DE-B412-4987-9624-9D06AD80D6A9/d/kicad.app/Contents/SharedSupport/scripting
/private/var/folders/5z/724347s91cz2c75nnpj1590w0000gn/T/AppTranslocation/1C2B78DE-B412-4987-9624-9D06AD80D6A9/d/kicad.app/Contents/SharedSupport/scripting/plugins
/Users/antonbushinskiy/Library/Preferences/kicad/scripting
/Users/antonbushinskiy/Library/Preferences/kicad/scripting/plugins

```

MitjaNemec, 2020-07-10:
Ok the folders listed with PLUGIN_DIRECTORIES_SEARCH are the ones where you can put the plugins. And I don't have the slightest idea what the first two ones are.

As for 2, you'll have to repeat and hit on extra enter so the for loop is going to execute, and it'll print out the window names. 

I'll still have to figure out why one version of plugins register and the other doesn't.

MitjaNemec, 2020-07-12:
I've looked at the files you've attached in https://github.com/MitjaNemec/Kicad_action_plugins/issues/99#issuecomment-656051559. The 'newest' version had a bug, that has since been fixed. Sorry about the confusion.

I was in the process of updating the plugins when you reported the issue and I thought that it was the recent changes that produce the error. So I hurried through the update process and I've introduced a couple of bugs. I' should have stopped the update process and fixed on solving the issue. I have little experience what is the proper process when deploying upgrades and an issue is reported during deployment. Lesson learned.

tulskij-pryanic, 2020-07-12:
Ok, I tryed! But still have issues. When I put plugins in 
Volumes/Macintosh HD/Users/antonbushinskiy/Library/Preferences/kicad/scripting/plugins
Some of them give error message like in [this](https://github.com/MitjaNemec/Kicad_action_plugins/issues/99#issuecomment-656051559) post. Some of them crash pcbnew (place footprint, save/restore layout, and screenshot not available.

In .app it's OK, but still not all plugins available
[image]

MitjaNemec, 2020-07-13:
For the error mentioned you'll have to repeat the following:

For 2. can you run the following code in pcbnew scripting console (you have to copy it line by line not all at once) and paste the output here.

import wx
a = wx.GetTopLevelWindows()
for item in a:
    print item.GetTitle()

please note that you have to hit 'enter' two times in order that 'for' loop executes.

tulskij-pryanic, 2020-07-15:
[image]
It's ok?

MitjaNemec, 2020-07-15:
Thnx.

Can you execute also
```python
repr([x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().startswith('pcbnew')])
```
and
```python
repr([x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().startswith('pcbnew')][0])
```

MitjaNemec, 2020-07-15:
You'll have to import wx before that

tulskij-pryanic, 2020-07-15:
```
import wx
repr([x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().startswith('pcbnew')])
"[<wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fda9f912c00> >]"
repr([x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().startswith('pcbnew')][0])
"<wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7fda9f912c00> >"

```
[image]

MitjaNemec, 2020-07-15:
I am stumped. The second line was the exact same line that you've had the problem with when running `replicate_layout` plugin reported in screenshot in post https://github.com/MitjaNemec/Kicad_action_plugins/issues/99#issuecomment-656051559.

Can you try running the plugin again?

tulskij-pryanic, 2020-07-15:
> Can you try running the plugin again?
Now it placed inside .app and it works)
After I move plugins to Lib folder:

> Traceback (most recent call last):
> 
>   File "/Users/antonbushinskiy/Library/Preferences/kicad/scripting/plugins/replicate_layout/action_replicate_layout.py", line 318, in Run
>     _pcbnew_frame = [x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().startswith('pcbnew')][0]
> 
> IndexError: list index out of range

MitjaNemec, 2020-07-15:
Whoa. This is totally surprising. Same code runs when runs from .app but if run from Lib folder results in an error. I'll have to get some outside help. Preferably someone familiar with macOs.

From what I can gather (and I'll prepare special version of replicate_layout plugin for you to run so that I can gather additional data) it looks like when the plugin is run from the Lib folder it runs as an separate instance and it does note have access to pcbnew GUI elements (wx). Therefore `wx.GetTopLevelWindows()` most likely returns an empty list. And an empty list obviously does not have an element with index [0].

Why this is so is beyond me.

MitjaNemec, 2020-07-15:
BTW. When you do get an error, do you have the `replicate_layout.log` file in the project folder and what is it's contents?

MitjaNemec, 2020-07-15:
OK, I've pushed a new version of the replicate layout plugin (just the file `action_replicate_layout.py` got updated). Can you run it from both folders and report back. I'd really like to get `replicate_layout.log` file for both cases (delete it before running each time so that we can be certain it is a current version of the file)

Again, thanks for your peristence

tulskij-pryanic, 2020-07-15:
I have no log in plugin folder.
[image]

Should I replace this 2 files in replicate layout folder?

MitjaNemec, 2020-07-15:
Look for the log in the **project** folder.

Which two files. You only need to replace the `replicate_layout.py` file. `.pyc` file is regenerated automatically when the plugin is run.

tulskij-pryanic, 2020-07-16:
Ok, I tried now and it work from both folders_)
[replicate_layout_app.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4932621/replicate_layout_app.log)
[replicate_layout_lib.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/4932622/replicate_layout_lib.log)

MitjaNemec, 2020-08-03:
Sorry for the delay, but I took some time off. Getting back to your issues. @OndrejPopp found and fixed a couple of bugs in #100. So this will help with `save/restore` and `place_footprints `plugins.

Can you load and use all of the plugins now? Is there any other issue, or can I close this issue?

MitjaNemec, 2020-08-03:
Sorry for the delay, but I took some time off. Getting back to your issues. @OndrejPopp found and fixed a couple of bugs in #100. So this will help with `save/restore` and `place_footprints `plugins.

Can you load and use all of the plugins now? Is there any other issue, or can I close this issue?

BTW there was [another user](https://forum.kicad.info/t/rf-tools-on-mac-osx-using-version-5-1-6/24098?u=mitjan) with plugin issues on macOs on KiCad 5.1.6. Might be connected.

OndrejPopp, 2020-08-03:
Thanks Mitja.
I will have a look at it later. At this moment I am busy with something else. But my issue is only about some missing white space indentation. That's all. So, if that got fixed in #100 that's fine by me. I will let you know, after I had a look at it whether you can close this issue ok?

In the meantime, I have started with a more elaborate modification of the replicate layout plugin. May you be interested in that. This may take a while to explain. I am working on a pcb layout of a SPI controllable switch matrix, which is quite repetitive, and it turned out quite difficult and un-maintainable to layout everything at once. The freerouter choked on this and is unable to recognize the repetitive structure. So I am trying to use hierarchical layout for this. Unfortunately hierarchical layout has never been implemented in Kicad, and from what I saw on the forum nobody is working on this either. So I am using (misusing) a project as a design entitiy that has a main schematic and a pcb-layout associated with it, and the whole design consists of several "projects" that instantiate the main files/pcbs of other projects in a certain hierarchy. So to make this work, I need the replicate layout plugin not only to replicate an instance within one pcb-layout, but also from other pcb layouts from other projects. So, a combination of import and replicate. I tried to use save and restore layout for that, but I find it inconvenient to make a separate project for each pcb layout I want to import where I have to instantiate the main schematic of that layout in order to make save/restore to pickup up the right hierarchy. In addition, I saw a bug somewhere in it, maybe that got fixed in #100, but what I was seeing was that save/restore got confused about two footprints and treated them as the same one, so after the restore, I see a gap where the first footprint is supposed to be, while that missing footprint is sitting on top of the other one.... I thought maybe this has something to do with (re)annotation but apparently not. In addition, replicate layout is much more sophisticated because it can also remove existing tracks while replicating and so I decided to extend it with an import function as well.

So, I played around with wx form builder and have extended the gui to be able show project hierarchies of all matching instances or top level pcb's of the anchor instance. That's all I have for now. I still need to write the functionality so I  can replicate from any matching project instance onto the currently opened board. I looked at the code, and I think this will work, I just need to extend the Replicator to have a source board as well alongside the destination board, to include the anchor instance for replication as well when replicating from other projects and refactor some module related code into a seperate Project class. For the rest all the functionality is already there including removing existing tracks, so layout sub blocks can be easily updated.

So that is what I am planning to work on, which is more elaborate than some white space indentation. And so when I have this ready I can generate an other pull request for you, may you be interested in this.

In any case, I will let you know whether the white space indentation problem has been fixed. 

Thanks and kind regards,
Ondrej

⁣Get BlueMail for Android ​

tulskij-pryanic, 2020-08-09:
Hello! Sorry for my delay. 
I think now it's ok:
[image]
I tested some of them and it works! Great!

MitjaNemec, 2020-08-11:
Thanks for the feedback and for persistence. Don't worry about the delay.I am aware that we have lives to live besides KiCad

closed 2020-08-11 by MitjaNemec