MitjaNemec/Kicad_action_plugins #104: swap_units - doesn't appear in the list of plugins

05CARM, 2020-09-22:
I downloaded all of the plugins folders into ~/.kicad_plugins
All of them appear in the drop down list in Pcbnew **except** 'swap_units'
I tried copying swap_units.py to ~/.kicad_plugins and that made no difference.

Application: Pcbnew
Version: 5.1.6-c6e7f7d~87~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-117-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.5.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

MitjaNemec, 2020-09-23:
This might be py2/py3 issue as I don't have any issues on MSWIN platform. Can you look into the swap _unit plugin folder for `Swap_units_error.log` file. If you find it, please attach it here.

05CARM, 2020-09-23:
[Swap_units_error.log](https://github.com/MitjaNemec/Kicad_action_plugins/files/5265877/Swap_units_error.log)

MitjaNemec, 2020-09-24:
Thanks. Sadly I'm still in the dark. Can you please start up pcbnew and run the scripting console.
Then you enter the following lines (copy them line by line and hit `enter` after each line)
```python
import pcbnew
pcbnew.NOT_LOADED_WIZARDS
pcbnew.FULL_BACK_TRACE
```
Then please send me back the feedback.

05CARM, 2020-09-24:
Will this do? Did I do it right?
[image]
If not, let me know how you want the results recorded.

MitjaNemec, 2020-09-25:
This is alright, but it does not point me to anywhere useful. So it'll take me some time to figure out how to proceed, but I'll need to add additional instrumentation to gather additional info what is going on and I'll need your help as I can not recreate the issue. Hang in there.

05CARM, 2020-09-25:
Thank you.
2 of your other plugins are very helpful to me.