MitjaNemec/Kicad_action_plugins #106: Plugin error on Nightly 5.99

finale-z, 2020-10-03:
Replace Layout,Place Layout and Save/Restore Layout run incorrectly on nightly 5.99.
The wrong code is this sentence
`
_pcbnew_frame = [x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().startswith('pcbnew')][0]
`
The contents of the error report are “IndexError:list index out of range"
The operation log of Replace Layout is as follows
`
10-03 15:03:33 replicate_layout.action_replicate_layout 305:Plugin executed on: 'linux'
10-03 15:03:33 replicate_layout.action_replicate_layout 306:Plugin executed with python version: '3.8.5 (default, Sep  5 2020, 10:50:12) \n[GCC 10.2.0]'
10-03 15:03:33 replicate_layout.action_replicate_layout 307:KiCad build version: (5.99.0-3484-g9efc91c10)
10-03 15:03:33 replicate_layout.action_replicate_layout 308:Replicate layout plugin version: 68 started
10-03 15:03:33 replicate_layout.action_replicate_layout 319:wx top level windows: WindowList: [<wx._core.Frame object at 0x7fdce0133f70>, <wx._core.Frame object at 0x7fdce0146040>, <wx._core.Frame object at 0x7fdce01460d0>, <wx._core.Frame object at 0x7fdce0146160>]
`

MitjaNemec, 2020-10-05:
Currently I am supporting only 5.1.x branch. Though for this bug (there are many more preventing use on 5.99) I'd ask you if you can paste complete KiCad build (version) info. I think it is connected to the wx version

finale-z, 2020-10-06:
This is my KiCad version info
`
Application: KiCad

Version: (5.99.0-3484-g9efc91c10), release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.72.0 OpenSSL/1.1.1h zlib/1.2.11 zstd/1.4.5 libidn2/2.3.0 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.41.0

Platform: Linux 5.4.67-1-MANJARO x86_64, 64 bit, Little endian, wxGTK

Build Info:
	Date: Sep 22 2020 19:04:49
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.72.0
	OCC: 7.4.0
	Curl: 7.72.0
	ngspice: 32
	Compiler: GCC 10.2.0 with C++ ABI 1014

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
	KICAD_SCRIPTING_ACTION_MENU=ON
	BUILD_GITHUB_PLUGIN=ON
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
`
And I tried to run these three scripts on the next two build versions and got the same error. It seems that Wx can't get the window  of pcbnew

Miceuz, 2021-01-24:
I have dabbled with this a bit.

I have noticed that now 'pcbnew' is at the end of title for 5.99 version of KiCad. Thus I have updated the code to

```
_pcbnew_frame = [x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().endswith('pcbnew')][0]
```

Miceuz, 2021-01-24:
After catching up with trivial changes like file name change from .sch to .kicad_sch and Modules renamed to Fooprints I am stuck at this:
[image]

Log does not give any hints:
```
01-24 22:46:24 replicate_layout.action_replicate_layout 305:Plugin executed on: 'linux'
01-24 22:46:24 replicate_layout.action_replicate_layout 306:Plugin executed with python version: '3.8.5 (default, Jul 28 2020, 12:59:40) \n[GCC 9.3.0]'
01-24 22:46:24 replicate_layout.action_replicate_layout 307:KiCad build version: (5.99.0-8578-g04cc64c008-dirty)
01-24 22:46:24 replicate_layout.action_replicate_layout 308:Replicate layout plugin version: 69 started
01-24 22:46:24 replicate_layout.action_replicate_layout 340:Preparing replicator with U6 as a reference
01-24 22:46:24 replicate_layout.replicatelayout 215:getting project hierarchy from schematics
01-24 22:46:24 replicate_layout.replicatelayout 217:Project hierarchy looks like:
{}
01-24 22:46:24 replicate_layout.replicatelayout 225:getting a list of all footprints on board
```

My KiCad setup is very unconventional - I have it compiled with wxWidgets 3.1.5:
```
Application: Pcbnew

Version: (5.99.0-8578-g04cc64c008-dirty), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.4.0-64-generic x86_64, 64 bit, Little endian, wxGTK, cinnamon, x11

Build Info:
	Date: Jan 24 2021 21:37:16
	wxWidgets: 3.1.5 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.65.1
	OCE: 6.9.1
	Curl: 7.68.0
	ngspice: 26
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=ON
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
	KICAD_SCRIPTING_ACTION_MENU=ON
	KICAD_USE_OCE=ON
	KICAD_USE_EGL=ON
	KICAD_SPICE=ON

```

MitjaNemec, 2021-01-25:
Thanks for the info regarding frame name.

As for the plugin and 5.99 You'll not be able to fix it. 5.99 changed schematics file format and the plugin is parsing the schematics files. So this is a big task to refactor the plugin. Also the timestamps got replaced with UUIDs

 So I am postponing it until it is clear if V6 will get eeschema python API or not, as this will have an impact on the schematics file parsing code. Once this is known, I'll start working on the plugins. But the process will be slow as I'll have first to evaluate what kind of architecture suits best for all of my plugins and how should I package the plugins so that they are ready for the promised content (plugin) manager that should also come in V6.

Miceuz, 2021-01-25:
Actually I hacked together a dirty but working solution yesterday - you can see it in pull request I have just created #115

I understand your point regarding doing proper software engineering, this PR is just a demo of what it takes.

EeliK, 2021-02-24:
> 
> 
> I have dabbled with this a bit.
> 
> I have noticed that now 'pcbnew' is at the end of title for 5.99 version of KiCad. Thus I have updated the code to
> 
> ```
> _pcbnew_frame = [x for x in wx.GetTopLevelWindows() if x.GetTitle().lower().endswith('pcbnew')][0]
> ```

Just a note on this... The string is now "PCB Editor", not pcbnew.

EeliK, 2021-02-24:
BOARD.GetModules() has been changed to GetFootprints(). (Line 321)

MitjaNemec, 2021-02-24:
@EeliK you should read the README.md in the master branch. This should take you to the `5.99_test` branch where most of these things have been handled.

Though I haven't tested the plugin on 5.99 and I am still waiting on positive (or negative) feedback. More discussion in #102

EeliK, 2021-02-24:
Dang, I thought I had 5.99_test...

EeliK, 2021-02-27:
> Just a note on this... The string is now "PCB Editor", not pcbnew.

This may be a bigger problem. What worries me is that the string "PCB Editor" is marked as translatable in the source code. I don't find it in a .po file, though.

MitjaNemec, 2021-02-28:
Thanks for the notice. The way to find parent handle always seemed a bit hacky to me. And now with the string becoming translatable I'll have to find another way. I'll have to look at how Qu1ck solves this. If anybody it would be him, finding proper solution