MitjaNemec/Kicad_action_plugins #107: Unable to install.

julian1, 2020-10-27:
README.md says 

```
To install the plugin copy relevant folder into 'scripting/plugins' subfolder of the KiCad configuration folder
```

What is the 'relevant folder'? The top level directory, or a specific directory? Regardless the result seems to be the same. 

``` 
me@debian:~/.kicad_plugins$ pwd
/home/me/.kicad_plugins

me@debian:~/.kicad_plugins$ git clone git@github.com:MitjaNemec/Kicad_action_plugins
Cloning into 'Kicad_action_plugins'...
remote: Enumerating objects: 72, done.
remote: Counting objects: 100% (72/72), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 2781 (delta 41), reused 44 (delta 26), pack-reused 2709
Receiving objects: 100% (2781/2781), 20.56 MiB | 3.74 MiB/s, done.
Resolving deltas: 100% (1947/1947), done.

start kicad

Tools -> external plugins -> refresh plugins. 
No plugins appear in plugins menu. No evidence of any other GUI change.
 
```

Whether I use the top level, or copy a subdir it's the same result.
 

```
me@debian:~/.kicad_plugins$ uname  -a
Linux debian 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1+deb10u1 (2020-04-27) x86_64 GNU/Linux

Application: kicad
Version: 5.0.2+dfsg1-1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.0 OpenSSL/1.1.1d zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3
Platform: Linux 4.19.0-8-amd64 x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 2.24
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.62.0
    Compiler: GCC 8.2.0 with C++ ABI 1013
Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

I have managed to successfully install plugins from other repos into ~/.kicad_plugins, so the path seems fine.

MitjaNemec, 2020-10-27:
Hi!

Let's first make sure where exactly you can install and run the plugins.
Run the `pcbnew`, open python console and run these commands
```python
import pcbnew
print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)
```
If you cannot find the icon to run the scripting console, you'll not be able to run my plugins, as the build you are running lacks support for wxpython. IIRC this is due to GTK libraries mismatch (available wxpython lib on your system seems to be built against GTK2, but the rest is built against GTK3.

And finally, I currently support only 5.1.x branch. You seem to be running 5.0.x branch, so even if you manage to install and run the plugins, they might crash.

julian1, 2020-10-27:
I have never seen and cannot find now a way to get a python console from pcbnew. 

So, it looks like my Kicad version is just too old. I did note the README.me mentioned testing against 5.1, but assumed based on the relative stability of Kicad file formats, that it should be recent enough.

I will check out Kicad_action_plugins again, in future, when I upgrade. 

Thanks for your help!.

closed 2020-10-27 by julian1

MitjaNemec, 2020-10-27:
As said IIRC your on some distributions default wxpython libraries are built against GTK2, but wxwidgets (C++ library) is build against GTK3. Therefore you can not build KiCad with support for wxpython (which is required for python console and action plugins). You might want to check if you can install wxpython libraries built against GTK3 (on some distributions this is possible. You might want to check on the forum.kicad.info. And if it is you might be able to build KiCad with action plugin support.