MitjaNemec/Kicad_action_plugins #108: Replicate Layour fails

iprepiprep, 2020-11-04:
11-03 21:37:28 replicate_layout.action_replicate_layout 305:Plugin executed on: 'win32'
11-03 21:37:28 replicate_layout.action_replicate_layout 306:Plugin executed with python version: '2.7.15 (default, Oct 11 2018, 12:09:51)  [GCC 8.2.0 64 bit (AMD64)]'
11-03 21:37:28 replicate_layout.action_replicate_layout 307:KiCad build version: Unknown
11-03 21:37:28 replicate_layout.action_replicate_layout 308:Replicate layout plugin version: 68 started
11-03 21:37:28 replicate_layout.action_replicate_layout 319:wx top level windows: wxWindowList: [<wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x32d4420> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x7ff890> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x2208c020> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x1fdd2750> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x21adc060> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x1fdd23e0> >, <wx._windows.Frame; proxy of <Swig Object of type 'wxFrame *' at 0x1fdd2e30> >]
11-03 21:37:28 replicate_layout.action_replicate_layout 343:Preparing replicator with U1 as a reference
11-03 21:37:28 replicate_layout.replicatelayout 215:getting project hierarchy from schematics
11-03 21:37:28 replicate_layout.replicatelayout 217:Project hierarchy looks like:
{u'5FB4C553': [u'CurrentControl1', u'C:\\DATA\\Projects\\BenchSensorSimulator\\Electronics\\CurrentControl.sch'], u'5FB4C5BC': [u'CurrentControl2', u'C:\\DATA\\Projects\\BenchSensorSimulator\\Electronics\\CurrentControl.sch']}
11-03 21:37:28 replicate_layout.replicatelayout 225:getting a list of all footprints on board
11-03 21:37:28 replicate_layout.action_replicate_layout 370:source anchor footprint is u'U1'
Located on:[u'CurrentControl1']
With filenames:[u'CurrentControl.sch']
With sheet_id:[u'CurrentControl1']
11-03 21:37:28 replicate_layout.action_replicate_layout 374:Corresponding footprints are 
[(u'U3', [u'CurrentControl2']), (u'U1', [u'CurrentControl1'])]
11-03 21:37:28 replicate_layout.action_replicate_layout 387:Showing dialog
11-03 21:37:28 replicate_layout.action_replicate_layout 391:Pcbnew position: wx.Point(-11, -11)
11-03 21:37:28 replicate_layout.action_replicate_layout 397:Display 0: wx.Rect(0, 0, 3840, 1600)
11-03 21:37:28 replicate_layout.action_replicate_layout 401:Toolbar position: wx.Point(3441, 167)
11-03 21:37:28 replicate_layout.action_replicate_layout 408:Dialog position: wx.Point(3128, 167)
11-03 21:37:33 replicate_layout.replicatelayout 249:construcing a list of sheets suitable for replication on level:u'CurrentControl1', file:u'CurrentControl.sch'
11-03 21:37:33 replicate_layout.replicatelayout 256:Source sheet path up to the level:[u'CurrentControl1']
11-03 21:37:33 replicate_layout.replicatelayout 260:Footprints on the sheets:
[u'U3', u'U1']
11-03 21:37:33 replicate_layout.replicatelayout 263:Footprints raw sheet ids:
[u'/5FB4C5BC/5FB2A6BF', u'/5FB4C553/5FB2A6BF']
11-03 21:37:33 replicate_layout.replicatelayout 279:All sheets to replicate:
[[u'CurrentControl2'], [u'CurrentControl1']]
11-03 21:37:33 replicate_layout.replicatelayout 284:All sheets to replicate sorted:
[[u'CurrentControl1'], [u'CurrentControl2']]
11-03 21:37:33 replicate_layout.replicatelayout 290:All sheets to replicate sorted and without source sheet:
[[u'CurrentControl2']]
11-03 21:37:37 replicate_layout.replicatelayout 249:construcing a list of sheets suitable for replication on level:u'CurrentControl1', file:u'CurrentControl.sch'
11-03 21:37:37 replicate_layout.replicatelayout 256:Source sheet path up to the level:[u'CurrentControl1']
11-03 21:37:37 replicate_layout.replicatelayout 260:Footprints on the sheets:
[u'U3', u'U1']
11-03 21:37:37 replicate_layout.replicatelayout 263:Footprints raw sheet ids:
[u'/5FB4C5BC/5FB2A6BF', u'/5FB4C553/5FB2A6BF']
11-03 21:37:37 replicate_layout.replicatelayout 279:All sheets to replicate:
[[u'CurrentControl2'], [u'CurrentControl1']]
11-03 21:37:37 replicate_layout.replicatelayout 284:All sheets to replicate sorted:
[[u'CurrentControl1'], [u'CurrentControl2']]
11-03 21:37:37 replicate_layout.replicatelayout 290:All sheets to replicate sorted and without source sheet:
[[u'CurrentControl2']]
11-03 21:37:37 replicate_layout.action_replicate_layout 179:Replicating layout
11-03 21:37:37 replicate_layout.replicatelayout 990:Starting replication of sheets: [[u'CurrentControl2']]
on level: [u'CurrentControl1']
with tracks=True, zone=True, text=True, containing=True, remove=False, locked=False
11-03 21:37:37 root 261:updating GUI message: 'Preparing for replication'
11-03 21:37:37 root 266:updating GUI with: 0
11-03 21:37:37 replicate_layout.replicatelayout 556:Getting the list of source footprints
11-03 21:37:37 replicate_layout.replicatelayout 560:Getting the list of all the remaining footprints
11-03 21:37:37 replicate_layout.replicatelayout 564:Getting nets local to source footprints
11-03 21:37:37 replicate_layout.replicatelayout 568:Getting source bounding box
11-03 21:37:37 root 266:updating GUI with: 37
11-03 21:37:37 replicate_layout.replicatelayout 572:Getting source tracks
11-03 21:37:37 replicate_layout.replicatelayout 576:Getting source zones
11-03 21:37:37 replicate_layout.replicatelayout 580:Getting source text items
11-03 21:37:37 replicate_layout.replicatelayout 584:Getting source text items
11-03 21:37:37 root 266:updating GUI with: 87
11-03 21:37:37 root 266:updating GUI with: 100
11-03 21:37:37 root 261:updating GUI message: 'Replicating footprints'
11-03 21:37:37 replicate_layout.replicatelayout 590:Replicating footprints
11-03 21:37:37 root 266:updating GUI with: 0
11-03 21:37:37 replicate_layout.replicatelayout 596:Replicating footprints on sheet [u'CurrentControl2']
11-03 21:37:38 replicate_layout.action_replicate_layout 231:Fatal error when running replicator
Traceback (most recent call last):
  File "C:\Users\A\AppData\Roaming\kicad\scripting\plugins/replicate_layout/action_replicate_layout.py", line 199, in OnOk
    rep_locked=rep_locked)
  File "C:\Users\A\AppData\Roaming\kicad\scripting\plugins/replicate_layout/replicatelayout.py", line 1024, in replicate_layout
    self.replicate_modules()
  File "C:\Users\A\AppData\Roaming\kicad\scripting\plugins/replicate_layout/replicatelayout.py", line 711, in replicate_modules
    dst_mod_text_items[index].SetKeepUpright(src_text.IsKeepUpright())
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 15106, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, TEXTE_MODULE, name)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'TEXTE_MODULE' object has no attribute 'SetKeepUpright'

MitjaNemec, 2020-11-04:
Thanks for reporting.

What is the build information of your KiCad installation?

It looks like you are either running custom 5.1.x build or an old 5.99 build. Anyhow, the plugin can not find the `SetKeepUpright` method when replicating text. This can be a bug in the plugin, or the pcbnew python API changed. So the build information would help me track this down.

iprepiprep, 2020-11-04:
Hi, I'm running (5.0.2)-1 ,release build

iprepiprep, 2020-11-04:
Just upgraded to 5.1.7 and replications works great !

MitjaNemec, 2020-11-04:
Ok, the `SetKeepUpright` is not available in 5.0.x branch. While I am currently supporting only 5.1.x branch you have two options how to proceed:
  1. you can upgrade your KiCad instalation to 5.1.x (5.1.8 has just become available, and you can install program only (without libraries))
  2. you can comment the line 711 in replicatelayout.py. Hopefully the plugin would work.

MitjaNemec, 2020-11-04:
Thanks for the feedback

closed 2020-11-04 by MitjaNemec